MiczFlor/RPi-Jukebox-RFID #1832: 🐛 | Failed to start dev environment in future3/spotify branch

varac, 2022-05-31:
## Bug

### What I did

I followed the [Phoniebox Development Runbook for Docker environments](https://rpi-jukebox-rfid.readthedocs.io/en/latest/developer/docker.html). I could build all docker images fine but when I start the containers with `docker compose -f docker/docker-compose.yml -f docker/docker-compose.linux.yml up` both the jukebox and spotify containter end up in a restart loop.

### What happened

### I expected this to happen

Following the docs should lead to a functional dev env.

### Further information that might help

Last lines of [spotify.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/8802606/spotify.log):

```
...
spotify  | Exception in thread "main" com.electronwill.nightconfig.core.io.ParsingException: Failed to parse data from Reader
spotify  |      at com.electronwill.nightconfig.core.io.ParsingException.readFailed(ParsingException.java:18)
spotify  |      at com.electronwill.nightconfig.core.io.ReaderInput.directRead(ReaderInput.java:23)
spotify  |      at com.electronwill.nightconfig.core.io.AbstractInput.read(AbstractInput.java:37)
spotify  |      at com.electronwill.nightconfig.core.io.CharacterInput.readAndSkip(CharacterInput.java:42)
spotify  |      at com.electronwill.nightconfig.toml.Toml.readUseful(Toml.java:34)
spotify  |      at com.electronwill.nightconfig.toml.TableParser.parseNormal(TableParser.java:50)
spotify  |      at com.electronwill.nightconfig.toml.TomlParser.parse(TomlParser.java:44)
spotify  |      at com.electronwill.nightconfig.toml.TomlParser.parse(TomlParser.java:37)
spotify  |      at com.electronwill.nightconfig.core.io.ConfigParser.parse(ConfigParser.java:113)
spotify  |      at com.electronwill.nightconfig.core.io.ConfigParser.parse(ConfigParser.java:219)
spotify  |      at com.electronwill.nightconfig.core.io.ConfigParser.parse(ConfigParser.java:202)
spotify  |      at com.electronwill.nightconfig.core.file.WriteAsyncFileConfig.load(WriteAsyncFileConfig.java:138)
spotify  |      at com.electronwill.nightconfig.core.file.SimpleCommentedFileConfig.load(SimpleCommentedFileConfig.java:38)
spotify  |      at xyz.gianlu.librespot.player.FileConfiguration.<init>(FileConfiguration.java:85)
spotify  |      at xyz.gianlu.librespot.api.Main.main(Main.java:37)
spotify  | Caused by: java.io.IOException: Is a directory
spotify  |      at java.base/sun.nio.ch.FileDispatcherImpl.read0(Native Method)
spotify  |      at java.base/sun.nio.ch.FileDispatcherImpl.read(FileDispatcherImpl.java:48)
spotify  |      at java.base/sun.nio.ch.IOUtil.readIntoNativeBuffer(IOUtil.java:276)
spotify  |      at java.base/sun.nio.ch.IOUtil.read(IOUtil.java:245)
spotify  |      at java.base/sun.nio.ch.FileChannelImpl.read(FileChannelImpl.java:223)
spotify  |      at java.base/sun.nio.ch.ChannelInputStream.read(ChannelInputStream.java:65)
spotify  |      at java.base/sun.nio.ch.ChannelInputStream.read(ChannelInputStream.java:107)
spotify  |      at java.base/sun.nio.ch.ChannelInputStream.read(ChannelInputStream.java:101)
spotify  |      at java.base/sun.nio.cs.StreamDecoder.readBytes(StreamDecoder.java:284)
spotify  |      at java.base/sun.nio.cs.StreamDecoder.implRead(StreamDecoder.java:326)
spotify  |      at java.base/sun.nio.cs.StreamDecoder.read(StreamDecoder.java:178)
spotify  |      at java.base/java.io.InputStreamReader.read(InputStreamReader.java:181)
spotify  |      at java.base/java.io.BufferedReader.fill(BufferedReader.java:161)
spotify  |      at java.base/java.io.BufferedReader.read(BufferedReader.java:182)
spotify  |      at com.electronwill.nightconfig.core.io.ReaderInput.directRead(ReaderInput.java:21)
```

Last lines of [jukebox.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/8802605/jukebox.log):

```
...
jukebox  | 31.05.2022 06:28:52 -  146:daemon.py          - jb.daemon            - MainThread      - ERROR    - Plugins with errors during load: publishing, volume, player, bluetooth_audio_buttons, gpio
jukebox  | 31.05.2022 06:28:52 -   67:server.py          - jb.rpc.server        - MainThread      - INFO     - Init RPC Server (Pyzmq version: 23.0.0; ZMQ version: 4.1.4; has draft API: False)
jukebox  | 31.05.2022 06:28:52 -   75:server.py          - jb.rpc.server        - MainThread      - DEBUG    - Connected to address 'inproc://JukeBoxRpcServer'
jukebox  | 31.05.2022 06:28:52 -   81:server.py          - jb.rpc.server        - MainThread      - DEBUG    - Connected to address 'tcp://*:5555'
jukebox  | Traceback (most recent call last):
jukebox  |   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/run_jukebox.py", line 67, in <module>
jukebox  |     main()
jukebox  |   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/run_jukebox.py", line 63, in main
jukebox  |     myjukebox.run()
jukebox  |   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/daemon.py", line 227, in run
jukebox  |     self.rpc_server = RpcServer()
jukebox  |   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/rpc/server.py", line 86, in __init__
jukebox  |     self.socket.bind(websocket_address)
jukebox  |   File "/usr/local/lib/python3.9/dist-packages/zmq/sugar/socket.py", line 226, in bind
jukebox  |     super().bind(addr)
jukebox  |   File "zmq/backend/cython/socket.pyx", line 562, in zmq.backend.cython.socket.Socket.bind
jukebox  |   File "zmq/backend/cython/checkrc.pxd", line 28, in zmq.backend.cython.checkrc._check_rc
jukebox  | zmq.error.ZMQError: Protocol not supported
jukebox  | 31.05.2022 06:28:52 -   31:daemon.py          - jb.daemon            - MainThread      - DEBUG    - Active Threads = [<_MainThread(MainThread, stopped 547871088656)>, <Thread(PlayerEventLoop, started daemon 547742544352)>, <Thread(Thread-1, started daemon 547734151648)>, <Thread(StartJingle, started daemon 547711771104)>, <MultiTimer(host.timer.cputemp, started daemon 547703378400)>]
```

## Software

### Base image and version

* Raspberry PI OS lite (bullseye, 64 bit), installed yesterday

### Branch / Release

* Branch: `future3/spotify`

```
On branch future3/spotify
Your branch is up to date with 'origin/future3/spotify'.
```

### Installscript

-

## Hardware

### RaspberryPi version

* Raspberry Pi 3 Model B Plus Rev 1.3

### Soundcard

* [Suptronics x400](http://www.suptronics.com/xseries/x400.html)

varac, 2022-05-31:
Related: #1575

Groovylein, 2022-06-02:
We had to delete this branch.

See https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1815#issuecomment-1144485647

closed 2022-06-02 by Groovylein