MiczFlor/RPi-Jukebox-RFID #1838: 🐛 | docker: zmq.error.ZMQError: Protocol not supported

varac, 2022-06-02:
## Bug

### What I did

I followed the [Phoniebox Development Runbook for Docker environments](https://rpi-jukebox-rfid.readthedocs.io/en/latest/developer/docker.html). I could build all docker images fine but when I start the containers with `docker compose -f docker/docker-compose.yml -f docker/docker-compose.linux.yml` up both the jukebox containter ends up in a restart loop.

### I expected this to happen

The jukebox container should run successfully.

### Further information that might help

```
jukebox  | Traceback (most recent call last):
jukebox  |   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/run_jukebox.py", line 67, in <module>
jukebox  |     main()
jukebox  |   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/run_jukebox.py", line 63, in main
jukebox  |     myjukebox.run()
jukebox  |   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/daemon.py", line 227, in run
jukebox  |     self.rpc_server = RpcServer()
jukebox  |   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/rpc/server.py", line 86, in __init__
jukebox  |     self.socket.bind(websocket_address)
jukebox  |   File "/usr/local/lib/python3.9/dist-packages/zmq/sugar/socket.py", line 226, in bind
jukebox  |     super().bind(addr)
jukebox  |   File "zmq/backend/cython/socket.pyx", line 562, in zmq.backend.cython.socket.Socket.bind
jukebox  |   File "zmq/backend/cython/checkrc.pxd", line 28, in zmq.backend.cython.checkrc._check_rc
jukebox  | zmq.error.ZMQError: Protocol not supported
```

## Software

### Base image and version

* Raspberry PI OS lite (bullseye, 64 bit)

### Branch / Release

* Git tag V3.2.0

## Hardware

Raspberry Pi 3 Model B Plus Rev 1.3

## Possible soluton

I found that  that this diff in `jukebox.Dockerfile` fixed it:

```
-RUN pip3 install pyzmq
+RUN pip install -vv --no-binary pyzmq pyzmq --install-option --zmq=/usr 
```

I could create a PR if this solution is accepted.

varac, 2022-06-02:
Related:  #1814

varac, 2022-06-03:
I just noticed that I get the same error following the [official installation docs](https://rpi-jukebox-rfid.readthedocs.io/en/latest/install.html) for future3.

s-martin, 2022-06-03:
As mentioned by @ChisSoc most likely because of 64bit OS.

abirke, 2023-05-14:
I am also still seeing this issue on Raspberry Pi Model 3B  V1.2, Raspbian GNU/Linux 11 (bullseye) 32-bit, Python 3.9.2, pip 23.1.2, and Commit ID bd7bb789a31d358b505c825c5c695e9c2fc1eace from the `future3/main`.

The above script seems to still fix the problem although the `pip install` command options have slightly changed.

```
pip uninstall pyzmq
apt-get install libzmq3-dev
pip install --no-binary=:all: pyzmq
```

s-martin, 2023-11-18:
Should be now fixed on `future3/develop`

https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2108#issuecomment-1809167778

can you check this?

closing for now. Feel free to post in #2108, if the issue still occurs.

closed 2023-11-18 by s-martin