MiczFlor/RPi-Jukebox-RFID #1839: 🐛 | AttributeError: module alsaaudio has no attribute "PCM_FORMAT_S24_3LE"

varac, 2022-06-03:
## Bug

### What I did

I installed the raspberry pi with current RPi OS bullseye image and ran the installer script

### What happened

This error shows up in the jukebox-daemon logs:

```
pi@phoniebox:~ $ cat RPi-Jukebox-RFID/shared/logs/errors.log
03.06.2022 08:54:57 -  593:plugs.py           - jb.plugin            - MainThread      - ERROR    - Failed to load package: components.jingle.alsawave
03.06.2022 08:54:57 -  615:plugs.py           - jb.plugin            - MainThread      - ERROR    - Ignoring failed package load for 'jingle.alsawave'
03.06.2022 08:54:57 -  616:plugs.py           - jb.plugin            - MainThread      - ERROR    - Reason: AttributeError: module 'alsaaudio' has no attribute 'PCM_FORMAT_S24_3LE'
03.06.2022 08:54:57 -  617:plugs.py           - jb.plugin            - MainThread      - ERROR    - Detailed reason:
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 612, in load_all_named
    load(package, load_as, prefix)
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 594, in load
    raise e
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 585, in load
    _PLUGINS[load_as].module = importlib.import_module(f'{package}', 'pkg')
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/components/jingle/alsawave/__init__.py", line 20, in <module>
    class AlsaWave:
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/components/jingle/alsawave/__init__.py", line 24, in AlsaWave
    3: alsaaudio.PCM_FORMAT_S24_3LE,
AttributeError: module 'alsaaudio' has no attribute 'PCM_FORMAT_S24_3LE'
```

### I expected this to happen

Logs should not contain this error, a sound should be played on jukebox startup.

### Further information that might help

[syslog.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/8830273/syslog.log)

## Software

### Base image and version

* Raspberry Pi OS stable lite (bullseye, 64bit)

### Branch / Release

* `Git state: 80710235 [2022-02-10] Release 3.2.0  (HEAD, tag: V3.2.0, origin/future3/main) [V3.2.0-dirty]`

### Installscript

I followed the [Installing Phoniebox future3](https://rpi-jukebox-rfid.readthedocs.io/en/latest/install.html#installing-phoniebox-future3) documentation:

    cd; bash <(wget -qO- https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/future3/develop/installation/install-jukebox.sh)

## Hardware

* Raspberry Pi 3 Model B Plus Rev 1.3
* [Suptronics x400](http://www.suptronics.com/xseries/x400.html) Soundcard
* MFRC522 RFID Reader

s-martin, 2022-06-03:
As mentioned by @ChisSoc most likely because of 64bit OS.

s-martin, 2023-12-11:
3.4 was just released, https://github.com/MiczFlor/RPi-Jukebox-RFID/releases/tag/v3.4.0

Could you check, if your problem was fixed?

s-martin, 2024-05-07:
The 64Bit issue is tracked in #2041, so closing this issue.

closed 2024-05-07 by s-martin