MiczFlor/RPi-Jukebox-RFID #1840: 🐛 | Several python modules missing after fresh install

varac, 2022-06-03:
## Bug

### What I did

I installed the raspberry pi with current RPi OS bullseye image and ran the installer script.

### What happened

Several python modules were missing, I needed to install them manually:

* `ModuleNotFoundError: No module named 'ruamel'`

    `sudo apt install python3-ruamel.yaml`

* `ModuleNotFoundError: No module named 'pulsectl'`

    `sudo pip install pulsectl`

* `from ruamel.yaml import YAML   ModuleNotFoundError: No module named 'ruamel'`

    `sudo apt install python3-evdev`

* `ModuleNotFoundError: No module named 'eyed3'`

    `sudo apt install python3-eyed3`

* `ModuleNotFoundError: No module named 'mpd'`

    `sudo apt install python3-mpd`

* `ModuleNotFoundError: No module named 'alsaaudio'`

    `sudo apt install python3-alsaaudio`

* [zmq.error.ZMQError: Protocol not supported](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1838)

### I expected this to happen

The installation should result in a working jukebox.

### Further information that might help

Syslog after installing the missing modules: [syslog.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/8830273/syslog.log)

## Software

### Base image and version

* Raspberry Pi OS stable lite (bullseye, 64bit)

### Branch / Release

* `Git state: 80710235 [2022-02-10] Release 3.2.0  (HEAD, tag: V3.2.0, origin/future3/main) [V3.2.0-dirty]`

### Installscript

I followed the [Installing Phoniebox future3](https://rpi-jukebox-rfid.readthedocs.io/en/latest/install.html#installing-phoniebox-future3) documentation:

    cd; bash <(wget -qO- https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/future3/develop/installation/install-jukebox.sh)

## Hardware

* Raspberry Pi 3 Model B Plus Rev 1.3
* [Suptronics x400](http://www.suptronics.com/xseries/x400.html) Soundcard
* MFRC522 RFID Reader

varac, 2022-06-03:
Related issues from the same installation: #1839, #1838

s-martin, 2022-06-03:
As mentioned by @ChisSoc most likely because of 64bit OS.

s-martin, 2023-12-11:
3.4 was just released, https://github.com/MiczFlor/RPi-Jukebox-RFID/releases/tag/v3.4.0

Could you check, if your problem was fixed?

Jane232, 2024-02-01:
I can confirm this bug is still occurring with 3.4.0, tested on a RPI 2

AlvinSchiller, 2024-02-01:
> I can confirm this bug is still occurring with 3.4.0, tested on a RPI 2 (armv7l)

Hi,
whats is your OS version?
When does error this occure? When the service is running or on manually calling scripts?
If your run the python scripts manually you need to activate the venv first (https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/develop/documentation/developers/pyhton.md)

Jane232, 2024-02-02:
Raspberry Pi Os Legacy Version 32-Bit
And it occurred while executing the scripts manually (without venv).
Therefore I think you might be right that it was just missing the venv, and is not related to the mentioned bug

Thank you

closed 2024-02-02 by pabera