MiczFlor/RPi-Jukebox-RFID #1841: 🐛 | Webapp doesn't show any audio files

varac, 2022-06-03:
## Bug

### What I did

I installed the raspberry pi with current RPi OS bullseye image and ran the installer script

### What happened

Webapp doesn't show any audio files, even thogh mpd imported them successfully (I could play audio with ncmpc).

### I expected this to happen

Imported audio files should show up in the webapp

### Further information that might help

[syslog.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/8830273/syslog.log)

## Software

### Base image and version

* Raspberry Pi OS stable lite (bullseye, 64bit)

### Branch / Release

* `Git state: 80710235 [2022-02-10] Release 3.2.0  (HEAD, tag: V3.2.0, origin/future3/main) [V3.2.0-dirty]`

### Installscript

I followed the [Installing Phoniebox future3](https://rpi-jukebox-rfid.readthedocs.io/en/latest/install.html#installing-phoniebox-future3) documentation:

    cd; bash <(wget -qO- https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/future3/develop/installation/install-jukebox.sh)

## Hardware

* Raspberry Pi 3 Model B Plus Rev 1.3
* [Suptronics x400](http://www.suptronics.com/xseries/x400.html) Soundcard
* MFRC522 RFID Reader

varac, 2022-06-03:
[image]

s-martin, 2022-06-03:
As mentioned by @ChisSoc most likely because of 64bit OS.

M4rt1nCh, 2022-06-05:
Having a fresh install on 32 bit. Although there are audio folders available, they are not shown / detected in the WebUI…

pabera, 2022-06-05:
Yes. 32-Bit is essential. But this continues to be a problem, a console screenshot of the browser would be helpful too.

varac, 2022-06-16:
> As mentioned by @ChisSoc most likely because of 64bit OS.

I reinstalled using RPi OS lite 32bit, and the issue was the same. Turns out I didn't see that [Firefox is not supported](https://rpi-jukebox-rfid.readthedocs.io/en/latest/issues.html#browsers).
I don't find any open issue for Firefox support. It's hard to believe that it's still not supported, it's currently the only project that I need chromium for, and I can't remember that I needed it in the last years for any web app that wouldn't work with Firefox.
Please support Firefox soon.

Also, please add a warning more prominently, i doubt many ppl will look at the `Known issues` section.

closed 2022-06-16 by varac