MiczFlor/RPi-Jukebox-RFID #1847: Neuftech USB RFID Reader doesn't get recognized

fab4x, 2022-06-05:
### Bug
Neuftech USB RFID Reader doesn't get recognized with the installer script. 
**I've also added the fix that worked for me. I hope that's okay.** 

### What I did
I've installed the latest RPi Buster Lite image and ran the installer script and at the end of the installation where I am able to choose one, the script ran into an error after I've picked the option Neuftech USB Reader. 

The error output said that the python package evdev is missing and that the user pi doesn't exist. 

### Fix
To fix the problem I've manually created the user pi (sudo adduser pi) on my system and instantly disabled the login (sudo passwd -l pi) to avoid security problems (even though it's just the music box for my daughter). Afterwards I installed the python package evdev with the command pip install evdev and started the installation script again and it ran through without any problems. 

### Source of the problem
I've checked the source code in the installation script (buster-install-default.sh) and found the issue in line 1228 (sudo chown pi:www-data "${jukebox_dir}"/scripts/deviceName.txt) - wouldn't it make sense to use the variable $USER or ask for a User on the system which really exists? It may also help to add the installation command for python evdev.

fab4x, 2022-06-05:
Fix added

closed 2022-06-05 by fab4x

s-martin, 2022-06-05:
It's the missing evdev package. The underlying issue was fixed by #1791, but is not yet part of master.

It's unrelated to the user pi.

vitospinelli, 2022-08-16:
I manually installed evdev but it still doesn't work

fab4x, 2022-08-16:
Did you check if the user pi exists? If it does, could you paste the exact error message please?

vitospinelli, 2022-08-16:
I just solved it. 

I manually installed RC522 following exactly the instructions at the link below.

https://github.com/pimylifeup/MFRC522-python

Then I reinstalled phoniebox with the one line command and it started working.

The error I had was this:
phoniebox git clone -q git://github.com/ondtryaso/pi-rc522.git /tmp/pip-install-5yjpx8ai/pi-rc522 failed with error code 128 in None

So it looks like the installation process was failing to install pi-rc522. A manual install and then redoing the installation did the trick.

s-martin, 2022-08-20:
Fixed by #1791, but not merged to master yet.