MiczFlor/RPi-Jukebox-RFID #1850: Firefox support

varac, 2022-06-16:
### Describe your problem 

I stumbled across this because on a fresh (future3) Phoniebox installation the [Webapp doesn't show any audio files](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1841). Then I found in the `Known issues` section that [Firefox is not supported](https://rpi-jukebox-rfid.readthedocs.io/en/latest/issues.html#browsers).

* Phoniebox setup fresh using today's Rpi OS lite 32bit (bullseye)
* Installation following the [Installing Phoniebox future3](https://rpi-jukebox-rfid.readthedocs.io/en/latest/install.html) instructions

Loule-Corro, 2023-01-28:
i can confirm too

s-martin, 2023-05-01:
As per Known Issues section in the future3 documentation:
_The Web UI will not work with Firefox, due to an issue with websockets and pyzmq. Please use a different browser for now._

This issue won’t be fixed for now, please use a different browser.

s-martin, 2023-12-08:
Fixed in #2135