MiczFlor/RPi-Jukebox-RFID #1851: Wrong playlist ordering on future3

varac, 2022-06-17:
### Describe your problem 

I noticed that the playlist order is alphabetical, not based on the id3 tag track numbers as it should.
My library is organized with this pattern: `Album Artist/Album/Artist - Album - Track nr - Title`. 
For album from one artist this is not a problem.
However, for sompilations or soundtracks the order is messed up.
I.e. for this [Sing 2 Soundtrack](https://musicbrainz.org/release/1ad0d21f-dc36-4288-b713-0e830a7a8f30)  the files are named like this:

```
❯ ls
'Adam Buxton - Sing 2 (Original Motion Picture Soundtrack) - 16 - Tippy Toes (with Fancy Feelings & DSCOSTU) - BONUS TRACK.mp3'
'Bomba Estéreo - Sing 2 (Original Motion Picture Soundtrack) - 10 - Soy Yo - Sing 2 Mix.mp3'
'Halsey - Sing 2 (Original Motion Picture Soundtrack) - 12 - Could Have Been Me.mp3'
'Keke Palmer - Sing 2 (Original Motion Picture Soundtrack) - 17 - Christmas (Baby Please Come Home) (feat. Scarlett Johansson, Taron Egerton, Reese Witherspoon & Tori Kelly).mp3'
'Kiana Ledé - Sing 2 (Original Motion Picture Soundtrack) - 03 - Can'\''t Feel My Face.mp3'
'Reese Witherspoon - Sing 2 (Original Motion Picture Soundtrack) - 14 - Break Free (with Nick Kroll).mp3'
'Sam i - Sing 2 (Original Motion Picture Soundtrack) - 08 - Suéltate (with Sam i & Jarina De Marco feat. Anitta, BIA) - From Sing 2.mp3'
'Scarlett Johansson - Sing 2 (Original Motion Picture Soundtrack) - 04 - Heads Will Roll.mp3'
'Scarlett Johansson - Sing 2 (Original Motion Picture Soundtrack) - 09 - Stuck In A Moment You Can'\''t Get Out Of.mp3'
'Scarlett Johansson - Sing 2 (Original Motion Picture Soundtrack) - 15 - I Still Haven’t Found What I’m Looking For (with Bono).mp3'
'Sing 2 Cast - Sing 2 (Original Motion Picture Soundtrack) - 05 - Sing 2 Audition Medley.mp3'
'Taron Egerton - Sing 2 (Original Motion Picture Soundtrack) - 07 - There'\''s Nothing Holdin'\'' Me Back (with Tori Kelly).mp3'
'Taron Egerton - Sing 2 (Original Motion Picture Soundtrack) - 11 - A Sky Full of Stars.mp3'
'Tori Kelly - Sing 2 (Original Motion Picture Soundtrack) - 02 - Let'\''s Go Crazy (with Taron Egerton, Reese Witherspoon, Nick Kroll).mp3'
'Tori Kelly - Sing 2 (Original Motion Picture Soundtrack) - 06 - Where the Streets Have No Name (with Taron Egerton, Scarlett Johansson, Reese Witherspoon, Nick Kroll).mp3'
'Tori Kelly - Sing 2 (Original Motion Picture Soundtrack) - 13 - I Say A Little Prayer (with Pharrell Williams).mp3'
'U2 - Sing 2 (Original Motion Picture Soundtrack) - 01 - Your Song Saved My Life - From Sing 2.mp3'
```

And the resulting playlist is sorted alphabetically, not in track order:

```
pi@phoniebox:~ $ mpc playlist
Adam Buxton, Fancy Feelings, DSCOSTU - Tippy Toes (with Fancy Feelings & DSCOSTU) - BONUS TRACK
Bomba Estéreo - Soy Yo - Sing 2 Mix
Halsey - Could Have Been Me
Keke Palmer, Scarlett Johansson, Taron Egerton, Reese Witherspoon, Tori Kelly - Christmas (Baby Please Come Home) (feat. Scarlett Johansson, Taron Egerton, Reese Witherspoon & Tori Kelly)
Kiana Ledé - Can't Feel My Face
Reese Witherspoon, Nick Kroll - Break Free (with Nick Kroll)
Sam i, Jarina De Marco, Anitta, BIA - Suéltate (with Sam i & Jarina De Marco feat. Anitta, BIA) - From Sing 2
Scarlett Johansson - Heads Will Roll
Scarlett Johansson - Stuck In A Moment You Can't Get Out Of
Scarlett Johansson, Bono - I Still Haven’t Found What I’m Looking For (with Bono)
Sing 2 Cast - Sing 2 Audition Medley
Taron Egerton, Tori Kelly - There's Nothing Holdin' Me Back (with Tori Kelly)
Taron Egerton - A Sky Full of Stars
Tori Kelly, Taron Egerton, Reese Witherspoon, Nick Kroll - Let's Go Crazy (with Taron Egerton, Reese Witherspoon, Nick Kroll)
Tori Kelly, Taron Egerton, Scarlett Johansson, Reese Witherspoon, Nick Kroll - Where the Streets Have No Name (with Taron Egerton, Scarlett Johansson, Reese Witherspoon, Nick Kroll)
Tori Kelly, Pharrell Williams - I Say A Little Prayer (with Pharrell Williams)
U2 - Your Song Saved My Life - From Sing 2
```

The files have the right id3 tag track number, i.e. this one is track 16, but is played first due to the alphabetical order:

```
❯ eyeD3 Adam\ Buxton\ -\ Sing\ 2\ \(Original\ Motion\ Picture\ Soundtrack\)\ -\ 16\ -\ Tippy\ Toes\ \(with\ Fancy\ Feelings\ \&\ DSCOSTU\)\ -\ BONUS\ TRACK.mp3
eyed3.plugins:WARNING: Plugin '('lastfm.py', '/usr/lib/python3/dist-packages/eyed3/plugins')' requires packages that are not installed: No module named 'pylast'
... - Tippy Toes (with Fancy Feelings & DSCOSTU) - BONUS TRACK.mp3              [ 5.03 MB ]
--------------------------------------------------------------------------------------------
Time: 02:09	MPEG1, Layer III	[ 320 kb/s @ 44100 Hz - Joint stereo ]
--------------------------------------------------------------------------------------------
ID3 v2.4:
title: Tippy Toes (with Fancy Feelings & DSCOSTU) - BONUS TRACK
artist: Adam Buxton, Fancy Feelings, DSCOSTU
album: Sing 2 (Original Motion Picture Soundtrack)
album artist: Various Artists
release date: 2021-12-17
track: 16		
disc: 1
Publisher/label: Sing 2 Soundtrack Album
FRONT_COVER Image: [Size: 136810 bytes] [Type: image/jpeg]
Description: cover
```

Is there any way to change this behaviour ?

#### What's your hardware set up? 

* Phoniebox setup fresh using today's Rpi OS lite 32bit (bullseye)
* Installation following the [Installing Phoniebox future3](https://rpi-jukebox-rfid.readthedocs.io/en/latest/install.html) instructions

varac, 2022-06-17:
I mean yes, I could re-organize my whole library because a pattern like `Album Artist/Album/Track nr - Title`instead of `Album Artist/Album/Artist - Album - Track nr - Title` would be a work around, but that's a lot of work, a I guess I'm not the only affected person by this.

pabera, 2022-08-23:
Interesting use case. this should be doable. I need to check how this could be done.