MiczFlor/RPi-Jukebox-RFID #1853: 🐛 | Phoniebox-RFID-Modul_RC522 Code 128

Borlizgaron, 2022-06-28:
## RC522-Modul doesn't work

### What I did
Hello!
I have successfully installed my Phoniebox and everything is running, except for the RFID reader RC522. It shows as disabled in the web interface under settings. It can be activated, but then does not respond. When I update the web interface, the reader is disabled again.

### What happened

I used this https://github.com/MiczFlor/RP...ponents/rfid-reader/RC522 and startet the Code:

`sudo sh RPi-Jukebox-RFID/components/rfid-reader/RC522/setup_rc522.sh`

The following error message:

`Command "git clone -q git://github.com/ondryaso/pi-rc522.git /tmp/pip-install-attrnxz5/pi-rc522" failed with error code 128 in None`

Can anyone help me? Is something wrong with git? Is something out of date with python?

## Software

### Buster light

s-martin, 2022-07-02:
Issue was fixed in https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1791, but is not yet on master.

s-martin, 2022-07-02:
Duplicate of #1721

vitospinelli, 2022-08-16:
Manually installing RC522 following exactly these instructions solved the issue for me:

https://github.com/pimylifeup/MFRC522-python

s-martin, 2022-08-20:
Fixed by #1791, but not merged to master yet.

closed 2022-08-20 by s-martin