MiczFlor/RPi-Jukebox-RFID #1867: Cover art problems

kris711, 2022-08-02:
Hi,

maybe someone knows how to search all logs seems to be not caring about....

the only way i see cover arts (cover.jpg existis in each folder) in web app is to start something via Lists-Tab and then using home Tab. 
Only in this order the cover is shown. As soon i move away from home tab (e.g. to preferences and back) there is no way to show it.

Reasons? I have no clue.

handfrog, 2022-08-11:
I cannot see any cover, no where in the webinterface.
Every folder contains a cover.jpg.

I already tried:
- File ownership to pi:www-data
- Rights even to 777
- Different sizes and depth of jpg
- including covers in id3v2 tags

I am also using future3/main and tried different browsers on different OSes.

If you need futher information just ask, my Linux is good, but my python is a bit "rusty" :smile:

s-martin, 2022-08-20:
@pabera @ChisSoc

pabera, 2022-08-23:
> Every folder contains a cover.jpg.

We are reading the image directly from the file, so no need of a `cover.jpg`

We only show the covers in the player for now. We have not yet built it into the list views.

kris711, 2022-08-23:
Ok but the player behave as you describe only if I come from list view.
(cover.jpg is shown if there is no mp3 embedded cover in tags)
Switching from preference tab to player does not show anything.

s-martin, 2023-12-20:
I use recent `future3/develop`, but still have the problem (with Chrome or Firefox).

The mp3s have the cover embedded (added with mp3tag and checked with vlc), but Phoniebox doesn't show anything.

s-martin, 2023-12-22:
Fixed just now on branch `future3/develop`.

be aware that you need to update the settings in `yukebox.yml`.
See 
[image]

closed 2023-12-22 by s-martin