MiczFlor/RPi-Jukebox-RFID #1872: default installation missing package "evdev"🐛 | BUG SUMMARY

MarkusF1895, 2022-08-28:
## Bug
The https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/docs/INSTALL-COMPLETE-GUIDE.md#6-install-phoniebox-software runs into an error because package evdev is missing! 
### What I did
manuelly install evdev

### What happened

### I expected this to happen

### Further information that might help

## Software

### Base image and version

### Branch / Release

### Installscript

## Hardware

### RaspberryPi version

### RFID Reader

### Soundcard

### Other notable hardware

s-martin, 2022-08-28:
Fixed in https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1791, but not yet merged to master.

MarkusF1895, 2022-08-28:
Thanks for the hint. Please close this issue. I tried again, as described in #1791. see here: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1874

Incidentally, I think it makes sense to either mention the installation of the requirements in the installation instructions or to include them in the "one-line command."

closed 2022-08-28 by MarkusF1895