MiczFlor/RPi-Jukebox-RFID #1874: 🐛 | RFID Reader does not work

MarkusF1895, 2022-08-28:
## Bug
The RFID Reader does work, but no information flows after a scan into the UI. 

### What I did
- run `cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh`
- https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#link-a-new-card-to-a-folder-or-stream-in-the-web-app tried, but the "last used chip" file is not created! 
- change to git+https://github.com/lthiery/SPI-Py.git#egg=spi-py in requirements.txt
- run `sudo python3 -m pip install --upgrade --force-reinstall -r requirements.txt`
- run `python3 RegisterDevice.py`, but an error appeared. 

### What happened
`pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python3 RegisterDevice.py`
`Choose the reader from list`
`0 Sycreader RFID Technology Co., Ltd SYC ID&IC USB Reader`
`Device Number: 0`
`Exception ignored in: <function BaseEventLoop.__del__ at 0xb61d3300>`
`Traceback (most recent call last):`
`  File "/usr/lib/python3.7/asyncio/base_events.py", line 622, in __del__`
`    warnings.warn(f"unclosed event loop {self!r}", ResourceWarning,`
`  File "/usr/lib/python3.7/asyncio/base_events.py", line 390, in __repr__`
`    f'<{self.__class__.__name__} running={self.is_running()} '`
`  File "/usr/lib/python3.7/asyncio/base_events.py", line 1794, in get_debug`
`    return self._debug`
`AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'`

### I expected this to happen
If I put a chip on the reader, the "last used chip" file should be generated. 

### Further information that might help

## Software

### Base image and version
`2022-08-28`

### Branch / Release
`'origin/master'` 

### Installscript
using  `cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh`

## Hardware

### RaspberryPi version
Raspberry Pi Bundle: Raspberry Pi 4 4GB + Gehäuse + Netzteil + Kabel 
https://www.amazon.de/gp/product/B084NW35M5/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

### RFID Reader
Neuftech USB RFID Reader ID Kartenlesegerät Kartenleser Kontaktlos Card Reader für EM4100
https://www.amazon.de/gp/product/B018OYOR3E/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

### Soundcard (not relevant)
CSL - Externe USB Soundkarte mit Virtual Surround Sound, Plug and Play
https://www.amazon.de/gp/product/B00C7LXUDY/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

### Other notable hardware
nope

MarkusF1895, 2022-09-05:
@MiczFlor, please tell me the possible mistakes in the code. I want to fix the bug but would be reluctant to understand the entire code.

s-martin, 2022-09-05:
Was fixed in #1791, but this is not part of master yet.

MarkusF1895, 2022-09-05:
> Was fixed in #1791, but this is not part of master yet.

I have tried. No remedy!

tinoton2022, 2022-09-14:
Installing evdev via $sudo pip3 install evdev fixed the issue for me with exactly the same hardware setup

MarkusF1895, 2022-09-19:
> Installing evdev via $sudo pip3 install evdev fixed the issue for me with exactly the same hardware setup

First of all. Thanks for your answer. 
Unfortunately, it doesn't seem to have been the problem for me. See the following: 
`pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ sudo pip3 install evdev
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: evdev in /usr/local/lib/python3.7/dist-packages (1.6.0)`

I went through all the steps again and noticed that it warns me when installing the requirements!
If I install it as shown: `pytest-pythonpath 0.7.4 has requirement pytest<7,>=2.5.2, but you'll have pytest 7.1.3 which is incompatible.`
If I install it with `pytest==2.5.2`: pytest-cov 3.0.0 has requirement pytest>=4.6, but you'll have pytest 2.5.2 which is incompatible.
If I install it with `pytest==5`: No warning but still the same error in the result. 

@tinoton2022 could you use `pip3 freeze > requirements123.txt` and post the result here?

Albusoline, 2022-11-10:
Hi.
After installing evdev via
python3 -m pip install evdev
and
sudo python3 -m pip install evdev,

then running 
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python3 RegisterDevice.py
Choose the reader from list
0 Sycreader RFID Technology Co., Ltd SYC ID&IC USB Reader
Device Number: 0

I received the following output:

Exception ignored in: <function BaseEventLoop.__del__ at 0x761855d0>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/base_events.py", line 622, in __del__
    warnings.warn(f"unclosed event loop {self!r}", ResourceWarning,
  File "/usr/lib/python3.7/asyncio/base_events.py", line 390, in __repr__
    f'<{self.__class__.__name__} running={self.is_running()} '
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1794, in get_debug
    return self._debug
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'

Any ideas?

MarkusF1895, 2022-11-14:
> Hi. After installing evdev via python3 -m pip install evdev and sudo python3 -m pip install evdev,
>
> then running pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python3 RegisterDevice.py Choose the [...]

Still have the same error but haven't had time to look closer lately. Thanks for the tip, though. At least I'm not the only one with this problem. If you open a new issue, feel free to refer to this ticket. Maybe then we'll finally get help.

swoop124, 2022-11-29:
Hello,

i have the same errors. With the stes described here I got it working so far.
But when I change the settings of `General RFID player control` to `Placing the card starts player, removal stops it` then it seems to work for about one or two seconds. After that the playback pause.
```
Nov 23 12:02:58 raspibox daemon_rfid_reader.py[7353]: 2022-11-23 12:02:58,954 - root - INFO - Trigger Play Cardid=0003611630
Nov 23 12:03:00 raspibox daemon_rfid_reader.py[7353]: 2022-11-23 12:03:00,613 - root - INFO - No RFID Signal detected.
Nov 23 12:03:00 raspibox daemon_rfid_reader.py[7353]: 2022-11-23 12:03:00,614 - root - INFO - Trigger Pause Force
Nov 23 12:03:00 raspibox daemon_rfid_reader.py[7353]: Some Audiobook in a Folder
Nov 23 12:03:00 raspibox daemon_rfid_reader.py[7353]: [paused] #1/6 10:45/47:52 (22%)
Nov 23 12:03:00 raspibox daemon_rfid_reader.py[7353]: volume: 45% repeat: off random: off single: off consume: off
```

When the card/RFID-Chip is placed for a longer time this is happening:
```
Nov 23 14:46:14 raspibox daemon_rfid_reader.py[553]: 2022-11-23 14:46:14,282 - root - INFO - Trigger Play Cardid=0003451567
Nov 23 14:46:15 raspibox daemon_rfid_reader.py[553]: 2022-11-23 14:46:15,925 - root - INFO - No RFID Signal detected.
Nov 23 14:46:15 raspibox daemon_rfid_reader.py[553]: 2022-11-23 14:46:15,926 - root - INFO - Trigger Pause Force
Nov 23 14:46:16 raspibox daemon_rfid_reader.py[553]: Lionel Richie - Just For You
Nov 23 14:46:16 raspibox daemon_rfid_reader.py[553]: [paused]  #1/1   0:23/0:00 (0%)
Nov 23 14:46:16 raspibox daemon_rfid_reader.py[553]: volume: 30%   repeat: off   random: off   single: off   consume: off
Nov 23 14:46:43 raspibox daemon_rfid_reader.py[553]: 2022-11-23 14:46:43,744 - root - INFO - Trigger Play Cardid=0003451567
Nov 23 14:46:45 raspibox daemon_rfid_reader.py[553]: 2022-11-23 14:46:45,363 - root - INFO - No RFID Signal detected.
Nov 23 14:46:45 raspibox daemon_rfid_reader.py[553]: 2022-11-23 14:46:45,364 - root - INFO - Trigger Pause Force
Nov 23 14:46:45 raspibox daemon_rfid_reader.py[553]: Lionel Richie - Just For You
Nov 23 14:46:45 raspibox daemon_rfid_reader.py[553]: [paused]  #1/1   0:25/0:00 (0%)
Nov 23 14:46:45 raspibox daemon_rfid_reader.py[553]: volume: 30%   repeat: off   random: off   single: off   consume: off
Nov 23 14:46:58 raspibox daemon_rfid_reader.py[553]: 2022-11-23 14:46:58,694 - root - INFO - Trigger Play Cardid=0003451567
Nov 23 14:47:00 raspibox daemon_rfid_reader.py[553]: 2022-11-23 14:47:00,344 - root - INFO - No RFID Signal detected.
Nov 23 14:47:00 raspibox daemon_rfid_reader.py[553]: 2022-11-23 14:47:00,345 - root - INFO - Trigger Pause Force
Nov 23 14:47:00 raspibox daemon_rfid_reader.py[553]: Lionel Richie - Just For You
Nov 23 14:47:00 raspibox daemon_rfid_reader.py[553]: [paused]  #1/1   0:27/0:00 (0%)
Nov 23 14:47:00 raspibox daemon_rfid_reader.py[553]: volume: 30%   repeat: off   random: off   single: off   consume: off
```

MarkusF1895, 2022-12-16:
> I started off from scratch with a new SD-Card. This time I used the one-line installer script for the developer-version. I didn't had to install evdev by myself, all was already done by the script. However I still got the error mentioned above.
> 
> Funny enough, the RFID-Card reader worked well in Phoniebox... didn't expect that.
> 
> Therefore, I found a solution, even if it doesn't look nice during the installation...

Same for me, thanks for the tipp with the developer-version!

MarkusF1895, 2022-12-16:
problem solved

closed 2022-12-16 by MarkusF1895

topas-rec, 2022-12-25:
The script where the issue arises `~/RPi-Jukebox-RFID/scripts/RegisterDevice.py` asks the user for the correct device and then writes the name of that device into a file called `deviceName.txt`.

After getting the same issue as the original poster oof this issue, I looked into that file and the device name was there.
Whatever is going wrong in python (looks like there is something wrong in the destructor of `BaseEventLoop`: `BaseEventLoop.__del__`) it does not break the functionality that this script should have.
The reader also works after installation through "stable".

We could hunt the cause for this, but it seems solved in the development branch already. So that seems an unnecessary effort.
Alternatively we could catch this exception and in the exception handling we do check if there is the correct string in `deviceName.txt`. If the name is correct then the exception handling is done and the users are not bothered with this.

topas-rec, 2022-12-28:
> Alternatively we could catch this exception

Doesn't work. It is not caught as an `AttributeError`.