MiczFlor/RPi-Jukebox-RFID #1878: Audio cut off

HereIsFrank, 2022-09-01:
Hi everybody, 

I found a strange behaviour, at least in my Phoniebox installation. Everytime the last track of a playlist has about 2 seconds cut off. It only affects the last track of a list, everything track before the last track is played completely. I have already searched the various forums - without result. Not even this error has someone described. 

The error occurs both when using the soundcard of the RasPi4 and with a HifiBerry.

The behavior is also reproducible on the command line when I call mpc manually. Does anyone know this and does anyone have a solution?

//Frank

Phoniebox version: 2.3 - 24258901 - master
Raspberry 4 with Raspian Buster (Linux 5.10.103-v7l+ armv7l)

s-martin, 2023-03-14:
I would assume, if this also happens when using mpc directly that this is an issue of mpc.

s-martin, 2023-12-05:
Closing as it seems to be an external issue of mpc and no interaction for a couple of months.
Feel free to reopen the issue, if the problem still occurs.

closed 2023-12-05 by s-martin