MiczFlor/RPi-Jukebox-RFID #1886: RPI Zero 2 W with i2s Audio (No sound) on future3

tactmaster, 2022-09-15:
### Describe your problem 

Sounds working great before install

No sounds after install 

#### What's your hardware set up? 

Adafruit I2S Audio Bonnet 
RPI Zero 2 W 
https://www.adafruit.com/product/4037
Used this to set it up 
curl -sS https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/i2samp.sh | bash

### Solution

Not sure where this should be a pull request or documenation update

load-module module-alsa-sink  added to .config/pulse/default.pa

it may also need the sink to be set in jukebox.yaml

However I am still playing..

#### If possible, try to attach logs from ... (paths from RPi)

[notwokring32.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/9577817/notwokring32.txt)
[alsanoworking.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/9577819/alsanoworking.txt)

s-martin, 2023-12-11:
3.4 was just released, https://github.com/MiczFlor/RPi-Jukebox-RFID/releases/tag/v3.4.0

Could you check, if your problem was fixed?

damaev, 2023-12-31:
I ran into some seemingly problems with my I2S PCM5102A DAC after upgrading the OS from jessy or stretch to the current bookworm. 

My new installation method is:
To make the following changes to `/boot/config.txt`
```
dtparam=i2s=on
#dtparam=audio=on
dtoverlay=hifiberry-dac
```

After a `reboot` the card should be detected. *My following issue was that in the old days you needed to configure alsa with a softvol control. It took me a lot of time to not make it work and to.finally realise that pulseaudio is now directly taking care of it.*
Changes to `/etc/.asound` or similar are not necessary/working.

Hopefully this helps. 
Best wishes

s-martin, 2023-12-31:
> I ran into some seemingly problems with my I2S PCM5102A DAC after upgrading the OS from jessy or stretch to the current bookworm. 
> 
> My new installation method is:
> To make the following changes to `/boot/config.txt`
> ```
> dtparam=i2s=on
> #dtparam=audio=on
> dtoverlay=hifiberry-dac
> dtoverlay=vc4-kms-v3d,noaudio
> ```
> 
> After a `reboot` the card should be detected. *My following issue was that in the old days you needed to configure alsa with a softvol control. It took me a lot of time to not make it work and to.finally realise that pulseaudio is now directly taking care of it.*
> Changes to `/etc/asound.conf` or similar are not necessary/working.
> 
> Hopefully this helps. 
> Best wishes 
> 

Looks like bookworm also changed the location of config.txt:

https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2169#issuecomment-1872617699

damaev, 2023-12-31:
> > I ran into some seemingly problems with my I2S PCM5102A DAC after upgrading the OS from jessy or stretch to the current bookworm.
> > My new installation method is:
> > To make the following changes to `/boot/config.txt`
> > ```
> > dtparam=i2s=on
> > #dtparam=audio=on
> > dtoverlay=hifiberry-dac
> > dtoverlay=vc4-kms-v3d,noaudio
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > After a `reboot` the card should be detected. _My following issue was that in the old days you needed to configure alsa with a softvol control. It took me a lot of time to not make it work and to.finally realise that pulseaudio is now directly taking care of it._
> > Changes to `/etc/asound.conf` or similar are not necessary/working.
> > Hopefully this helps.
> > Best wishes
> 
> Looks like bookworm also changed the location of config.txt:
> 
> [#2169 (comment)](https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2169#issuecomment-1872617699)

Yes, I was mistaken. I'm currently running bullseye...
I edited my post accordingly.

pabera, 2024-01-02:
Closing this. Feel free to reopen if not solved. Also have a look at #2169.

closed 2024-01-02 by pabera