MiczFlor/RPi-Jukebox-RFID #1890: Installation of default values 2.3 under latest Raspberry Pi OS

DonPavlov, 2022-09-24:
Multiple issues found.
1. Dependency git is not installed, missing an apt-install git -y
2. PHP 7.3 no longer exists. better to just use php-common and php-cgi
E: Package 'php7.3-common' has no installation candidate
E: Unable to locate package php7.3-cgi
E: Couldn't find any package by glob 'php7.3-cgi'
E: Couldn't find any package by regex 'php7.3-cgi'
3. SPI-py installation fails. Installing it manually to circumvent the issue.
  ERROR: Command errored out with exit status 128:
   command: git clone -q git://github.com/lthiery/SPI-Py.git /tmp/pip-install-83i9or7j/spi-py_9104836be2c44344b31184f0449f6827
       cwd: None

4. Selecting the USB Reader has the following issue:
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py", line 4, in <module>
    from Reader import get_devices
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 23, in <module>
    from evdev import InputDevice, ecodes, list_devices
ModuleNotFoundError: No module named 'evdev'
chown: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory
chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory

This results in missing python3-evdev apt package

5. After running the RegisterDevice.py script. It just fails to stop. I don't know why and i have to quit the script now.( Only fails if evdev is already installed)

s-martin, 2022-10-26:
PHP version fixed by #1849, but not in master yet.

s-martin, 2022-10-26:
3, 4 and 5: Was fixed in https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1791, but this is not part of master yet.

s-martin, 2022-10-26:
1: not sure, if you can checkout the script without git installed?

s-martin, 2022-12-18:
V2.4.0 was released today - fixed.

closed 2022-12-18 by s-martin