MiczFlor/RPi-Jukebox-RFID #1896: Banana Pi BPI-Zero?

conkydong, 2022-10-25:
Does this work with a Banana Pi BPI-Zero?

conkydong, 2022-10-27:
Help?

micxer, 2022-11-07:
Taking into account that regular Pi's are almost impossible to buy this is a valid question. Is there anything that would make it impossible to use a Banana pi for a Phoniebox?

bjentsch, 2022-11-24:
I'm a bit late to the party, but I managed to get Phoniebox up and running on a [Pine A64](https://www.pine64.org/devices/single-board-computers/pine-a64/) which [costs about 40 Euros](https://www.amazon.de/dp/B01LWUNWNR/) at the time of writing, of course excluding all accessories like WiFi, rotary encoders, power supplies, buttons, NFC reader...

I'll just dump my experiences here and create some tickets here for the bugs I found here in the repo after I have finished scanning the open issues for them 😁 

Here we go, this is what you need for Pine64!

## Step 0 - preparation (partly because of differences between RPi and Pine64 (or Banana Pi?)

1. Prepare a user account called "pi", make them administrator

```
$ useradd pi
$ passwd pi # default on raspbian is "raspberry"
$ sudo usermod -aG sudo pi
$ groups pi # should say pi sudo
```

2. Install some software that will be of use later on

```
$ sudo apt install libasound2-dev lighttpd-mod-deflate
```

## Step 1 - download buster-install-default.sh ("One line install command" doesn't work out of the box as expected)

Download https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh

**Edit the file** as follows:

[buster-install-default.sh **line 693**](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/installscripts/buster-install-default.sh#L693): change from
```
local php_ini="/etc/php/7.3/cgi/php.ini"
```
to
```
local php_ini="/etc/php/$(ls -1 /etc/php)/cgi/php.ini"
```

Why?

The `php@7.3` package isn't available anymore in major distros, most offer 7.4 as their oldest supported version.

This is fixed in #1849 but not merged yet; it's plain broken in `master`.

Make another amendment to the install script:

[buster-install-default.sh **line 841**](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/installscripts/buster-install-default.sh#L841): add a line to **pause the install script there** so you have the chance to edit the cloned repository **before python dependencies are installed**.

Why?

Because the python `requirements.txt` contains a git repository on GitHub that is referenced with the `git://` protocol, which isn't supported by GitHub any more apparently.

The line to add:

```
read -rp "MODIFY requirements.txt in the cloned directory now! After finishing, hit ENTER to proceed to the next step." INPUT
```

What does it do? It stops the install script, allowing you to make the changes from #1791 inside `requirements.txt`: https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1791/files#diff-4d7c51b1efe9043e44439a949dfd92e5827321b34082903477fd04876edb7552

**Open a second Terminal / SSH connection to your soon-to-be phoniebox.**

Then just change `git://` to `https://` in [line 8](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/requirements.txt#L8).

## Step 2 - run `buster-install-default.sh`

Go through the installation configuring the phoniebox as you wish. Complete it, ignoring the errors. If you want, you can restart now.

A few of the things that will fail (because of hardware differences to RPi, some config fails in the config templates, missing permissions, and general Linux sissyness):

[... 177 lines cut ...]
```
$ sudo amixer -c 2 scontrols
# Simple mixer control 'Speaker',0 # <-- this is it on the Xmod!
# Simple mixer control 'PCM Capture Source',0
# Simple mixer control 'Line',0
# Simple mixer control 'Mic',0
# Simple mixer control 'Auto Gain Control',0
# Simple mixer control 'Up Down',0
```

So, `Speaker` it is for the `RPi-Jukebox-RFID/settings/Audio_iFace_Name`:

```
$ echo Speaker > RPi-Jukebox-RFID/settings/Audio_iFace_Name
```

## Step 4 - REBOOT

I mean it!

## Step 5 - actually use this thing

This is where I'm at. I'm in the process of selecting and purchasing a case, buttons, rotary dials for the volume, cabling...

I hope these steps help you manage building your own phoniebox on non-RPi hardware. If you have any questions left, or the instructions are unclear, let me know.

bjentsch, 2022-12-08:
**An update for you guys trying to get this stuff to work on any other system than the original RPi.**

I got everything up and running in a pretty satisfying way. Here's what works:

- Network access (lighttpd, webapp)
- Audio output (mpd Server, Hardware: Creative Xmod)
- Speakers (ripped apart a pair of old Altec Lansing Desktop Speakers)
- Audio amplifier using 5V output from Pine A64 (https://www.amazon.de/gp/product/B07KPTNPLJ/)
- Buttons via USB (!) using this arcade encoder board: https://www.amazon.de/gp/product/B01LZBFMWG/
- Status LED via GPIO

The system is consuming about 4W of energy during playback.

### Having to use USB buttons for my setup

I tried to get GPIO buttons to work for one or two eternities. I read metric tons of code to understand if it was an issue in my configuration or in the phoniebox code. Turns out it was neither.

After reading the RPi.GPIO port's [README](https://github.com/swkim01/RPi.GPIO-PineA64/blob/master/README.md) for the 374th time, I stumbled across this remark in the "features" list:

> GPIO interrupts(callbacks when events occur on input gpios) Not Implemented yet!!!

Yeah, well. That's exactly what the phoniebox code is using for GPIO button control. So, the conclusion in this case is:

⚠ **You can't use GPIO buttons or rotary encoders with your phoniebox with Pine A64!** ⚠

💡 I'm pretty sure that this doesn't apply to Orange/Banana Pi, as their GPIO implementations don't mention any non-implemented functionality.

This still leaves me with output functionality for GPIO - meaning I have to connect my physical buttons to the above-mentioned USB encoder board while still being able to connect the Status LED to the GPIO pins.

### Tackling final permission problems

Things that got in the way in the meantime:

- GPIO service permissions and
- USB buttons service permissions

...and for the same reason, namely running as `pi:pi` user. If you want to change this, head for the config file:

```
$ sudo nano /etc/systemd/system/phoniebox-buttons-usb-encoder.service
```
or
```
$ sudo nano /etc/systemd/system/phoniebox-gpio-control.service
```
and just delete these two lines:
```
User=pi
Group=pi
```
This will cause the services to run as root, eliminating permission problems.

## Conclusion

Whew, one hell of a ride. It took me about one week of net working time to get here, split up across about six weeks.

I'd lie if I said it was fun all the time, but it was definitely worth it. I got all the hardware for about the same price as a regular RPi 3B/4, maybe 20 or 30 Euros more. It was an awesome experience to get the phoniebox ecosystem up and running literally bit by bit.

And now it's time to go for the casing, which will be a tiny little wooden suitcase :-D