MiczFlor/RPi-Jukebox-RFID #1897: python3-evdev not installed - needs to be installed

LeDucCestMoin, 2022-10-25:
## Bug

### What I did

I installed the raspberry pi with above mentioned buster image and ran the installer script`

### What happened
ModuleNotFoundError: No module named 'evdev'

### I expected this to happen

No ModuleNotFoundError
### Further information that might help

## Software

### Base image and version

2022-09-22-raspbian-buster-lite.img

### Branch / Release
 
master
### Installscript

bust-install-default.sh

## Hardware

### RaspberryPi version

Raspberry Pi 3 Model B Rev 1.2

### RFID Reader
ffff:0035 Sycreader RFID Technology Co., Ltd SYC ID&IC USB Reader

### Soundcard
iqaudio-dacplus

### Other notable hardware
none

s-martin, 2022-10-26:
Was fixed in https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1791, but this is not part of master yet.

s-martin, 2022-12-18:
V2.4.0 was released today - fixed.

closed 2022-12-18 by s-martin