MiczFlor/RPi-Jukebox-RFID #1912: 🐛 | Could not register RFID cards with RC522

Lippsson, 2022-11-15:
## Bug
i installed 2021-12-02-raspios-buster-armhf-lite on a raspberry pi 3
i used the B. Developer install script for Buster
i use the PC522 reader
Hifiberry Miniamp as Soundcard

i could not register new cards with the Phoniebox software.
It works when i use the simple script from https://pimylifeup.com/raspberry-pi-rfid-rc522/
i also checked with: 

sudo systemctl status phoniebox-rfid-reader

● phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: loaded (/etc/systemd/system/phoniebox-rfid-reader.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2022-11-15 21:35:22 CET; 5min ago
 Main PID: 537 (python3)
    Tasks: 2 (limit: 871)
   CGroup: /system.slice/phoniebox-rfid-reader.service
           └─537 python3 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py

Nov 15 21:35:22 raspberrypi systemd[1]: Started Phoniebox RFID-Reader Service.
Nov 15 21:35:24 raspberrypi daemon_rfid_reader.py[537]: 2022-11-15 21:35:24,634 - root - INFO - Dir_PATH: /home/pi/RPi-Jukebox-RFID/scripts
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
lines 1-10/10 (END)

i also did the troubleshootings steps from the wiki side without any success.

Lippsson, 2022-11-15:
one comment: when i use the simple script from https://pimylifeup.com/raspberry-pi-rfid-rc522/ i get this:

/usr/local/lib/python3.7/dist-packages/mfrc522/MFRC522.py:151: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(pin_rst, GPIO.OUT)
Waiting for you to scan an RFID sticker/card
AUTH ERROR!!
AUTH ERROR(status2reg & 0x08) != 0
The ID for this card is: 177137037973
pi@raspberrypi:~ $ python Read.py
Waiting for you to scan an RFID sticker/card
AUTH ERROR(status2reg & 0x08) != 0
Error while reading!
Error while reading!
The ID for this card is: 584191750583
pi@raspberrypi:~ $

bjentsch, 2022-12-09:
What happens if you try to start the rfid service? 

`$ sudo systemctl start phoniebox-rfid-reader`

If it crashes on launch, what does this output:

`$ sudo journalctl -xeu phoniebox-rfid-reader`

Lippsson, 2023-01-14:
Hello! I tried with the 2.4 installation and now it works really nice. Thank you all for this update!

closed 2023-01-14 by Lippsson