MiczFlor/RPi-Jukebox-RFID #1915: 🚀 | phoniebox all in one image

erickeller, 2022-11-19:
## Feature Description

### What functionality would you like to see in your phoniebox?

It would be great to pre-bake a phoniebox image, so one could just write it on the sd card without having to go through installing all dependencies and compiling some parts.
This image could be produced automatically on tags using something like get-edi.io

Basically there could be a similar curses phoniebox-config to configure it on first boot / or bake it in on the first web-ui login e.g. home-assistant does.
If the image would using the mender.io mechanism one could also create A/B update of new versions released over the air without user interaction.

### How do you envision the feature to work from a users perspective?

```
dd if=Downloads/phoniebox_arch_x.y.z.img of=/dev/sdX bs=4M status=progress
sudo mount /dev/sdX1 /mnt/ && sudo touch /mnt/ssh && sudo umount /mnt
# over ssh configuration
ssh ...
phoniebox-config 
# or over browser configuration
firefox phoniebox.local
```

### Further information that might help

https://www.get-edi.io/
https://mender.io/

MoJo2600, 2023-09-16:
I'm currently working on a [Buildroot setup](https://github.com/MoJo2600/RPi-Jukebox-RFID-buildroot-package) to create this image. My main intention is to reduce the boot time significantly. But the result would be a prebuild image one can just write to the SD card.

But there are some limitations which I think makes your idea very hard to realize. Right now the configuration is done by the install script which does a lot of different system configurations based on user input. It would be necessary to switch this completely to a configuration based approach where the user can setup everything with a web ui or config file. Then there is the vast amount of hardware options that would have to be supported.

Maybe somebody can prove me wrong. I'd love to hear ideas.

pabera, 2023-09-17:
This has been brought up a few times in the past and it can help. There are a few things to be considered before exploring this route. The fact that the Phoniebox software uses an install script is a historical one but is also related to the dynamic set up. Various Pi's, Soundcards, GPIO, Bluetooth etc might be difficult to handle with just one image.

I like the idea of the config approach. We already have a config (they slightly look different between 2.x and future3). But all decisions taken during installation could be placed in such a config file. In this case, the installation could be done in a single run. This could than be used in a Buildroot set up.

s-martin, 2023-09-17:
My idea would be to offer 2-3 images (3.x, 2.x, Spotify? …) with a couple of hints, which one to choose and after the first start of the image you run a script for all the details?

MoJo2600, 2023-09-17:
Yes, I've seen it in V3 and I'm only working with this version. I was thinking the same @s-martin - there will be dedicated images for the different pis. Some USB devices can be supported out of the box.
My goal would be: Flash SD-Card, start the pi. A wifi hotspot will be created and there is a setup wizard to setup everything. No ssh or script execution by the user.

MoJo2600, 2023-09-18:
Is there another way besides github issues or the forum to get in touch with you developers? I have a few questions about the code that would not fit here.

AlvinSchiller, 2023-09-18:
> Is there another way besides github issues or the forum to get in touch with you developers? I have a few questions about the code that would not fit here.

That would be via Matrix: 
https://matrix.to/#/#phoniebox_community:gitter.im

The update of the badge is already in a pull request.