MiczFlor/RPi-Jukebox-RFID #1919: PN532 (v3 from Elechome) not working

conkydong, 2022-12-05:
Trying to get my PN532 v3 from Elechome to work.
nfc-list does work and is reading NFC tags but the RFID daemon isn't.

```
daemon_rfid_reader.py[31756]: Traceback (most recent call last):
daemon_rfid_reader.py[31756]:   File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 20, in <module>
 daemon_rfid_reader.py[31756]:     reader = Reader()
 daemon_rfid_reader.py[31756]:   File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 182, in __init__
 daemon_rfid_reader.py[31756]:     self.devs.append(Pn532Reader())
 daemon_rfid_reader.py[31756]:   File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 145, in __init__
 daemon_rfid_reader.py[31756]:     pn532 = Pn532_i2c()
 daemon_rfid_reader.py[31756]: NameError: name 'Pn532_i2c' is not defined
 systemd[1]: phoniebox-rfid-reader.service: Main process exited, code=exited, status=1/FAILURE
 systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
 systemd[1]: phoniebox-rfid-reader.service: Service RestartSec=100ms expired, scheduling restart.
 systemd[1]: phoniebox-rfid-reader.service: Scheduled restart job, restart counter is at 5.
 systemd[1]: Stopped Phoniebox RFID-Reader Service.
```

conkydong, 2022-12-06:
Solved. I just had a case error in one file (couldn't say which one anymore): Pn532_i2c() was written as PN532_i2c().

s-martin, 2022-12-18:
Can you close the issue, if it works?

closed 2022-12-19 by conkydong