MiczFlor/RPi-Jukebox-RFID #1922: 🐛 | BUG SUMMARY evdev not installed -> cannot register rfid reader

llasse, 2022-12-10:
## Bug

### What I did
After following install procedure (current raspbeery lite 32bit fresh install) and running the installation script installation seems to finish, but I cannot properly setup  the rfid reader, as this error occurs:

### What happened

 ```
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py", line 4, in <module>
    from Reader import get_devices
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 23, in <module>
    from evdev import InputDevice, ecodes, list_devices
ModuleNotFoundError: No module named 'evdev'
chown: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory
chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory

DONE. Let the sounds begin.
Find more information and documentation on the github account:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/
Reboot is needed to activate all settings
Would you like to reboot now? [Y/n
```
### I expected this to happen
proper finish of scripts/RegisterDevice.py 

### Further information that might help
 `find logfiles at [https://paste.ubuntu.com/p/cRS7qM8ZmP/`](https://pastebin.com/tKvcthaS)

I then also tried to troubleshoot via

`pip3 install evdev`

and ran `python3 RegisterDeviy.py` again  chose (even though I have a neutech usb rfid reader, but it was the only option that had USB in it :))
```
Choose the reader from list
0 vc4
1 Sycreader RFID Technology Co., Ltd SYC ID&IC USB Reader
Device Number: 1
```

Then it worked. 

## Software

### Base image and version
Raspbian GNU/Linux 11 (bullseye)
Lite 32bit

### Branch / Release
`master`

### Installscript
scripts/installscripts/buster-install-default.sh`

## Hardware

### RaspberryPi version
3 B+

### RFID Reader

Sycreader RFID Technology Co., Ltd SYC ID&IC USB Reader
### Soundcard

HifiBerry DAC

### Other notable hardware
-

s-martin, 2022-12-14:
Fixed in #1791 on develop. Not on master yet.

MarkusF1895, 2022-12-17:
Had the same problem. Try the develop bash. See here: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#b-developer-install-script-for-buster

tiffoknee, 2022-12-17:
I installed python3-evdev. (sudo apt-get install python3-evdev) then re-ran registerdevice.py and it sees the device - although it's currently stuck after selecting the reader

`pi@phonieone:~/RPi-Jukebox-RFID/scripts $ python RegisterDevice.py
Choose the reader from list
0 vc4
1 GeneralPlus USB Audio Device
2 Sycreader RFID Technology Co., Ltd SYC ID&IC USB Reader
Device Number: 2
.....`

https://github.com/joaorb64/joycond-cemuhook/issues/1

tiffoknee, 2022-12-17:
tried reinstalling - no joy. Tried the developer branch and now I get this error on registerdevice:

pi@raspberrypi:~ $ python3 /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py
Choose the reader from list
0 GeneralPlus USB Audio Device
1 Sycreader RFID Technology Co., Ltd SYC ID&IC USB Reader
Device Number: 1
Exception ignored in: <function BaseEventLoop.__del__ at 0x761cf6a8>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/base_events.py", line 622, in __del__
    warnings.warn(f"unclosed event loop {self!r}", ResourceWarning,
  File "/usr/lib/python3.7/asyncio/base_events.py", line 390, in __repr__
    f'<{self.__class__.__name__} running={self.is_running()} '
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1794, in get_debug
    return self._debug
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
Exception ignored in: <function BaseEventLoop.__del__ at 0x761cf6a8>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/base_events.py", line 622, in __del__
    warnings.warn(f"unclosed event loop {self!r}", ResourceWarning,
  File "/usr/lib/python3.7/asyncio/base_events.py", line 390, in __repr__
    f'<{self.__class__.__name__} running={self.is_running()} '
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1794, in get_debug
    return self._debug
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'

tiffoknee, 2022-12-17:
Think my problem might be the USB reader i have - swapped to rc522 and it's ok

s-martin, 2022-12-18:
V2.4.0 was released today - fixed.

closed 2022-12-18 by s-martin