MiczFlor/RPi-Jukebox-RFID #1924: 🐛 | ID of the RFID card does not appear in the web interface

MarkusF1895, 2022-12-16:
## Bug

### What I did

Install with Developer install script for Buster: cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; GIT_BRANCH=develop bash ./buster-install-default.sh
I had previously tried the default version and had problems with it. 
Connected with ssh command in Linux Terminal. 

### What happened

I can place audio folders and files in the web interface. I can also play them with the web interface.
However, when I try to register commands or audio folders on an RFID card, the problem arises. 
The ID is not written in the "Last used Chip ID" field of the web interface!
Instead, the ID is displayed in the terminal of the Raspberry Pi.

### I expected this to happen

I expect that the ID will be automatically entered in the corresponding field of the web interface after scanning.

### Further information that might help

I would like to receive a csv file with example RFID links to commands and audio folders.
Please upload one. 

## Software

### Base image and version

Raspbian GNU/Linux 11 (bullseye)
Released: 2022-09-22
with Raspberry Pi Imager v1.7.3

### Branch / Release

GIT_BRANCH=develop bash

### Installscript

cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; GIT_BRANCH=develop bash ./buster-install-default.sh

## Hardware

### RaspberryPi version

Raspberry Pi Bundle: Raspberry Pi 4 4GB + Gehäuse + Netzteil + Kabel
https://www.amazon.de/gp/product/B084NW35M5/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

### RFID Reader

RFID Reader
Neuftech USB RFID Reader ID Kartenlesegerät Kartenleser Kontaktlos Card Reader für EM4100
https://www.amazon.de/gp/product/B018OYOR3E/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

### Soundcard

Soundcard (not relevant)
CSL - Externe USB Soundkarte mit Virtual Surround Sound, Plug and Play
https://www.amazon.de/gp/product/B00C7LXUDY/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

### Other notable hardware
nope

note to myself: ssh pi@192.168.178.84

tiffoknee, 2022-12-17:
Does your reader beep when you swipe a card?  Trying to work out if my problem is the reader. My concern atm is that when I plug it into my mac it tries to setup as a HID which.. not ideal.

MarkusF1895, 2022-12-18:
> Does your reader beep when you swipe a card? Trying to work out if my problem is the reader. My concern atm is that when I plug it into my mac it tries to setup as a HID which.. not ideal.

Yes, the reader beeps. I am a bit confused. I connected the reader to the Pi, not the PC. Should I connect the reader to the PC?

tiffoknee, 2022-12-18:
No no, I just plugged mine into my pc to see if I could get any data out of it that way - I think my problem is different to yours. The fact that yours beeps means that it's detecting the card, which is a good start!  Your problem seems to be that the rfid is printing out to the command line and not going to the web UI?  This seems to imply that somehow the line with 'print' on it might be uncommented?  can you check in your Reader.py file and see what the last few lines look like?

[image]

tiffoknee, 2022-12-18:
Also, I had trouble with the bullseye version of the OS - it might be worth swapping to the 'legacy' version buster - I didn't read the instructions properly where it said the code is not fully compatible with bullseye!

MarkusF1895, 2022-12-18:
> No no, I just plugged mine into my pc to see if I could get any data out of it that way - I think my problem is different to yours. The fact that yours beeps means that it's detecting the card, [...]

First of all, thank you for the quick replies. Here is the Reader.py:
`pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ cat Reader.py`
`#!/usr/bin/env python3
# There are a variety of RFID readers out there, USB and non-USB variants.
# This might create problems in recognizing the reader you are using.
# We haven't found the silver bullet yet. If you can contribute to this
# quest, please comment in the issue thread or create pull requests.
# ALTERNATIVE SCRIPTS:
# If you encounter problems with this script Reader.py
# consider and test one of the alternatives in the same scripts folder.
# Replace the Reader.py file with one of the following files:
# * Reader.py.experimental
#     This alternative Reader.py script was meant to cover not only USB readers but more.
#     It can be used to replace Reader.py if you have readers such as
#     MFRC522, RDM6300 or PN532.
# * Reader.py.kkmoonRFIDreader
#     KKMOON RFID Reader which appears twice in the devices list as HID 413d:2107
#     and this required to check "if" the device is a keyboard.

# import string
# import csv
import os.path
import sys

from evdev import InputDevice, ecodes, list_devices
from select import select

def get_devices():
    return [InputDevice(fn) for fn in list_devices()]

class Reader:
    reader = None

    def __init__(self):
        self.reader = self
        path = os.path.dirname(os.path.realpath(__file__))
        self.keys = "X^1234567890XXXXqwertzuiopXXXXasdfghjklXXXXXyxcvbnmXXXXXXXXXXXXXXXXXXXXXXX"
        if not os.path.isfile(path + '/deviceName.txt'):
            sys.exit('Please run RegisterDevice.py first')
        else:
            with open(path + '/deviceName.txt', 'r') as f:
                deviceName = f.read()
            devices = get_devices()
            for device in devices:
                if device.name == deviceName:
                    self.dev = device
                    break
            try:
                self.dev
            except:
                sys.exit('Could not find the device %s\n. Make sure is connected' % deviceName)

    def readCard(self):
        stri = ''
        key = ''
        while key != 'KEY_ENTER':
            r, w, x = select([self.dev], [], [])
            for event in self.dev.read():
                if event.type == 1 and event.value == 1:
                    stri += self.keys[event.code]
                    # print( keys[ event.code ] )
                    key = ecodes.KEY[event.code]
        return stri[:-1]`

tiffoknee, 2022-12-18:
hmm, that is weird! I don't really see _how_ the OS could impact the code such that it returns to the command line instead of the UI but given all the other troubles people have had I think it might be worth reflashing with buster instead of bullseye. Or if you have a spare, a different SD card?

tiffoknee, 2022-12-18:
unless there's some debug code embedded somewhere - although why it would do that for you on bullseye and not for me on buster (I am also using the dev branch). Although! I am using the experimental Reader.py because I had to swap to an RC522 so it's a little apples and oranges for me.

MarkusF1895, 2022-12-18:
> hmm, that is weird! I don't really see _how_ the OS could impact the code such that it returns to the command line instead of the UI but given all the other troubles people have had I think it might be worth reflashing with buster instead of bullseye. Or if you have a spare, a different SD card?

Could you please send me a link to the buster version? Wouldn't like to try the wrong one. 😄

MarkusF1895, 2022-12-18:
> experimental Reader.py

I would also be happy about the experimental Reader.py. Then I can at least rule out a few causes.

tiffoknee, 2022-12-18:
in the raspberry pi imager, you select 'pi OS (other)'
[image]

and then whichever Legacy version is appropriate for you (I use the no desktop one because I ssh in and don't use a monitor)
[image]

MarkusF1895, 2022-12-18:
Strange. With the buster version I get exactly the error that I originally had. #1874 

Here is the error that occurs both during installation and when running the "RegisterDevice.py" script: 

`pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python3 RegisterDevice.py
Choose the reader from list
0 C-Media Electronics Inc.       USB PnP Sound Device
1 Sycreader RFID Technology Co., Ltd SYC ID&IC USB Reader
2 HID 04d9:2518 System Control
3 HID 04d9:2518 Consumer Control
4 HID 04d9:2518 Mouse
Device Number: 1
Exception ignored in: <function BaseEventLoop.__del__ at 0xb6130660>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/base_events.py", line 622, in __del__
    warnings.warn(f"unclosed event loop {self!r}", ResourceWarning,
  File "/usr/lib/python3.7/asyncio/base_events.py", line 390, in __repr__
    f'<{self.__class__.__name__} running={self.is_running()} '
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1794, in get_debug
    return self._debug
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
Exception ignored in: <function BaseEventLoop.__del__ at 0xb6130660>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/base_events.py", line 622, in __del__
    warnings.warn(f"unclosed event loop {self!r}", ResourceWarning,
  File "/usr/lib/python3.7/asyncio/base_events.py", line 390, in __repr__
    f'<{self.__class__.__name__} running={self.is_running()} '
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1794, in get_debug
    return self._debug
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
Exception ignored in: <function BaseEventLoop.__del__ at 0xb6130660>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/base_events.py", line 622, in __del__
    warnings.warn(f"unclosed event loop {self!r}", ResourceWarning,
  File "/usr/lib/python3.7/asyncio/base_events.py", line 390, in __repr__
    f'<{self.__class__.__name__} running={self.is_running()} '
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1794, in get_debug
    return self._debug
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
Exception ignored in: <function BaseEventLoop.__del__ at 0xb6130660>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/base_events.py", line 622, in __del__
    warnings.warn(f"unclosed event loop {self!r}", ResourceWarning,
  File "/usr/lib/python3.7/asyncio/base_events.py", line 390, in __repr__
    f'<{self.__class__.__name__} running={self.is_running()} '
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1794, in get_debug
    return self._debug
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
Exception ignored in: <function BaseEventLoop.__del__ at 0xb6130660>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/base_events.py", line 622, in __del__
    warnings.warn(f"unclosed event loop {self!r}", ResourceWarning,
  File "/usr/lib/python3.7/asyncio/base_events.py", line 390, in __repr__
    f'<{self.__class__.__name__} running={self.is_running()} '
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1794, in get_debug
    return self._debug
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'`

MarkusF1895, 2022-12-18:
By the way, the device name was written in deviceName.txt, but the error still appears! This surprises me, since the "RegisterDevice.py" script ends after writing the txt.

Content of deviceName.txt: `Sycreader RFID Technology Co., Ltd SYC ID&IC USB Reader`

tiffoknee, 2022-12-18:
Hmm, you have the same device that I do. That sycreader one - i couldn’t
even get mine to beep or output to the command line. Hopefully someone with
a working neuftech will come along to confirm whether their one displays as
that. I gave up, thinking mine was wrong in some way and swapped to the
rc522.

Is it still beeping and writing the rfid to the commansline?

MarkusF1895, 2022-12-18:
Yes it beeps. But I can't say if the number is written somewhere. Before it was visible on the desktop in the command window.

MarkusF1895, 2022-12-20:
I want to receive a csv file with example RFID links to commands and audio folders.
Please upload one.

MarkusF1895, 2022-12-20:
Perfect, thank you. I will test it out on Thursday or tomorrow.

Regarding the commands: In addition to the CSV, you probably have to maintain them in the "RPi-Jukebox-RFID/settings/rfid_trigger_play.conf.sample" file. @tiffoknee, could you look and check it out for me?

MarkusF1895, 2022-12-20:
I'll try the latest version also then.

s-martin, 2022-12-27:
Version 2.4 was released couple of days ago fixing also the evdev bug.

 you might want to check, if this fixes your issue.

lom-matze, 2023-02-17:
Hi there,
I really need your help!

My neuftech RFID reader is not recognized via 
`python3 RegisterDevice.py`

I've :
- neutech RFID reader
- - it is peeping and LED is on, but never show an RFID card ID in interface
- - it works, I've tested it with RFID software and it shows me the IDs
- my `deviceName.txt` is empty
- everything else is runnging (can play music via display or webinterface etc.)
- installed `sudo pip3 install evdev`
- the script `python3 RegisterDevice.py` is now running, but not showing any device to select.
[image]
- reinstall with `rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh
` did NOT help as well

What can I try? 

Would be so thankful for your help!

Thanks and cheers!
Matthias

s-martin, 2023-02-17:
What’s the output of `tail /var/log/syslog -n 200`?

lom-matze, 2023-02-18:
> What’s the output of `tail /var/log/syslog -n 200`?

Thanks for quick help.
I've added some syslog logs here below. The last 200 (at the bottom) lines and a fail -f syslog while I'm trying to activate the RFID reader in the webinterface ( it is off there). But the problem is, that it is not registered or intialized.

If I try to ` python3 RegisterDevice.py`, there is this output:
[image]
and non output in syslog.

here the syslog, tail f-, while I try to activate the RFID reader in the web interface:
`Feb 18 17:36:20 ghostbox lighttpd[1519]: Unit phoniebox-gpio-control.service could not be found.
Feb 18 17:36:26 ghostbox systemd[1]: Started Phoniebox RFID-Reader Service.
Feb 18 17:36:27 ghostbox lighttpd[1549]: Unit phoniebox-gpio-control.service could not be found.
Feb 18 17:36:27 ghostbox daemon_rfid_reader.py[1531]: Could not find the device
Feb 18 17:36:27 ghostbox daemon_rfid_reader.py[1531]: . Make sure is connected
Feb 18 17:36:27 ghostbox systemd[1]: phoniebox-rfid-reader.service: Main process exited, code=exited, status=1/FAILURE
Feb 18 17:36:27 ghostbox systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
Feb 18 17:36:27 ghostbox systemd[1]: phoniebox-rfid-reader.service: Scheduled restart job, restart counter is at 1.
Feb 18 17:36:27 ghostbox systemd[1]: Stopped Phoniebox RFID-Reader Service.
Feb 18 17:36:27 ghostbox systemd[1]: Started Phoniebox RFID-Reader Service.
Feb 18 17:36:28 ghostbox daemon_rfid_reader.py[1558]: Could not find the device
Feb 18 17:36:28 ghostbox daemon_rfid_reader.py[1558]: . Make sure is connected
Feb 18 17:36:28 ghostbox systemd[1]: phoniebox-rfid-reader.service: Main process exited, code=exited, status=1/FAILURE
Feb 18 17:36:28 ghostbox systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
Feb 18 17:36:28 ghostbox systemd[1]: phoniebox-rfid-reader.service: Scheduled restart job, restart counter is at 2.
Feb 18 17:36:28 ghostbox systemd[1]: Stopped Phoniebox RFID-Reader Service.
Feb 18 17:36:28 ghostbox systemd[1]: Started Phoniebox RFID-Reader Service.
Feb 18 17:36:28 ghostbox daemon_rfid_reader.py[1559]: Could not find the device
Feb 18 17:36:28 ghostbox daemon_rfid_reader.py[1559]: . Make sure is connected
Feb 18 17:36:29 ghostbox systemd[1]: phoniebox-rfid-reader.service: Main process exited, code=exited, status=1/FAILURE
Feb 18 17:36:29 ghostbox systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
Feb 18 17:36:29 ghostbox systemd[1]: phoniebox-rfid-reader.service: Scheduled restart job, restart counter is at 3.
Feb 18 17:36:29 ghostbox systemd[1]: Stopped Phoniebox RFID-Reader Service.
Feb 18 17:36:29 ghostbox systemd[1]: Started Phoniebox RFID-Reader Service.
Feb 18 17:36:29 ghostbox daemon_rfid_reader.py[1561]: Could not find the device
Feb 18 17:36:29 ghostbox daemon_rfid_reader.py[1561]: . Make sure is connected
Feb 18 17:36:29 ghostbox systemd[1]: phoniebox-rfid-reader.service: Main process exited, code=exited, status=1/FAILURE
Feb 18 17:36:29 ghostbox systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
Feb 18 17:36:30 ghostbox systemd[1]: phoniebox-rfid-reader.service: Scheduled restart job, restart counter is at 4.
Feb 18 17:36:30 ghostbox systemd[1]: Stopped Phoniebox RFID-Reader Service.
Feb 18 17:36:30 ghostbox systemd[1]: Started Phoniebox RFID-Reader Service.
Feb 18 17:36:30 ghostbox daemon_rfid_reader.py[1563]: Could not find the device
Feb 18 17:36:30 ghostbox daemon_rfid_reader.py[1563]: . Make sure is connected
Feb 18 17:36:30 ghostbox systemd[1]: phoniebox-rfid-reader.service: Main process exited, code=exited, status=1/FAILURE
Feb 18 17:36:30 ghostbox systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
Feb 18 17:36:30 ghostbox systemd[1]: phoniebox-rfid-reader.service: Scheduled restart job, restart counter is at 5.
Feb 18 17:36:30 ghostbox systemd[1]: Stopped Phoniebox RFID-Reader Service.
Feb 18 17:36:30 ghostbox systemd[1]: phoniebox-rfid-reader.service: Start request repeated too quickly.
Feb 18 17:36:30 ghostbox systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
Feb 18 17:36:30 ghostbox systemd[1]: Failed to start Phoniebox RFID-Reader Service.`

here the syslog (last 200 lines):
`Feb 17 21:00:49 ghostbox systemd[1]: Started Phoniebox RFID-Reader Service.
Feb 17 21:00:49 ghostbox dhcpcd[456]: wlan0: Router Advertisement from fe80::3ea6:2fff:feb2:5c4e
Feb 17 21:00:49 ghostbox dhcpcd[456]: wlan0: adding address 2a01:c23:8405:5300:9e:9c62:ea32:d893/64
Feb 17 21:00:49 ghostbox dhcpcd[456]: wlan0: adding route to 2a01:c23:8405:5300::/64
Feb 17 21:00:49 ghostbox dhcpcd[456]: wlan0: requesting DHCPv6 information
Feb 17 21:00:49 ghostbox avahi-daemon[383]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::d41d:71bd:f120:ff9a.
Feb 17 21:00:49 ghostbox avahi-daemon[383]: Joining mDNS multicast group on interface wlan0.IPv6 with address 2a01:c23:8405:5300:9e:9c62:ea32:d893.
Feb 17 21:00:49 ghostbox avahi-daemon[383]: Registering new address record for 2a01:c23:8405:5300:9e:9c62:ea32:d893 on wlan0.*.
Feb 17 21:00:49 ghostbox avahi-daemon[383]: Withdrawing address record for fe80::d41d:71bd:f120:ff9a on wlan0.
Feb 17 21:00:49 ghostbox dhcpcd[456]: wlan0: adding default route via fe80::3ea6:2fff:feb2:5c4e
[... cut ...]
Feb 18 17:31:24 ghostbox systemd[1]: phpsessionclean.service: Succeeded.
Feb 18 17:31:24 ghostbox systemd[1]: Finished Clean php session files.
Feb 18 17:31:26 ghostbox systemd[1]: man-db.service: Succeeded.
Feb 18 17:31:26 ghostbox systemd[1]: Finished Daily man-db regeneration.
Feb 18 17:31:31 ghostbox systemd[1]: systemd-hostnamed.service: Succeeded.
Feb 18 17:31:55 ghostbox systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
Feb 18 17:33:22 ghostbox systemd[1]: Started Session 3 of user pi.`

s-martin, 2023-02-26:
What’s the output of `ls -la /dev/input/by-id/`?

s-martin, 2023-02-26:
And can you try `sudo python3 -m pip install --upgrade --force-reinstall evdev`?

MarkusF1895, 2023-02-27:
Thank you for your efforts. Sorry forgot to close the bug. I gave up on this code; too big and intransparent. I implemented this one instead: https://github.com/talaexe/Spotify-RFID-Record-Player

By the way, the "Neuftech USB RFID Reader ID" never works with this code here. The hardware must be released for its use. I haven't seen them anywhere here. Troubleshooting was much easier with the other code, and I don't need the fancy GUI either.

closed 2023-02-27 by MarkusF1895