MiczFlor/RPi-Jukebox-RFID #1926: 🐛 | Next track command is not navigating chapters in m4b audiobooks

tiffoknee, 2022-12-18:
## Bug

### What I did

I have the develop branch of v2 running on a buster OS on 3b+

### What happened

I imported some m4b files

### I expected this to happen

I expected the chapters to import as per @sandreas PR here: https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1057 and for the 'next track' command to skip to the next chapter.

### Further information that might help

my syslog is a bit swamped with some unrelated issues..

`Dec 18 14:31:59 raspberrypi systemd[1]: phoniebox-buttons-usb-encoder.service: Main process exited, code=exited, status=203/EXEC
Dec 18 14:31:59 raspberrypi systemd[1]: phoniebox-buttons-usb-encoder.service: Failed with result 'exit-code'.
Dec 18 14:32:09 raspberrypi systemd[1]: phoniebox-buttons-usb-encoder.service: Service RestartSec=10s expired, scheduling restart.
Dec 18 14:32:09 raspberrypi systemd[1]: phoniebox-buttons-usb-encoder.service: Scheduled restart job, restart counter is at 104.
Dec 18 14:32:09 raspberrypi systemd[1]: Stopped Phoniebox Buttons USB Encoder Service.
Dec 18 14:32:09 raspberrypi systemd[1]: Started Phoniebox Buttons USB Encoder Service.
Dec 18 14:32:09 raspberrypi systemd[9851]: phoniebox-buttons-usb-encoder.service: Failed to execute command: Permission denied
Dec 18 14:32:09 raspberrypi systemd[9851]: phoniebox-buttons-usb-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/components/controls/buttons_usb_encoder/buttons_usb_encoder.py: Permission denied
Dec 18 14:32:09 raspberrypi systemd[1]: phoniebox-buttons-usb-encoder.service: Main process exited, code=exited, status=203/EXEC
Dec 18 14:32:09 raspberrypi systemd[1]: phoniebox-buttons-usb-encoder.service: Failed with result 'exit-code'.
Dec 18 14:32:14 raspberrypi systemd[1]: Started Session 4 of user pi.`

## Software

### Base image and version
Raspbian GNU/Linux 10 (buster)

### Branch / Release
develop

### Installscript
`scripts/installscripts/buster-install-default.sh`

## Hardware
### RaspberryPi version
`3 B+`

s-martin, 2023-05-01:
Are you using USB buttons?
If yes I assume the error messages in syslog are in fact related as the button commands don’t go through.

Do other button functions work?

s-martin, 2023-12-05:
Closing as no interaction for a couple of months.
Feel free to reopen the issue, if the problem still occurs.

closed 2023-12-05 by s-martin