MiczFlor/RPi-Jukebox-RFID #1942: 🚀 | Add Event Device plugin to support USB controllers

votti, 2022-12-30:
## Feature Description

### What functionality would you like to see in your phoniebox?

It should be easy to configure the phoniebox to listen to generic event devices, such as an USB controller/joystick or keyboard.

### How do you envision the feature to work from a users perspective?

The user should be able to configure an USB controller and the button mapping -> RPC in his main configuration file.

### Further information that might help

This can be implemented by adding a plugin around the `EvDevKeyListener`

s-martin, 2023-05-01:
For the V2 implementation of USB encoder see https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/controls/buttons_usb_encoder

s-martin, 2024-02-20:
Implemented in #1943

closed 2024-02-20 by s-martin