MiczFlor/RPi-Jukebox-RFID #1951: .flake8 broken on future3

votti, 2023-01-02:
### Describe your problem 

The .flake8 config file is currently broken in future3/develop, causing all CICD/pre commit runs to fail.

### Error
```
  File "/opt/hostedtoolcache/Python/3.9.16/x64/lib/python3.9/configparser.py", line 697, in read
    self._read(fp, filename)
  File "/opt/hostedtoolcache/Python/3.9.16/x64/lib/python3.9/configparser.py", line 1116, in _read
    raise e
configparser.ParsingError: Source contains parsing errors: '.flake8'
	[line 15]: 'count\n'
```

### Potential mitigation

remove line 15: `count`

(But I dont know what the purpose of this line should be.)

s-martin, 2023-01-02:
It probably should be `count = True`.
I wonder, if this never worked?

s-martin, 2023-01-03:
I merged #1954 to future3/develop, should work now.

@votti, can you check in your environment?

votti, 2023-01-04:
Thanks a lot @s-martin - everything looks good now and the pre-commit hook is working again.

PR #1954 (2023-01-03): Fix count in flake8 config
Should fix #1951