MiczFlor/RPi-Jukebox-RFID #1963: 🚀 | Add MQTT component to support control via Smart Home solution

c0un7-z3r0, 2023-01-29:
## Feature Description

Add MQTT component to Future3. 

### What functionality would you like to see in your phoniebox v3.x?

Integration with Smart Home solutions like Home Assistant via MQTT. Similar to how it is implemented in version 2.x

### How do you envision the feature to work from a users perspective?

As the user I can configure the connection to my MQTT broker via the jukebox.yml. Then the phoniebox is connecting to the broker and starts broadcasting various status messages e.g. Playerstate, Volume .... 
But It can also then be controlled via MQTT to allow for example via Home Assistant to mute the player.
The status of the connection to the broker should be easily visible in the webapp. 

### Further information that might help

The features should be on par with to how version 2.x has implemented plus potentially additional features that might be useful.

s-martin, 2023-01-31:
Not yet implemented https://rpi-jukebox-rfid.readthedocs.io/en/latest/featurelist.html#feature-status

pabera, 2023-03-16:
Wonderful feature which could use a pull request :-)

s-martin, 2023-12-28:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3%2Fdevelop/src%2Fjukebox%2Fjukebox%2Fpublishing%2Fserver.py#L139-L146

c0un7-z3r0, 2024-06-20:
More then a year later. I started working on this topic. Finally. Basic mqtt updates are working already now need to see which other values I need to map and how I implement the mqtt commands. Once this is done I will add some test and open up the PR.

But this is the first time I work with zeromq. Not sure if I followed the architecture and the ideas you linked to @s-martin . So any initial thoughts or quick feedback would be appreciated. 

Here is the WIP implementation https://github.com/c0un7-z3r0/RPi-Jukebox-RFID/blob/future3/add-mqtt/src/jukebox/components/mqtt/__init__.py

s-martin, 2024-06-21:
Thanks for starting the work on this feature! 🎉

We will look into it.

s-martin, 2025-02-05:
Implemented. Will be part of the 3.6 release.

closed 2025-02-05 by s-martin