MiczFlor/RPi-Jukebox-RFID #1964: 🐛 | No GPIO Functions after 2.1

baschior, 2023-01-31:
## Bug
After Installation of the 2.1 Version the gpio service is not working anymore.

### What I did

I installed the raspberry pi with above mentioned buster image and ran the installer script.
Erverything works fine as before but the configured gpio buttons are not working.

### I expected this to happen

On older versions the gpio service was working properly after configuration.

s-martin, 2023-01-31:
What's the output of `tail /var/log/sylph -n 200`?

baschior, 2023-02-01:
pi@phoniebox:~ $ tail /var/log/syslog -n 200
Feb  1 21:57:39 phoniebox systemd[1]: phoniebox-gpio-control.service: Service RestartSec=3s expired, scheduling restart.
Feb  1 21:57:39 phoniebox systemd[1]: phoniebox-gpio-control.service: Scheduled restart job, restart counter is at 35.
Feb  1 21:57:39 phoniebox systemd[1]: Stopped Phoniebox GPIO Control Service.
Feb  1 21:57:39 phoniebox systemd[1]: Started Phoniebox GPIO Control Service.
Feb  1 21:57:39 phoniebox mopidy[5756]: INFO     [MpdSession-154] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38270
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: INFO:__main__:GPIO Started
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: INFO:__main__:['VolumeControl', 'PrevNextControl', 'PlayPause', 'Shutdown', 'PauseShutdown', 'Volume0', 'VolumeUp', 'VolumeDown', 'NextSong', 'PrevSong', 'FastForward', 'Rewind', 'Halt', 'RFIDDevice']
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: INFO:__main__:Device VolumeControl not enabled
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: INFO:__main__:Device PrevNextControl not enabled
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: INFO:__main__:adding GPIO-Device, PlayPause
Feb  1 21:57:39 phoniebox mopidy[5756]: INFO     [MpdSession-155] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38274
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: INFO:__main__:adding GPIO-Device, Shutdown
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: INFO:__main__:Device PauseShutdown not enabled
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: INFO:__main__:Device Volume0 not enabled
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: INFO:__main__:adding GPIO-Device, VolumeUp
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: PlayPause
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: Shutdown
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: VolumeUp
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: Traceback (most recent call last):
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 127, in <module>
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]:     devices = gpio_controler.get_all_devices(config)
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 95, in get_all_devices
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]:     device = self.generate_device(config[section], section)
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 62, in generate_device
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]:     pull_up_down=config.get('pull_up_down', fallback='pull_up'))
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py", line 87, in __init__
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]:     bouncetime=self.bouncetime)
Feb  1 21:57:39 phoniebox PhonieboxGPIOControl[18355]: RuntimeError: Failed to add edge detection
Feb  1 21:57:39 phoniebox systemd[1]: phoniebox-gpio-control.service: Main process exited, code=exited, status=1/FAILURE
Feb  1 21:57:39 phoniebox systemd[1]: phoniebox-gpio-control.service: Failed with result 'exit-code'.
Feb  1 21:57:40 phoniebox mopidy[5756]: INFO     [MpdSession-156] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38278
Feb  1 21:57:41 phoniebox mopidy[5756]: INFO     [MpdSession-157] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38282
Feb  1 21:57:42 phoniebox mopidy[5756]: INFO     [MpdSession-158] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38286
Feb  1 21:57:42 phoniebox systemd[1]: phoniebox-gpio-control.service: Service RestartSec=3s expired, scheduling restart.
Feb  1 21:57:42 phoniebox systemd[1]: phoniebox-gpio-control.service: Scheduled restart job, restart counter is at 36.
Feb  1 21:57:42 phoniebox systemd[1]: Stopped Phoniebox GPIO Control Service.
Feb  1 21:57:42 phoniebox systemd[1]: Started Phoniebox GPIO Control Service.
Feb  1 21:57:43 phoniebox PhonieboxGPIOControl[18380]: INFO:__main__:GPIO Started
Feb  1 21:57:43 phoniebox PhonieboxGPIOControl[18380]: INFO:__main__:['VolumeControl', 'PrevNextControl', 'PlayPause', 'Shutdown', 'PauseShutdown', 'Volume0', 'VolumeUp', 'VolumeDown', 'NextSong', 'PrevSong', 'FastForward', 'Rewind', 'Halt', 'RFIDDevice']
Feb  1 21:57:43 phoniebox PhonieboxGPIOControl[18380]: INFO:__main__:Device VolumeControl not enabled
Feb  1 21:57:43 phoniebox PhonieboxGPIOControl[18380]: INFO:__main__:Device PrevNextControl not enabled
Feb  1 21:57:43 phoniebox PhonieboxGPIOControl[18380]: INFO:__main__:adding GPIO-Device, PlayPause
Feb  1 21:57:43 phoniebox PhonieboxGPIOControl[18380]: INFO:__main__:adding GPIO-Device, Shutdown
Feb  1 21:57:43 phoniebox PhonieboxGPIOControl[18380]: INFO:__main__:Device PauseShutdown not enabled
Feb  1 21:57:43 phoniebox PhonieboxGPIOControl[18380]: INFO:__main__:Device Volume0 not enabled
Feb  1 21:57:43 phoniebox PhonieboxGPIOControl[18380]: INFO:__main__:adding GPIO-Device, VolumeUp
Feb  1 21:57:43 phoniebox PhonieboxGPIOControl[18380]: PlayPause
Feb  1 21:57:43 phoniebox PhonieboxGPIOControl[18380]: Shutdown
Feb  1 21:57:43 phoniebox PhonieboxGPIOControl[18380]: VolumeUp
[... cut ...]
Feb  1 21:57:59 phoniebox mopidy[5756]: INFO     [MpdSession-181] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:38370
Feb  1 21:58:00 phoniebox systemd[1]: phoniebox-gpio-control.service: Service RestartSec=3s expired, scheduling restart.
Feb  1 21:58:00 phoniebox systemd[1]: phoniebox-gpio-control.service: Scheduled restart job, restart counter is at 41.
Feb  1 21:58:00 phoniebox systemd[1]: Stopped Phoniebox GPIO Control Service.
Feb  1 21:58:00 phoniebox systemd[1]: Started Phoniebox GPIO Control Service.

s-martin, 2023-02-01:
```

Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]: INFO:__main__:adding GPIO-Device, VolumeUp
Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]: PlayPause
Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]: Shutdown
Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]: VolumeUp
Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]: Traceback (most recent call last):
Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 127, in <module>
Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]:     devices = gpio_controler.get_all_devices(config)
Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 95, in get_all_devices
Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]:     device = self.generate_device(config[section], section)
Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 62, in generate_device
Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]:     pull_up_down=config.get('pull_up_down', fallback='pull_up'))
Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py", line 87, in __init__
Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]:     bouncetime=self.bouncetime)
Feb  1 21:57:57 phoniebox PhonieboxGPIOControl[18482]: RuntimeError: Failed to add edge detection
Feb  1 21:57:57 phoniebox systemd[1]: phoniebox-gpio-control.service: Main process exited, code=exited, status=1/FAILURE
Feb  1 21:57:57 phoniebox systemd[1]: phoniebox-gpio-control.service: Failed with result 'exit-code'.
```

seems to be the issue here

baschior, 2023-02-02:
Thanks for your quick reaction...
I wasnt able to fix this myself.
All rest including the display works perfectly.
Besides spotify obviously

s-martin, 2023-03-12:
Are you using pin 7 (GPIO 4), which may cause this? https://forums.raspberrypi.com/viewtopic.php?t=205327

Google lists many other issues, but I’m not sure, if this applies here.
https://www.google.de/search?q=failed+to+add+edge+detection

baschior, 2023-03-13:
Actually yes, I use gpio 4 for shutdown... I will change that and try... Why is that gpio a Problem in the new version? It used to work...

s-martin, 2023-03-13:
To be honest I’m not sure.

in some of the search results I read that some GPIO behavior was changed from Python 2 to Python 3. Maybe you were using a Phoniebox version, which was still Python 2 code.

s-martin, 2023-04-28:
Does this work now?

baschior, 2023-05-11:
No sadly not. I cant get the gpio buttons to work ar all. Strange because they worked in a previous version. I will test again next week amd come back to you. Thanks for your help

topas-rec, 2023-12-28:
I did an installation on a pi 3 today and saw the same symptoms.
Reverting to Phoniebox 2.3 made the issue go away.

s-martin, 2023-12-28:
Did you install 2.5 or the new 3.4 (future3)?

topas-rec, 2023-12-28:
2.5

baschior, 2023-12-28:
> I did an installation on a pi 3 today and saw the same symptoms.
> Reverting to Phoniebox 2.3 made the issue go away.

How did you actually revert to 2.3?

s-martin, 2023-12-28:
Do you want to try version 3.x or do you have reasons to stay on V2?

topas-rec, 2023-12-28:
> How did you actually revert to 2.3?

I copied the link to the install script and replaced this link in the oneline install (line).

The link is:
https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/24258901330d9d7d433e0ed05cd46bbc7f22ec57/scripts/installscripts/buster-install-default.sh

Since the file name has changed it also had to be replaced.

topas-rec, 2023-12-28:
> Do you want to try version 3.x or do you have reasons to stay on V2?

I don't have the box in my hands anymore. It is not planned to update it to version 3.
I'll do that with my own one soon™.

AlvinSchiller, 2023-12-29:
Could you share your gpio configuration?

I tested with a push button on version 2.5.0 and as far as i can say its working. Tested `VolUp` and `PlayPause ` on the predefinied pins and also gpio 4.
Did you used a fresh configuration or did you have one from an older installation? and if yes what version number was the old?

topas-rec, 2024-01-06:
Not as a file. As I said, I cannot access the box any more.
It is the sample one with enabled volume, seek and play pause buttons.
This one: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/misc/sampleconfigs/gpio_settings.ini.sample

oxygi, 2024-01-30:
Same here. I have installed today a fresh 2.5 version with a RC522 connected and a button on PIN 29.
Everything works, except the button. `tail /var/log/syslog -n 200` after `sudo systemctl restart phoniebox-gpio-control`
:

Jan 30 21:40:23 raspberrypiJukebox systemd[1]: Stopping Phoniebox GPIO Control Service...
Jan 30 21:40:23 raspberrypiJukebox systemd[1]: phoniebox-gpio-control.service: Succeeded.
Jan 30 21:40:23 raspberrypiJukebox systemd[1]: Stopped Phoniebox GPIO Control Service.
Jan 30 21:40:23 raspberrypiJukebox systemd[1]: Started Phoniebox GPIO Control Service.
Jan 30 21:40:24 raspberrypiJukebox PhonieboxGPIOControl[3300]: INFO:__main__:GPIO Started
Jan 30 21:40:24 raspberrypiJukebox PhonieboxGPIOControl[3300]: INFO:__main__:['VolumeUp', 'VolumeDown']
Jan 30 21:40:24 raspberrypiJukebox PhonieboxGPIOControl[3300]: INFO:__main__:adding GPIO-Device, VolumeUp
Jan 30 21:40:24 raspberrypiJukebox PhonieboxGPIOControl[3300]: INFO:__main__:adding GPIO-Device, VolumeDown
Jan 30 21:40:24 raspberrypiJukebox PhonieboxGPIOControl[3300]: INFO:__main__:Ready for taking actions

You can see, that it is ready but nothing happens...

This is my gpio_settings.ini:

[DEFAULT]
enabled: True
antibouncehack: False

[VolumeUp]
Type: Button
Pin: 29
pull_up_down: pull_up
hold_time: 0.3
hold_mode: Repeat
functionCall: functionCallVolU

[VolumeDown]
Type: Button
Pin: 31
pull_up_down: pull_up
hold_time: 0.3
hold_mode: Repeat
functionCall: functionCallVolD

What am I doing wrong? I use a Rasberry Pi 2B

AlvinSchiller, 2024-01-30:
What edition are you using? Classic or Spotify?
And could you supply the full log of `tail /var/log/syslog -n 200` and the output of `sudo journalctl -u phoniebox-gpio-control.service -n 200`?
We couldn't narrow it down so far :/

s-martin, 2024-01-30:
This sounds strange
```
Jan 30 22:01:01 raspberrypiJukebox PhonieboxGPIOControl[427]: /home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py:88: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
Jan 30 22:01:01 raspberrypiJukebox PhonieboxGPIOControl[427]:   GPIO.setup(self.pin, GPIO.IN, pull_up_down=self.pull_up_down)
Jan 30 22:01:01 raspberrypiJukebox PhonieboxGPIOControl[427]: INFO:__main__:Ready for taking actions

```

AlvinSchiller, 2024-01-30:
This could be the problem

> /home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py:88: RuntimeWarning: This channel is already in use, ...

Can you share your whole gpio config file?

s-martin, 2024-01-30:
Is there a reason why you use `hold_mode`?

From my point of view this is only necessary, if you use TwoButtonControl, see https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/01734deb7c872c2ef241eb6b5c1990231aeee062/components/gpio_control/README.md#button

s-martin, 2024-01-30:
I suggest you remove hold_time and hold_mode.

Also if you set pull_up you should not use any resistors.

I.e.
```
[VolumeUp]
Type: Button
Pin: 29
pull_up_down: pull_up
functionCall: functionCallVolU
```

please check https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/components/gpio_control/README.md

AlvinSchiller, 2024-01-30:
> Is there a reason why you use `hold_mode`?
> 
> From my point of view this is only necessary, if you use TwoButtonControl, see https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/01734deb7c872c2ef241eb6b5c1990231aeee062/components/gpio_control/README.md#button

using `hold_mode` should be fine also for simplebuttons, as its also in the examples:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/4101984b1031996239903796ae8fc4ce180106a5/components/gpio_control/example_configs/gpio_settings.ini#L63

AlvinSchiller, 2024-01-30:
I think the problem is simple a misunderstanding with number used for `Pin`.
The docs state it has to be the GPIO number, not the board pin number.

Can you try and change your config from `Pin: 29` to `Pin: 5` and  from `Pin: 31` to `Pin: 6` ?

oxygi, 2024-01-31:
OMG - I am so sorry!! I counted the pins... Now it is working.

s-martin, 2024-01-31:
Glad it's working! Have fun.

s-martin, 2024-04-22:
Seems to be working. Feel free to reopen, if the issue still persists with version 2.6.

closed 2024-04-22 by s-martin

baschior, 2024-07-02:
Just to let you know and to thank you for your work. I skippes 2.6 and gave 2.7 with Spotify reintegration a Chance... And everything including Spotify works Like a Charm! Thanks a lot for your effort