MiczFlor/RPi-Jukebox-RFID #1972: 🐛 | Jukebox don't updates / reads RFID ID over RC522 reader.

spiderbit, 2023-02-17:
## Bug

### What I did
I installed like in the wiki described the raspios-buster-armhf-lite image 2021-12-02.
Set up wlan.
Used the stable script to install Jukebox. Setup my Pi with a RC522 Reader. 

### What happened

The RFID ID never get's updated / read

### I expected this to happen

The RFID ID get's updated / read

### Further information that might help
Manually using that script with deactivated rfid in the Webui gives me the ID:

#!/usr/bin/env python

```
import RPi.GPIO as GPIO
from mfrc522 import SimpleMFRC522

reader = SimpleMFRC522()

try:
        id, text = reader.read()
        print(id)
        print(text)
finally:
        GPIO.cleanup()
```
had to install mfrc522 python bib with:
`sudo pip3 install mfrc522`

## Software

### Base image and version

2021-12-02-raspbios-buster-armhf-lite.img

### Branch / Release
master

### Installscript
scripts/installscripts/buster-install-default.sh

## Hardware

### RaspberryPi version
1 B+

### RFID Reader
RC522
followed the wiring described as default in the wiki.

### Other notable hardware

This file doesn't exist in my installation.

s-martin, 2023-02-19:
What’s the output of `tail /var/log/syslog -n 200`?

spiderbit, 2023-02-20:
Here:
http://ix.io/4oBq

Deleted mpd update data...

spiderbit, 2023-02-22:
@s-martin sorry don't want to nag at you, but would be nice to get some feedback.

Is that normal:`
Unit phoniebox-gpio-control.service could not be found.`

s-martin, 2023-02-26:
That’s not normal and I’m not sure why it’s missing.

In `~/RPi-Jukebox-RFID/phoniebox_logs` should be logs from the installation.

could you paste the log of the installation?

spiderbit, 2023-02-27:
GPIOconfig="NO"

Has it maybe something to do with me having no hardware buttons should not matter right?

I don't see anything here you go:
http://ix.io/4pk1

spiderbit, 2023-03-07:
@s-martin so it's nice to find out what went wrong, but is there some sort of fix? Is there a package to install to get the gpio-control.service file installed? Or do you need more logs or anything, just would love to go forward with that project. 

You have all the logs relevant right? So no reason to keep it in the bad state?

AlvinSchiller, 2023-03-24:
Hi, have you already tried a reinstallation with a fresh image (up to date image aswell, as yours is from 2021)? Does your problem persist?

spiderbit, 2023-03-24:
@AlvinSchiller No, so you mean the bullseye image from here:
https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-32-bit

Well it's just that the wiki recommends to use buster over this, now I gone the save "stable" and recommend route and it does not work. Well sure I will try it when I come to it maybe later today.

spiderbit, 2023-03-24:
@AlvinSchiller Well in the Wiki there is a link to that image if you are not supposed to download that image should the wiki not point to the website like somewhere here maybe:

https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy

I don't know their release circle so I didn't know there are newer buster images that said should a apt update / upgrade not update everything?

AlvinSchiller, 2023-03-24:
Yes that should (in theory) make no difference and I cannot say it will not work.
But as I can confirm that this image will work with the installation and a rc522 reader, it will hopefully make it easier to find the issue, if you are going to try a fresh installation.

spiderbit, 2023-03-25:
Well just am trying again, not much done after dd the image on the system wlan config (stick) ssh then copy the one-line command from the stable branch.

```
################################################
WIFIconfig=NO
WIFIcountryCode=DE
EXISTINGuse=NO
AUDIOiFace="Headphone"
SPOTinstall="NO"
MPDconfig="YES"
DIRaudioFolders="/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
GPIOconfig="NO"
################################################

W: GPG error: https://apt.mopidy.com buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 78FD980E271D2943
E: The repository 'https://apt.mopidy.com buster InRelease' is not signed.
E: Unable to locate package libspotify-dev

Please select the RFID reader you want to use
1) USB-Reader (e.g. Neuftech)  3) PN532			      5) Multiple RFID reader
2) RC522		       4) Manual configuration
#? 2

[   26.172173] Under-voltage detected! (0x00050000)
[   28.252192] Voltage normalised (0x00000000)
```

Again same problem.
```
pi@raspberrypi:~ $ sudo pip3 install mfrc522
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting mfrc522
  Downloading https://files.pythonhosted.org/packages/d5/b5/d33c0634cece0931c3c4e0978b0db58f248045c3b379ccf2d512b76fe044/mfrc522-0.0.7-py3-none-any.whl
Requirement already satisfied: spidev in /usr/local/lib/python3.7/dist-packages (from mfrc522) (3.6)
Requirement already satisfied: RPi.GPIO in /usr/local/lib/python3.7/dist-packages (from mfrc522) (0.7.1)
Installing collected packages: mfrc522
Successfully installed mfrc522-0.0.7
pi@raspberrypi:~ $ python3 reader.py 
/usr/local/lib/python3.7/dist-packages/mfrc522/MFRC522.py:151: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(pin_rst, GPIO.OUT)
Error while reading!
Error while reading!
Error while reading!
841556827034

pi@raspberrypi:~ $ python3 reader.py 
^CTraceback (most recent call last):
  File "reader.py", line 8, in <module>
    id, text = reader.read()
  File "/usr/local/lib/python3.7/dist-packages/mfrc522/SimpleMFRC522.py", line 19, in read
    id, text = self.read_no_block()
  File "/usr/local/lib/python3.7/dist-packages/mfrc522/SimpleMFRC522.py", line 38, in read_no_block
    (status, TagType) = self.READER.MFRC522_Request(self.READER.PICC_REQIDL)
  File "/usr/local/lib/python3.7/dist-packages/mfrc522/MFRC522.py", line 259, in MFRC522_Request
    (status, backData, backBits) = self.MFRC522_ToCard(self.PCD_TRANSCEIVE, TagType)
  File "/usr/local/lib/python3.7/dist-packages/mfrc522/MFRC522.py", line 217, in MFRC522_ToCard
    n = self.Read_MFRC522(self.CommIrqReg)
  File "/usr/local/lib/python3.7/dist-packages/mfrc522/MFRC522.py", line 162, in Read_MFRC522
    val = self.spi.xfer2([((addr << 1) & 0x7E) | 0x80, 0])
KeyboardInterrupt
pi@raspberrypi:~ $ python3 reader.py 
841556827034

pi@raspberrypi:~ $ 
```
Don't know why it did not work the 2nd time but well now it works with the reader.py script. Very strange everything.

AlvinSchiller, 2023-03-25:
So still not working with phoniebox?
The undervoltage error seems strange. Is your power supply sufficient (this can cause the weirdest errors)?

And sry to bug with this, but can you post your current pinout configuration?

spiderbit, 2023-03-25:
It would not make sense even if the cpu undervolts, which seems not the case because it states "normalized", well I use a smartphone powersupply, but why would phonieboxes reader need more power than the python script?

I find following about this in the internet:

> Most “standard” 5V Micro-USB mobile phone, tablet, and digital camera chargers should work with the Raspberry Pi, although we (of course) would recommend that you utilise a high quality dedicated Raspberry Pi power supply to get the best results!

The powersupply I use currently and have used another one before has 2,4A max (at 5V), and because there is not much extra modules or I do no gaming or anything.

> The recommended input voltage is 5V, and the recommended input current is 2A.

It works down to 1A according to that website (ThePiHut) but on heavy cpu/gpu load that can cause instabilities. I use not even X so what GPU load should there exist and I am way over the recommondated values.

AlvinSchiller, 2023-03-27:
Just trying to figure out what could be wrong and checking the errors. 
So powersupply shouldn't be the issue.
The "W: GPG error" is caused from a change in mopidy, but as you are not using the spotify version, that shouldn't impact you.

Can you post your current pinout configuration, just to double check? 
That it works with the script isn't a valid proof that it will work in phoniebox.

spiderbit, 2023-03-28:
@AlvinSchiller I followed the wiki here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Wiring_for_RC522_card_reader

Should I post a photograph or something?

spiderbit, 2023-03-29:
UPDATE: I think it somehow works now, when I tried to make a photo of the pins, I noticed one cable not being connected, after connecting it first I had no success, but also the script didn't find something, but and gave me, input output error with any comment. so some sort of crash, but after boot the web interface updates the ID.

I am not 100% sure but I would argue that with my second attempt it had to do with cable connection issues but with my first installation attempt it was the software.

Because back then I tried it much more extensively forth and back and the script always worked while the webinterface didn't so I suspect that still there is a software issue if you install the version directly linked in the wiki so even if we can't determine it definitely I would suggest to update the first buster link in the wiki on the install site.

AlvinSchiller, 2023-04-01:
Glad that it works now. I hope the rest will work too :)

I will update the wiki as soon as i can (there are some other links that don't match and could be confusing).

AlvinSchiller, 2023-04-01:
So this issue can be closed?

spiderbit, 2023-04-01:
I think so yes

closed 2023-04-01 by spiderbit