MiczFlor/RPi-Jukebox-RFID #1991: Placing card starts the player, removal stops it doesn't work anymore

TobiBu, 2023-03-15:
## Bug

### What I did

### What happened

### I expected this to happen

### Further information that might help

## Software

### Base image and version

### Branch / Release

### Installscript

## Hardware

### RaspberryPi version

### RFID Reader

TobiBu, 2023-03-15:
I could narrow this down to the fact that for some reasons the while loop in daemon_rfid_reader.py takes quite long up to 5 seconds. this clashes with setting of the signal to force stop the playout which is set to 1 second.
Any ideas why sometimes one loop might take 5 seconds?

AlvinSchiller, 2023-03-23:
Hi, 
i tested your situation and couldn't reproduce it on a fresh installation (also with a RPi 4 and RC522 Reader).
If i add a manuell "sleep" after the reading of the card (before the signal will be reseted), then it behaves just like you described.
So something seems to be causing a delay while reading/processing the card.

Have you tried the rfid reader without phoniebox, if this delay is there noticable too (simple script to read the id)?

TobiBu, 2023-06-24:
Hi Alvin,

thanks for sharing these insights! 
I checked with top and for me there is no command "mpg123" running. Also manually invoking it does not result in any error.
So I assume my issue is something else.

Anyways, I fixed it by changing one line in "~/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py"

I changed the signal alarm from 1 second to 10 seconds which circumvents my issues. But that's only a very crude quick fix...

`if swipe_or_place == "PLACENOTSWIPE":
        # enable the signal alarm (if no card is present for 1 second)
        signal.alarm(10)`

Mannshoch, 2024-01-08:
As advised in (https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/2058#discussioncomment-7522310)  
I have a Raspberry Pi 3 Model B Rev 1.2
Debian
Beschreibung
Debian GNU/Linux 11 (bullseye)
2.4 - abcf6e92 - master

```
Jan  8 18:49:35 phoniebox daemon_mqtt_client.py[15891]:     client.tls_set(
Jan  8 18:49:35 phoniebox daemon_mqtt_client.py[15891]:   File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 796, in tls_set
Jan  8 18:49:35 phoniebox daemon_mqtt_client.py[15891]:     context.load_cert_chain(certfile, keyfile, keyfile_password)
Jan  8 18:49:35 phoniebox daemon_mqtt_client.py[15891]: FileNotFoundError: [Errno 2] No such file or directory
Jan  8 18:49:36 phoniebox systemd[1]: phoniebox-mqtt-client.service: Main process exited, code=exited, status=1/FAILURE
Jan  8 18:49:36 phoniebox systemd[1]: phoniebox-mqtt-client.service: Failed with result 'exit-code'.
Jan  8 18:49:36 phoniebox systemd[1]: phoniebox-mqtt-client.service: Consumed 1.124s CPU time.
Jan  8 18:49:46 phoniebox systemd[1]: phoniebox-mqtt-client.service: Scheduled restart job, restart counter is at 211.
Jan  8 18:49:46 phoniebox systemd[1]: Stopped Phoniebox MQTT Client Service.
Jan  8 18:49:46 phoniebox systemd[1]: phoniebox-mqtt-client.service: Consumed 1.124s CPU time.
Jan  8 18:49:46 phoniebox systemd[1]: Started Phoniebox MQTT Client Service.
Jan  8 18:49:47 phoniebox daemon_mqtt_client.py[15990]: Traceback (most recent call last):
Jan  8 18:49:47 phoniebox daemon_mqtt_client.py[15990]:   File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_mqtt_client.py", line 601, in <module>
Jan  8 18:49:47 phoniebox daemon_mqtt_client.py[15990]:     client.tls_set(
Jan  8 18:49:47 phoniebox daemon_mqtt_client.py[15990]:   File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 796, in tls_set
Jan  8 18:49:47 phoniebox daemon_mqtt_client.py[15990]:     context.load_cert_chain(certfile, keyfile, keyfile_password)
Jan  8 18:49:47 phoniebox daemon_mqtt_client.py[15990]: FileNotFoundError: [Errno 2] No such file or directory
Jan  8 18:49:47 phoniebox systemd[1]: phoniebox-mqtt-client.service: Main process exited, code=exited, status=1/FAILURE
Jan  8 18:49:47 phoniebox systemd[1]: phoniebox-mqtt-client.service: Failed with result 'exit-code'.
Jan  8 18:49:47 phoniebox systemd[1]: phoniebox-mqtt-client.service: Consumed 1.125s CPU time.
Jan  8 18:49:56 phoniebox lighttpd[16090]: No LSB modules are available.
Jan  8 18:49:57 phoniebox lighttpd[16203]: tail: cannot open '../logs/debug.log' for reading: No such file or directory
Jan  8 18:49:57 phoniebox systemd[1]: phoniebox-mqtt-client.service: Scheduled restart job, restart counter is at 212.
Jan  8 18:49:57 phoniebox systemd[1]: Stopped Phoniebox MQTT Client Service.
Jan  8 18:49:57 phoniebox systemd[1]: phoniebox-mqtt-client.service: Consumed 1.125s CPU time.
Jan  8 18:49:57 phoniebox systemd[1]: Started Phoniebox MQTT Client Service.
Jan  8 18:49:58 phoniebox daemon_mqtt_client.py[16207]: Traceback (most recent call last):
Jan  8 18:49:58 phoniebox daemon_mqtt_client.py[16207]:   File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_mqtt_client.py", line 601, in <module>
Jan  8 18:49:58 phoniebox daemon_mqtt_client.py[16207]:     client.tls_set(
Jan  8 18:49:58 phoniebox daemon_mqtt_client.py[16207]:   File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 796, in tls_set
Jan  8 18:49:58 phoniebox daemon_mqtt_client.py[16207]:     context.load_cert_chain(certfile, keyfile, keyfile_password)
Jan  8 18:49:58 phoniebox daemon_mqtt_client.py[16207]: FileNotFoundError: [Errno 2] No such file or directory
Jan  8 18:49:58 phoniebox systemd[1]: phoniebox-mqtt-client.service: Main process exited, code=exited, status=1/FAILURE
Jan  8 18:49:58 phoniebox systemd[1]: phoniebox-mqtt-client.service: Failed with result 'exit-code'.
Jan  8 18:49:58 phoniebox systemd[1]: phoniebox-mqtt-client.service: Consumed 1.128s CPU time.
Jan  8 18:50:08 phoniebox systemd[1]: phoniebox-mqtt-client.service: Scheduled restart job, restart counter is at 213.
Jan  8 18:50:08 phoniebox systemd[1]: Stopped Phoniebox MQTT Client Service.
Jan  8 18:50:08 phoniebox systemd[1]: phoniebox-mqtt-client.service: Consumed 1.128s CPU time.
Jan  8 18:50:08 phoniebox systemd[1]: Started Phoniebox MQTT Client Service.
Jan  8 18:50:09 phoniebox daemon_mqtt_client.py[16269]: Traceback (most recent call last):
Jan  8 18:50:09 phoniebox daemon_mqtt_client.py[16269]:   File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_mqtt_client.py", line 601, in <module>
Jan  8 18:50:09 phoniebox daemon_mqtt_client.py[16269]:     client.tls_set(
Jan  8 18:50:09 phoniebox daemon_mqtt_client.py[16269]:   File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 796, in tls_set
Jan  8 18:50:09 phoniebox daemon_mqtt_client.py[16269]:     context.load_cert_chain(certfile, keyfile, keyfile_password)
Jan  8 18:50:09 phoniebox daemon_mqtt_client.py[16269]: FileNotFoundError: [Errno 2] No such file or directory
Jan  8 18:50:09 phoniebox systemd[1]: phoniebox-mqtt-client.service: Main process exited, code=exited, status=1/FAILURE
Jan  8 18:50:09 phoniebox systemd[1]: phoniebox-mqtt-client.service: Failed with result 'exit-code'.
Jan  8 18:50:09 phoniebox systemd[1]: phoniebox-mqtt-client.service: Consumed 1.152s CPU time.
Jan  8 18:50:20 phoniebox systemd[1]: phoniebox-mqtt-client.service: Scheduled restart job, restart counter is at 214.
[... cut ...]
Jan  8 18:52:02 phoniebox systemd[1]: phoniebox-mqtt-client.service: Main process exited, code=exited, status=1/FAILURE
Jan  8 18:52:02 phoniebox systemd[1]: phoniebox-mqtt-client.service: Failed with result 'exit-code'.
Jan  8 18:52:02 phoniebox systemd[1]: phoniebox-mqtt-client.service: Consumed 1.162s CPU time.
```

```
WIFIconfig=NO
WIFIcountryCode=DE
EXISTINGuse=NO
AUDIOiFace="Headphone"
SPOTinstall="NO"
SPOTIuser=""
SPOTIpass=""
SPOTIclientid=""
SPOTIclientsecret=""
MPDconfig="YES"
DIRaudioFolders="/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
GPIOconfig="YES"
```

AlvinSchiller, 2024-01-08:
Hi, thanks for the log. 
Did you activate the mqtt service? It seems it is running but constantly restarting due to failure in the configuration.
Can you deactivate the service and see if the situation gets better?

Mannshoch, 2024-01-08:
I played around with mqtt for a while. As you said I deactivated it.
```
sudo systemctl stop phoniebox-mqtt-client
sudo systemctl disable phoniebox-mqtt-client
sudo systemctl reboot
```
The problem still exist.

```
Jan  8 22:26:22 phoniebox systemd[1]: Starting Samba NMB Daemon...
Jan  8 22:26:22 phoniebox systemd[1]: Starting /etc/rc.local Compatibility...
Jan  8 22:26:22 phoniebox systemd[1]: Started /etc/rc.local Compatibility.
Jan  8 22:26:22 phoniebox systemd[1]: Started Getty on tty1.
Jan  8 22:26:22 phoniebox systemd[1]: Reached target Login Prompts.
Jan  8 22:26:22 phoniebox systemd[1]: Started Samba NMB Daemon.
Jan  8 22:26:22 phoniebox systemd[1]: Starting Samba SMB Daemon...
Jan  8 22:26:23 phoniebox exim4[751]: Starting MTA: exim4.
Jan  8 22:26:23 phoniebox systemd[1]: Started LSB: exim Mail Transport Agent.
Jan  8 22:26:23 phoniebox systemd[1]: Started Samba SMB Daemon.
Jan  8 22:26:23 phoniebox lighttpd[752]: 2024-01-08 22:26:22: configfile.c.461) Warning: "mod_compress" is DEPRECATED and has been replaced with "mod_deflate".  A future release of lighttpd 1.4.x will not contain mod_compress and lighttpd may fail to start up
Jan  8 22:26:23 phoniebox lighttpd[752]: 2024-01-08 22:26:22: mod_deflate.c.567) DEPRECATED: compress.filetype replaced with deflate.mimetypes
Jan  8 22:26:23 phoniebox lighttpd[752]: 2024-01-08 22:26:22: mod_deflate.c.580) DEPRECATED: compress.cache-dir replaced with deflate.cache-dir
Jan  8 22:26:23 phoniebox systemd[1]: Started Lighttpd Daemon.
Jan  8 22:26:24 phoniebox systemd[1]: systemd-rfkill.service: Succeeded.
Jan  8 22:26:24 phoniebox bthelper[1034]: Changing power off succeeded
Jan  8 22:26:24 phoniebox lighttpd[1026]: 2024-01-08 22:26:23: configfile.c.461) Warning: "mod_compress" is DEPRECATED and has been replaced with "mod_deflate".  A future release of lighttpd 1.4.x will not contain mod_compress and lighttpd may fail to start up
Jan  8 22:26:24 phoniebox bthelper[594]: Changing power on succeeded
Jan  8 22:26:25 phoniebox bash[720]: [0:04] Decoding of startupsound.mp3 finished.
Jan  8 22:26:25 phoniebox bash[1041]: Updating DB (#1) ...
Jan  8 22:26:25 phoniebox bash[1041]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jan  8 22:26:25 phoniebox systemd[1]: Finished Phoniebox Startup.
Jan  8 22:26:25 phoniebox systemd[1]: Reached target Multi-User System.
Jan  8 22:26:25 phoniebox systemd[1]: Starting Update UTMP about System Runlevel Changes...
Jan  8 22:26:25 phoniebox systemd[1]: systemd-update-utmp-runlevel.service: Succeeded.
Jan  8 22:26:25 phoniebox systemd[1]: Finished Update UTMP about System Runlevel Changes.
Jan  8 22:26:25 phoniebox systemd[1]: Startup finished in 4.014s (kernel) + 18.664s (userspace) = 22.678s.

(List of Media)

Jan  8 22:26:42 phoniebox systemd[1]: systemd-fsckd.service: Succeeded.
Jan  8 22:26:57 phoniebox systemd-timesyncd[325]: Initial synchronization to time server 216.229.0.50:123 (2.debian.pool.ntp.org).
Jan  8 22:27:02 phoniebox systemd[1]: Stopping Phoniebox RFID-Reader Service...
Jan  8 22:27:02 phoniebox systemd[1]: phoniebox-rfid-reader.service: Succeeded.
Jan  8 22:27:02 phoniebox systemd[1]: Stopped Phoniebox RFID-Reader Service.
Jan  8 22:27:02 phoniebox systemd[1]: Started Phoniebox RFID-Reader Service.
Jan  8 22:27:02 phoniebox daemon_rfid_reader.py[1154]: 2024-01-08 22:27:02,945 - root - INFO - Dir_PATH: /home/pi/RPi-Jukebox-RFID/scripts
Jan  8 22:27:03 phoniebox systemd[1]: systemd-hostnamed.service: Succeeded.
Jan  8 22:27:04 phoniebox daemon_rfid_reader.py[1154]: 2024-01-08 22:27:04,149 - root - INFO - No RFID Signal detected.
Jan  8 22:27:04 phoniebox daemon_rfid_reader.py[1154]: 2024-01-08 22:27:04,149 - root - INFO - Trigger Pause Force
Jan  8 22:27:19 phoniebox daemon_rfid_reader.py[1154]: 2024-01-08 22:27:19,487 - root - INFO - Trigger Play Cardid=0662287364
Jan  8 22:27:20 phoniebox daemon_rfid_reader.py[1328]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jan  8 22:27:20 phoniebox daemon_rfid_reader.py[1362]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jan  8 22:27:20 phoniebox daemon_rfid_reader.py[1363]: loading: Lieder % kinderlieder - 100 Kinderlieder zum Mitsingen _ Die 30 Besten
Jan  8 22:27:20 phoniebox daemon_rfid_reader.py[1372]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jan  8 22:27:20 phoniebox daemon_rfid_reader.py[1381]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jan  8 22:27:20 phoniebox daemon_rfid_reader.py[1390]: OK MPD 0.22.4
Jan  8 22:27:20 phoniebox daemon_rfid_reader.py[1390]: OK
Jan  8 22:27:20 phoniebox mpd: exception: Failed to open mixer for 'My ALSA Device': no such mixer control: Headphone
Jan  8 22:27:22 phoniebox daemon_rfid_reader.py[1154]: 2024-01-08 22:27:22,943 - root - INFO - No RFID Signal detected.
Jan  8 22:27:22 phoniebox daemon_rfid_reader.py[1154]: 2024-01-08 22:27:22,944 - root - INFO - Trigger Pause Force
Jan  8 22:27:23 phoniebox daemon_rfid_reader.py[1448]: Kinderlieder zum Mitsingen || Kinderlieder
Jan  8 22:27:23 phoniebox daemon_rfid_reader.py[1448]: [paused]  #1/98   0:04/2:14 (2%)
Jan  8 22:27:23 phoniebox daemon_rfid_reader.py[1448]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jan  8 22:27:29 phoniebox daemon_rfid_reader.py[1154]: 2024-01-08 22:27:29,337 - root - INFO - Trigger Play Cardid=0662287364
[... cut ...]
Jan  8 22:31:23 phoniebox systemd[3642]: Listening on GnuPG cryptographic agent and passphrase cache.
Jan  8 22:31:23 phoniebox systemd[3642]: Reached target Sockets.
Jan  8 22:31:23 phoniebox systemd[3642]: Reached target Basic System.
Jan  8 22:31:23 phoniebox systemd[1]: Started User Manager for UID 1000.
Jan  8 22:31:23 phoniebox systemd[3642]: Reached target Main User Target.
Jan  8 22:31:23 phoniebox systemd[3642]: Startup finished in 486ms.
Jan  8 22:31:23 phoniebox systemd[1]: Started Session 1 of user pi.
```

AlvinSchiller, 2024-01-08:
Could you also post the result of `top`,  `amixer scontrols` and the content of the file `./setting/Audio_iFace_Name` to get a better picture of the possible problem?

Mannshoch, 2024-01-09:
top
```
top - 20:09:35 up 4 min,  1 user,  load average: 0.08, 0.18, 0.09
Tasks: 157 total,   1 running, 156 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.7 us,  1.8 sy,  0.0 ni, 97.5 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :    909.6 total,    478.3 free,    128.1 used,    303.2 buff/cache
MiB Swap:    100.0 total,    100.0 free,      0.0 used.    716.9 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                         
   2420 pi        20   0   10000   2936   2420 R   0.7   0.3   0:00.10 top                                                                                                             
     98 root       0 -20       0      0      0 I   0.3   0.0   0:00.33 kworker/2:1H-kblockd                                                                                            
    150 root      20   0   57180  21292  20232 S   0.3   2.3   0:01.47 systemd-journal                                                                                                 
    498 root      20   0    6692   3068   2756 S   0.3   0.3   0:01.41 cleanshutd                                                                                                      
    580 root       0 -20       0      0      0 I   0.3   0.0   0:00.16 kworker/u9:2-brcmf_wq/mmc1:0001:1                                                                               
    753 root      20   0   73412  14320  11964 S   0.3   1.5   0:00.31 nmbd                                                                                                            
      1 root      20   0  167044  10276   7288 S   0.0   1.1   0:04.47 systemd                                                                                                         
      2 root      20   0       0      0      0 S   0.0   0.0   0:00.03 kthreadd                                                                                                        
      3 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 rcu_gp                                                                                                          
      4 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 rcu_par_gp                                                                                                      
      5 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 slub_flushwq                                                                                                    
      6 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 netns                                                                                                           
      7 root      20   0       0      0      0 I   0.0   0.0   0:00.00 kworker/0:0-events                                                                                              
      8 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 kworker/0:0H-mmc_complete                                                                                       
      9 root      20   0       0      0      0 I   0.0   0.0   0:01.93 kworker/u8:0-events_unbound                                                                                     
     10 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 mm_percpu_wq                                                                                                    
     11 root      20   0       0      0      0 I   0.0   0.0   0:00.00 rcu_tasks_kthread                                                                                               
     12 root      20   0       0      0      0 I   0.0   0.0   0:00.00 rcu_tasks_rude_kthread                                                                                          
     13 root      20   0       0      0      0 I   0.0   0.0   0:00.00 rcu_tasks_trace_kthread                                                                                         
     14 root      20   0       0      0      0 S   0.0   0.0   0:00.04 ksoftirqd/0                                                                                                     
     15 root      20   0       0      0      0 I   0.0   0.0   0:00.41 rcu_preempt                                                                                                     
     16 root      rt   0       0      0      0 S   0.0   0.0   0:00.00 migration/0                                                                                                     
     17 root      20   0       0      0      0 S   0.0   0.0   0:00.00 cpuhp/0                                                                                                         
     18 root      20   0       0      0      0 S   0.0   0.0   0:00.00 cpuhp/1                                                                                                         
     19 root      rt   0       0      0      0 S   0.0   0.0   0:00.00 migration/1                                                                                                     
     20 root      20   0       0      0      0 S   0.0   0.0   0:00.03 ksoftirqd/1                                                                                                     
[... cut ...]
     32 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 kworker/3:0H-events_highpri                                                                                     
     33 root      20   0       0      0      0 S   0.0   0.0   0:00.01 kdevtmpfs                                                                                                       
     34 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 inet_frag_wq 
```

amixer scontrols
`Simple mixer control 'PCM',0`

`./setting/Audio_iFace_Name` contain Headphone

Mannshoch, 2024-01-09:
Until now I did not remove the Beeper from the scanner. How does the play until remove work? shouldn't there be a beep all x seconds?

AlvinSchiller, 2024-01-09:
> amixer scontrols `Simple mixer control 'PCM',0`
> 
> `./setting/Audio_iFace_Name` contain Headphone

This is a misconfiguration. Can you update your settings and replace `Headphone` with `PCM`, then reboot and try again?

The relevant files are:
/etc/mpd.conf
/home/pi/RPi-Jukebox-RFID/settings/global.conf
/home/pi/RPi-Jukebox-RFID/settings/Audio_iFace_Name
/home/pi/PhonieboxInstall.conf

AlvinSchiller, 2024-01-09:
> Until now I did not remove the Beeper from the scanner. How does the play until remove work? shouldn't there be a beep all x seconds?

It looks like the neuftech usb reader does not support the "place" option.
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1498#issuecomment-912650061

Mannshoch, 2024-01-10:
Could I somhow thest this?
I think it should work. because the auto shutdown not work if a nfc card is placed on the box.