MiczFlor/RPi-Jukebox-RFID #1993: 🚀 | Wifi Plugin

deimi, 2023-03-16:
## Feature Description
A plugin for controlling/monitoring the wifi interface 

### What functionality would you like to see in your phoniebox?
If a Phoniebox is powered by battery it would be very useful to be able to turn off Wifi completely with an RFID card or turn it on. This increases the battery runtime significantly (around 13% less power consumption on a Zero2). 

### How do you envision the feature to work from a users perspective?
The plugin provides functions for controlling the wifi interface and callbacks for monitoring it (e.g. status, signal strength, etc.)

deimi, 2023-03-16:
I am totally willing and happy to contribute this via a pull request to `future3`. 
With this issue I only want to check if such a plugin has a chance to be merged or if there are some reasons which speak against such a feature.

pabera, 2023-03-16:
This is a great idea. It shouldn't be that hard. We might want to define a few requirements because there are some edge cases we want to at least think about.

The future3 branch hasn't seen many updates recently. I believe we must bring it up to speed before you can start. I am actually trying to update all dependencies for the frontend at the moment.

s-martin, 2024-04-19:
> 
> The future3 branch hasn't seen many updates recently. I believe we must bring it up to speed before you can start. I am actually trying to update all dependencies for the frontend at the moment.
> 

@deimi the future3 branch is now up to speed, so we could think about implementing it.

As suggested I think it makes sense that we  discuss the the wanted use cases now.