MiczFlor/RPi-Jukebox-RFID #1994: 🚀 | Linux command plugin

deimi, 2023-03-16:
## Feature Description
A plugin which provides an RPC call for executing any linux terminal command.

### What functionality would you like to see in your phoniebox?
Being able to execute any linux command via an RPC call / RFID swipe as this would be especially useful when doing some special stuff which is not generic enough for an upstream plugin (e.g. controlling special hardware). This feature could also be useful to have when someone has an idea for a plugin but before implementing it a whole plugin, he could just try it out in a prototyping manner. 

### How do you envision the feature to work from a users perspective?
An RPC call which just takes the overhanded arguments and executes them in separate process.

deimi, 2023-03-16:
I'm also willing to contribute this to `future3`, but I want to ask if a pull request has a chance here.

deimi, 2023-03-16:
Also I'm not sure if it is a good idea to open such a big security hole, because it would mean that you can execute ANY command with the same user rights via RPC without any authentication.  
Probably it should be disabled by default and pretty good documented what the consequences could be if the user enables this plugin and it lies in his hands to mitigate the risk (e.g. with a good sudoers configuration)

pabera, 2023-03-16:
I don't think we should be doing this. It's a risk. If you need this feature and I can understand why you could use it, I believe it should be straight forward to implement. But then on your own private fork ;-)

On the other side, it might be worth thinking about the things you want to execute. Maybe you can implement them instead and allow every future3 to benefit from it.

deimi, 2023-03-16:
Yes I also think it may be a bit too risky. It was just an idea that popped up while thinking about the Wifi plugin.

closed 2023-03-16 by deimi