MiczFlor/RPi-Jukebox-RFID #2014: 🚀 | future3 - Add Hifiberry (Miniamp) support

s-martin, 2023-04-27:
## Feature Description

### What functionality would you like to see in your phoniebox?

I would love to have support for [https://www.hifiberry.com/shop/boards/miniamp/](Hifiberry Miniamp) (or other Hifiberry boards) in future3 

### How do you envision the feature to work from a users perspective?

while configuring Phoniebox Hifiberry should be automatically detected and configured.

### Further information that might help

Configuring Hifiberry https://www.hifiberry.com/docs/software/configuring-linux-3-18-x/

pabera, 2023-05-04:
It's been a while since I tried, but I had it working already in the past. What's the blocker that you are facing?

s-martin, 2023-12-10:
After I read https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/develop/documentation/builders/audio.md#audio-configuration correctly it wasn't so difficult though :D

closed 2023-12-10 by s-martin