MiczFlor/RPi-Jukebox-RFID #2023: Volume buttons don't work, iFace "Master"

Rommsel, 2023-05-02:
Hi, 
I know a few similar issues like this have been reported but unfortunately none of the answers has helped so far.

My hardware:
- Raspberry pi 3b+
- Trust Leto speaker with AUX directly plugged in raspberry pi (no external sound card)
- arcade buttons (one for volU, one for volD) connected to GPIO16 and GPIO19
- other buttons (pause, forward, back) are working, I'm pretty sure all solder connections are fine

outputs:
`cat /proc/asound/modules`:
 0 snd_bcm2835
 1 vc4
--> and 0 is also selected in the alsa.config

`cat /home/pi/RPi-Jukebox-RFID/settings/version`:
2.4 - abcf6e92 - master

`amixer scontrols`:
Simple mixer control 'Master',0
Simple mixer control 'Capture',0

`amixer sget Master`:
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 65536 [100%] [on]
  Front Right: Playback 65536 [100%] [on]

my audio_iFace_Name is "Master"

`~/RPi-Jukebox-RFID/settings/gpio_settings.ini`
[...]
[VolumeUp]
enabled: True
Type:  Button
Pin: 16
pull_up_down: pull_up
functionCall: functionCallVolU

[VolumeDown]
enabled: True
Type:  Button
Pin: 19
pull_up_down: pull_up
functionCall: functionCallVolD

I haven't tried different GPIO ports yet because I find it unlikely that both are not working. 
I also can't change the volume through the web interface. 
The speakers have a volume wheel on the case, could this interfere with the volume control?

Any ideas what else I could try?

AlvinSchiller, 2023-05-02:
Hi, can you provide your audio_output configuration of mpd.conf?

I had a similar problem with a miniamp and had to change the configuration of "mixer_type" from "hardware" to "software".

Rommsel, 2023-05-03:
Hi,
that did the trick. I actually changed it to "software" yesterday already but was too tired and desperate to try it properly. Now it seems to be working :) Thank you so much for your answer!

audio_output {
        type            "alsa"
        name            "My ALSA Device"
#       device          "hw:0,0"        # optional
        mixer_type      "software"      # optional
#       mixer_device    "default"       # optional
        mixer_control   "Master"        # optional
#       mixer_index     "0"             # optional
}

closed 2023-05-03 by Rommsel

kevindelord, 2023-12-14:
Editing /etc/mpd.conf with the snippet from @Rommsel worked for me as well! 
Thanks a ton