MiczFlor/RPi-Jukebox-RFID #2029: Release 2.5

s-martin, 2023-05-07:
This is an issue to coordinate all necessary steps to release [version 2.5](https://github.com/MiczFlor/RPi-Jukebox-RFID/milestone/8).

Some PRs and issues for 2.5 are still open and should be fixed, before 2.5 can be released.

- [ ] Fix all [issues for 2.5](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues?q=is%3Aissue+is%3Aopen+milestone%3A2.5)
- [ ] Merge all [PRs for 2.5](https://github.com/MiczFlor/RPi-Jukebox-RFID/pulls?q=is%3Apr+is%3Aopen+milestone%3A2.5)
- [ ] Create a PR to merge develop to master (i.e. release 2.5)
- [ ] update wiki(?)
- [ ] update news section in main Readme file
- [ ] fix version info

AlvinSchiller, 2023-05-08:
Question regarding "update wiki" as i stumbled across it during https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2008:
Should the wiki links that point to files in **develop** (".../tree/develop/..." and  ".../blob/develop/...") be updated to point to **master** after the release?

AlvinSchiller, 2023-12-21:
> Question regarding "update wiki" as i stumbled across it during #2008:
> Should the wiki links that point to files in **develop** (".../tree/develop/..." and ".../blob/develop/...") be updated to point to **master** after the release?

@s-martin @pabera any thoughts about that?

s-martin, 2023-12-21:
It would make sense that it points to master, do the docs fit to the implementation. But then updates and fixes of the docs are much slower.

But if we simplified the release process now, we could release more often.

AlvinSchiller, 2023-12-22:
Ok then I change it.

The wiki is also easy to change, so if this is going to be a problem, we can switch back anytime.

PR #2030 (2023-12-21): Release 2.5
Closes #2029
closed 2023-12-25 by AlvinSchiller