MiczFlor/RPi-Jukebox-RFID #2041: future3 | Install on bullseye not working (Wrong OS type)

BugHunter2k, 2023-07-01:
### Describe your problem 

Installation aborts with:

> Checking OS type ... aarch64
> ERROR: Only 32 bit operating systems supported. Please use a 32bit version of RaspianOS!

#### What's your hardware set up? 

RPi 4 Model B with 4GB 
OS: Raspberry Pi OS Lite (32-Bit)

installation Checks `uname -m` that gives `aarch64` (Procesortype)

Better Checks would be:
| command  | output |
|-----------------|------------| 
| `dpkg --print-architecture` | `armhf`
| `getconf LONG_BIT` | `32`

AlvinSchiller, 2023-07-01:
On Pi 4 a 64-bit kernel is running by default. See https://github.com/raspberrypi/firmware/issues/1795

It can be switched to 32-bit by adding arm_64bit=0 to config.txt.

s-martin, 2023-07-01:
> It can be switched to 32-bit by adding arm_64bit=0 to config.txt.
> 
> After that the installation should work.

Would make sense to add that to the docs.

pabera, 2023-08-14:
> > It can be switched to 32-bit by adding arm_64bit=0 to config.txt.
> > After that the installation should work.
> 
> Would make sense to add that to the docs.

+1

We can directly link to this in the install script as well.

pabera, 2023-08-14:
This has been added to the documentation for now https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2057
It might be interesting to use @BugHunter2k's solutions long-term. Maybe you find the time to put it into a pull request?

AlvinSchiller, 2023-11-20:
With the error reports from https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1815#issuecomment-1816263508 on using a mixed setup (32 bit OS and 64 bit Kernel), maybe this check is the right thing to do after all? @pabera 

If so, the message should be updated accordingly.

s-martin, 2023-12-04:
Do we know what does not work with 64 bit exactly?

pabera, 2023-12-04:
> Do we know what does not work with 64 bit exactly?

> ERROR: Only 32 bit operating systems supported. Please use a 32bit version of RaspianOS!

[We are defining this boundary ourselves](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/installation/includes/02_helpers.sh#L40-L58), and in fact we haven't tested if this actually already works. Docker Development environments run on 64bit, so I would believe this can also work for RPi with `amd64`/`aarch64`.

We just need to test and in case all works well, we can remove this check entirely.

But, the question is, what is the optimal solution for a Phoniebox when it comes to speed (on start up) and compatibility (of newer and older libaries). Here are a few considerations.

**32-bit Architecture**
* Pros: 
  1. Compatibility with older software
  2. Lower RAM usage
  3. Better stability on older Raspberry Pi models.
Cons:
  1. Limited RAM utilization on models with 4GB or more
  2. Slower performance for some tasks
  3. Potential future compatibility issues

**64-bit Architecture:**
* Pros
  1. Full RAM utilization on models with 4GB or more
  2. Improved performance for resource-intensive tasks
  3. Better future compatibility
* Cons
  1. Potential compatibility issues with older software
  2. Higher RAM usage

s-martin, 2023-12-04:
I think once we have tested 64 bit, we could remove the check.

Older Pi’s run on 32 bit anyway, so it shouldn’t be a problem for those, right?

wrsmcode, 2024-01-14:
Hello tother...
i have tested the installation prozess and can agree with this settings to config.txt:
arm_64bit=0

AND, i think an important step, that is not documented yet --> Expand filesystem!
by reading the install-steps, i didn't do that, and after about 3/4h  my installation-prozess terminates by an error.
lookin in the log file, i'll knew the error --> not enough space :/

would be nice, if someone can write this in the installation-description.

cheers
Joko

AlvinSchiller, 2024-01-16:
Hi Joko,

> AND, i think an important step, that is not documented yet --> Expand filesystem!

Usually the expansion is performed by default on first boot.
How did you flash your SD Card and what image did you use?

s-martin, 2024-06-07:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2313#issuecomment-2155357857

64 bit seems to work, we could discuss, if we remove that boundary @pabera

s-martin, 2024-07-14:
#2402 may also be 64bit related.

harrydoddnoble, 2024-09-15:
the 5 cannot run 32 bit kernel - is there any way to run Jukebox on a 5?

s-martin, 2024-11-07:
#2425 fixed.

closed 2024-11-07 by s-martin