MiczFlor/RPi-Jukebox-RFID #2048: 🐛 | BUG SUMMARY Replacing pirc with mfrc

Sanduur, 2023-07-14:
## pirc522 is not working in Reader.py

I installed this github on a freshly clean Raspian Environment, but for me, the code in the Reader.py.experimental file is not working. I have the blue RFID-RC522 module, connected to GPIO. Here is the original Code:

class Mfrc522Reader(object):
    def __init__(self):
        import pirc522
        self.device = pirc522.RFID()

    def readCard(self):
        # Scan for cards
        self.device.wait_for_tag()
        (error, tag_type) = self.device.request()

        if not error:
            logger.info("Card detected.")
            # Perform anti-collision detection to find card uid
            (error, uid) = self.device.anticoll()
            if not error:
                card_id = ''.join((str(x) for x in uid))
                logger.info(card_id)
                return card_id
        logger.debug("No Device ID found.")
        return None

    @staticmethod
    def cleanup():
        GPIO.cleanup()

**However I found a Tutorial (Sorry, its german language: https://devdrik.de/pi-mit-rfid-rc522/) and I found out that my Module is working fine.**

Unfortunately I am not a really good python developer but in 2023 we have access to chatGPT :) I asked to convert the class to work with SimpleMFRC522 and chatGPT entered this code for me (which is working well):

class Mfrc522Reader(object):
    def __init__(self):
        from mfrc522 import SimpleMFRC522
        self.device = SimpleMFRC522()

    def readCard(self):
        try:
            # Scan for cards
            (uid, text) = self.device.read()

            if uid:
                card_id = str(uid)
                logger.info(card_id)
                return card_id
        except Exception as e:
            logger.error("Error reading card: {}".format(str(e)))
            logger.error("uid: {}".format(uid))

        logger.debug("No card found.")
        return None

    @staticmethod
    def cleanup():
        GPIO.cleanup()

Just wanted to share because some others may have the same issue. Feel free to import this code in the github, I love this project!! I use the following version:

$ uname -r; python -V
6.1.21-v8+
Python 3.9.2

AlvinSchiller, 2023-07-14:
Hi, just for clarification: do you have the IRQ pin connected to Pin 18/GPIO 24? https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Wiring_for_RC522_card_reader

This is needed in this project and differs from your linked tutorial.

Sanduur, 2023-07-15:
I did NOT connect Pin 18 / GPIO 24... Even if "**You have to wire the IRQ Pin from RC522 to Raspberry Pin 18/GPIO 24**" in the manual is written in bold.

Thank you very much! Now it works as designed

AlvinSchiller, 2023-08-01:
So this can be closed, right?

Sanduur, 2023-08-07:
Yes. Thank you very much.

closed 2023-08-07 by Sanduur