MiczFlor/RPi-Jukebox-RFID #2050: Websocket failure on future3

Shaka-Agina, 2023-07-26:
Hey Just fount out there's a websocket failure after installing future 3. Any Idea of what could be causing this?

This is the error found on the dev console on Chrome:

"webSocketEndpoint.ts:56 WebSocket connection to 'ws://raspberrypi.local:5557/' failed: "

Hardware Raspberry Pi4 8GB

Dabylpmis, 2023-08-14:
I got the same issue. First at all the Jukebox daemon won't start because it found no pyzmq. After installing it via 
pip3 install pyzmq the daemon tries to start and the boot up sound starts to play but stops after 1 sec. 

Here is the output from:
systemctl --user status jukebox-daemon 

```python
592:plugs.py           - jb.plugin            - MainThread      - ERROR    - Ignoring failed package load for 'publishing'
593:plugs.py           - jb.plugin            - MainThread      - ERROR    - Reason: ZMQError: Protocol not supported (addr='ws://*:5557')
jb.plugin            - MainThread      - ERROR    - Detailed reason:
python3[1865]: Traceback (most recent call last):
python3[1865]:   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 589, in load_all_named
python3[1865]:     load(package, load_as, prefix)
python3[1865]:   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 579, in load
python3[1865]:     raise e
python3[1865]:   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 576, in load
python3[1865]:     func()
python3[1865]:   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/components/publishing/__init__.py", line 36, in initialize
python3[1865]:     _PUBLISH_SERVER_THREAD = pub.server.PublishServer(tcp_port=tcp_port, websocket_port=ws_port)
python3[1865]:   File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/publishing/server.py", line 211, in __init__
python3[1865]:     self.backend.bind(f'ws://*:{websocket_port}')
python3[1865]:   File "/home/pi/.local/lib/python3.9/site-packages/zmq/sugar/socket.py", line 302, in bind
python3[1865]:     super().bind(addr)
python3[1865]:   File "zmq/backend/cython/socket.pyx", line 564, in zmq.backend.cython.socket.Socket.bind
python3[1865]:   File "zmq/backend/cython/checkrc.pxd", line 28, in zmq.backend.cython.checkrc._check_rc
python3[1865]: zmq.error.ZMQError: Protocol not supported (addr='ws://*:5557')
```

votti, 2023-08-14:
I had the same issue, re-installing jukebox on a brand new Rasbian.
The solution for me was to force the re-compilation of pyzmq as also reported here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1838#issuecomment-1546893862

```
pip uninstall pyzmq
apt-get install libzmq3-dev
pip install --no-binary=:all: pyzmq
```

I think somehow the installation of the pre-compiled `zmq` does currently not work any more on `future3/develop`.

pabera, 2023-08-14:
@votti is correct. We need to look into the install script again and make sure we update all dependencies. `pyzmq` has seen some updates and they need to be incorporated.

Chilipp, 2023-09-21:
hey! just a quick comment on the solution by @votti in https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2050#issuecomment-1677140157. you should rather use `pip3 install --no-binary=pyzmq pyzmq`. Otherwise I ran into https://github.com/pypa/setuptools_scm/issues/918 (that is actually already closed, but apparently not for `pyzmq`).

DonPavlov, 2023-10-22:
I tried following the steps. But my pi failed to install pip packages. I had to do the following before I could install something via pip.
  
 Create and add the following lines to ~/.config/pip/pip.conf:

    [global]
    break-system-packages = true

s-martin, 2023-10-22:
I assume you use new Raspbian bookworm?

That's because bookworm now enforces PEP668, see also here: https://stackoverflow.com/a/75696359

Not sure yet, what's the best option...

DivineDominion, 2023-11-09:
I believe that current `future3/develop` contains all the fixes needed to close this issue. @Shaka-Agina could you try to install the latest dev version? Command from the docs for your convenience:

    cd; GIT_USER='MiczFlor' GIT_BRANCH='future3/develop' bash <(wget -qO- https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/future3/develop/installation/install-jukebox.sh)

Chilipp, 2023-11-09:
I just tried it @DivineDominion on a fresh 32bit bookworm os lite (`6.1.0-rpi4-rpi-v7l #1 SMP Raspbian 1:6.1.54-1+rpt2 (2023-10-05) armv7l`) and RPI4 and the problem still persists. The only solution was to run 

```bash
ZMQ_DRAFT_API=1 ZMQPREFIX="/usr/local" pip3 install --no-binary=pyzmq --break-system-packages pyzmq
```

i.e. adding the `--break-system-packages` option to the `pip3 install` command as mentioned in https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2050#issuecomment-1774101077 above by @DivineDominion  and @s-martin (I'd prefer setting this option instead of modifying the `pip.conf` file as in https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2050#issuecomment-1774081800 by @DivineDominion). The same option had to be added in `_jukebox_core_install_python_requirements`, see https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/0660586f7ec96aea75006cc8df8b86e1d3d11542/installation/routines/setup_jukebox_core.sh#L111

I also added a fix for the installation of npm as the current installation method has been deprecated (and does not install npm anymore) and there has been an issue with the installation of `pyalsaaudio` during the installation of the `requirements.txt` which is why I had to install `libasound2-dev` via `apt-get`

<details>

Here is the traceback during the invocation of `pip3 install ... -r .../requirements.txt`

```
Installing collected packages: snowballstemmer, pyalsaaudio, pulsectl, evdev, tornado, sphinxcontrib-jsmath, ruamel.yaml.clib, python_mpd2, Pygments, pyflakes, pycodestyle, pluggy, packaging, mock, mccabe, MarkupSafe, iniconfig, imagesize, docutils, babel, alabaster, ruamel.yaml, pytest, Jinja2, flake8, sphinxcontrib-serializinghtml, sphinxcontrib-qthelp, sphinxcontrib-htmlhelp, sphinxcontrib-devhelp, sphinxcontrib-applehelp, sphinx, sphinxcontrib-jquery, sphinx_rtd_theme
  Running setup.py install for pyalsaaudio ... error
  error: subprocess-exited-with-error
  
  × Running setup.py install for pyalsaaudio did not run successfully.
  │ exit code: 1
  ╰─> [14 lines of output]
      running install
      /usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      running build_ext
      building 'alsaaudio' extension
      creating build
      creating build/temp.linux-armv7l-cpython-311
      arm-linux-gnueabihf-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/usr/include/python3.11 -c alsaaudio.c -o build/temp.linux-armv7l-cpython-311/alsaaudio.o
      alsaaudio.c:28:10: fatal error: alsa/asoundlib.h: No such file or directory
         28 | #include <alsa/asoundlib.h>
            |          ^~~~~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> pyalsaaudio

```
</details>

If you like, I can open a PR with these changes, see https://github.com/MiczFlor/RPi-Jukebox-RFID/compare/future3/develop...Chilipp:RPi-Jukebox-RFID:fix-python-install?expand=1 

the installation (after implement the fix for the arm64 OS in https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2041) was successful via

```bash
cd; GIT_USER='Chilipp' GIT_BRANCH='fix-python-install' bash <(wget -qO- https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/future3/develop/installation/install-jukebox.sh)
```

and the `jukebox-daemon.service` was running successfully and the webapp has been built

pabera, 2023-11-10:
> If you like, I can open a PR with these changes

That would be great. Bookworm has not been tested with future3. Maybe we should use your PR also to align to future3/main.

Chilipp, 2023-11-10:
alright, PR is open in #2100

fellek, 2023-11-11:
Maybe related Errors for Jukebox in Docker-Compose
Running Bookworm on Raspberry Pi 4B

```
server.py - jb.rpc.server - MainThread - DEBUG - Connected to address 'tcp://*:5555'
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/run_jukebox.py", line 67, in <module>
    main()
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/run_jukebox.py", line 63, in main
    myjukebox.run()
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/daemon.py", line 227, in run
    self.rpc_server = RpcServer()
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/rpc/server.py", line 86, in __init__
    self.socket.bind(websocket_address)
  File "/home/pi/.local/lib/python3.9/site-packages/zmq/sugar/socket.py", line 302, in bind
    super().bind(addr)
  File "zmq/backend/cython/socket.pyx", line 564, in zmq.backend.cython.socket.Socket.bind
  File "zmq/backend/cython/checkrc.pxd", line 28, in zmq.backend.cython.checkrc._check_rc
zmq.error.ZMQError: Protocol not supported (addr='ws://*:5556')
````

s-martin, 2023-11-16:
Should be now fixed on `future3/develop`

https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2108#issuecomment-1809167778

can you check this?

s-martin, 2023-12-08:
Fixed

PR #2100 (2023-11-11): Fix future3 installation for bookworm
Hey! as described in https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2050#issuecomment-1804785695 this PR fixes the installation of the future3/branch on bookworm. In particular it adds additional options for the `pip3` installations and it installs nodejs and npm from the official sources.

closes #2050
closed 2023-12-08 by s-martin