MiczFlor/RPi-Jukebox-RFID #2056: Service does not start on reboot on future3

votti, 2023-08-14:
### Describe your problem 

Installing jukebox from `future3/develop` did not autostart the service on a brand new Rasbian lite.
The reason was that on the recent Raspbian user services do not seem to start with reboots per default.
This can be fixed by running once:
```
loginctl enable-linger
```
which enables user services to start at reboot.
This should be done by default

#### What's your hardware set up? 

Raspberry Pi 3 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
Linux phnonie 6.1.21-v7+

#### If possible, try to attach logs from ... (paths from RPi)
Running `systemctl --user list-units --type=target ` immediately after reboot and login shows that the `default.target` (=target of jukebox service) is not started yet but just starting.

pabera, 2023-09-17:
Just a question

> did not autostart the service

Which "service" are you referring to? `loginctl` is required to to login as a user but this was never a problem before.

s-martin, 2023-12-17:
3.4 has been released and so far this was not reproducible.
Closing, but feel free to reopen, if the issue still occurs for you.

closed 2023-12-17 by s-martin