MiczFlor/RPi-Jukebox-RFID #2066: Installation of RFID reader RC522 fails due to deleted repo

jo-me, 2023-09-18:
### Describe your problem 

The repository used to install the RC522 library apparently disappeared: https://github.com/ChisSoc/pi-rc522.git

It is configured in 
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/src/jukebox/components/rfid/hardware/rc522_spi/requirements.txt

The commented out one (kevinvalk) seems to work.

#### What's your hardware set up? 

RPi2. RC522

s-martin, 2023-09-18:
Hey @pabera, @MiczFlor,

looks like ChisSoc deleted his GitHub account.

Do you remember why we forked the original repo?

pabera, 2023-09-18:
I don't remember it. This was @ChrisSoc's contribution only. In the [config](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/src/jukebox/components/rfid/hardware/rc522_spi/requirements.txt) mentioned above they explain the reasoning behind it. We should be good with what @jo-me suggested. Would required a PR.

s-martin, 2023-10-10:
@AlvinSchiller i think we need to fix this in 2.x as well: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/rfid-reader/RC522/requirements.txt

s-martin, 2023-10-11:
I made a PR #2078 :)

s-martin, 2023-11-09:
#2075 and #2078 are merged, so this should be fixed on the develop branches

closed 2023-11-09 by s-martin