MiczFlor/RPi-Jukebox-RFID #2094: setup_jukebox_core.sh on device doesn't use pre-built pyzmq on future3

DivineDominion, 2023-11-07:
### Describe your problem 

`_jukebox_core_download_prebuilt_pyzmq`, introduced in e8945ed67bf37adc4d7a3a562f3174d37e4de592, is never called/used.

So in the end, after `libzmq` is fetched from Google Drive, `pyzmq` is then fetched without WebSockets support via `pip3`. So the intended workaround to use a pre-built binary mentioned here is not being utilized:

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/5e35ce86f05c377cf935d432ac00b8138503acaa/installation/routines/setup_jukebox_core.sh#L72-L78

#### What's your hardware set up?

- Raspberry Pi 3 Model B Plus Rev 1.3
- RPi OS Lite, 32 Bit, v11 (Bullseye)

DivineDominion, 2023-11-07:
Maybe actually using  the pre-built binary would also fix #2050

Or build pyzmq properly as mentioned there. I'll explore both and report back

pabera, 2023-11-08:
The reason why we implemented both options was the following: 1) Build zmq on low performance Pi's failed. This could even be the case for a Pi 3a or 2, 1 or Zero. They don't have enough memory. Therefore we pre-built the binaries and offer them via download. It's arguable whether a Google Drive link is a great download resource (maybe this should be a Github link). 2) Building zmq takes a long time on any Pi and it speeds up the installation process of the Phoniebox.
The reason why we offer building it on the Pi was the variations of the systems, Pi 4, now Pi 5 and 32bit vs. 64bit.
Maybe ZMQ now supports WebSockets by default and all of this is not required. This should be explored :-)

AlvinSchiller, 2023-11-08:
@pabera i currently have a 3A+ i work with. And the installation failed at the first try. But after increasing the swapfile the build was successfull (still takes a while ^^).
This could also be part of the installation, if it makes things easier.

DivineDominion, 2023-11-08:
@pabera Thanks! I understand the reasons. I could not find any call of `_jukebox_core_download_prebuilt_pyzmq` in git (via `git -S"_jukebox_core_download_prebuilt_pyzmq"`). So it looks to me like this function was never used after it was added? That is the confusing part: because the idea to offer a binary download sounds sensible, but it's not taking any effect and the installation boils down to use `pip3 install pyqmt` in the end, including download progress and such in the log.

DivineDominion, 2023-11-08:
> @pabera i currently have a 3A+ i work with. And the installation failed at the first try. But after increasing the swapfile the build was successfull (still takes a while ^^). This could also be part of the installation, if it makes things easier.

May I ask how you did install, what error you encountered, and how and increased the swap file? (My attempt to install doesn't try to compile anything and I didn't run out of swap space AFAICT, so maybe you're doing something different?)

AlvinSchiller, 2023-11-08:
> @AlvinSchiller
> 
> > i currently have a 3A+ i work with. And the installation failed at the first try. But after increasing the swapfile the build was successfull (still takes a while ^^). This could also be part of the installation, if it makes things easier.
> 
> May I ask how you did install, what error you encountered, and how and increased the swap file? (My attempt to install doesn't try to compile anything and I didn't run out of swap space AFAICT, so maybe I'm doing something different?)

I installed the `develop` branch, and choose to build locally. But i can try again later and get more details for you. The 3A+ only has 500 MB Ram which is slightly to low. The 3B+ with 1 GB shouldn't need the swapfile change.
Haven't tried `main` yet. Which branch you are using?

pabera, 2023-11-08:
@DivineDominion I really don't remember why this code was added (https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/e8945ed67bf37adc4d7a3a562f3174d37e4de592) but never called. I remember the reasons (explained above), but maybe it was more for testing purposes. I recall that it was pretty costly to prebuild all these environments etc, maybe it never worked out well. 🤷‍♀️

DivineDominion, 2023-11-08:
@AlvinSchiller

> I installed the `develop` branch, and choose to build locally. But i can try again later and get more details for you. The 3A+ only has 500 MB Ram which is slightly to low. The 3B+ with 1 GB shouldn't need the swapfile change. Haven't tried `main` yet. Which branch you are using?

Sorry for repeating myself so bluntly -- how did you do that exactly? 

I'm on `feature3/develop`, too, because of the more recent compatibility changes to `pip3` usage. 

I would love to rebuild pyzmq, too, but it never does. All I get is a download. I found [how to rebuild the web app in the update instructions](https://rpi-jukebox-rfid.readthedocs.io/en/latest/migration.html#updating-your-jukebox-version-3) but that's not affecting pyzmq.

DivineDominion, 2023-11-08:
Good news. The command should include `--no-binary "pyzmq"` to avoid downloading the latest version and be this:

    sudo ZMQ_PREFIX="${ZMQ_PREFIX}" ZMQ_DRAFT_API=1 \
      pip3 install --no-cache-dir --no-binary "pyzmq" --pre pyzmq

See the updated pyzmq docs:

https://github.com/zeromq/pyzmq/blob/main/docs/source/howto/draft.md

I'm currently executing everything to double check.

DivineDominion, 2023-11-08:
Took a while! But the build instructions work 🎉

--------

See #2096 for a fix.

Also see #2097 to remove the unused function. That would close this issue, because the observation was that this helper isn't used, and the build process didn't work. Now the build process works with the fix. 👍

We could still re-enable using the binary, though? But that sounds like an annoying CI task -- one we can hardly test manually for all architectures. I would suggest to use the manual build process that does what it is supposed to do and leave the convenience download preparation for an enhancement (new issue) for later, because it'd be such a tedious blocker.

AlvinSchiller, 2023-11-09:
> @AlvinSchiller
>
>
> Sorry for repeating myself so bluntly -- how did you do that exactly?
>
> I'm on `feature3/develop`, too, because of the more recent compatibility changes to `pip3` usage.
>
> I would [...]

Sorry for the delay and not sure if it is still relevant, but here my steps for the installation:

- Run installer (from my own fork)
  - cd; GIT_USER='AlvinSchiller' GIT_BRANCH='future3/develop' bash <(wget -qO- https://raw.githubusercontent.com/AlvinSchiller/RPi-Jukebox-RFID/future3/develop/installation/install-jukebox.sh)
- Choose to build WebApp
- Reboot after installation
- Extend Swapfile to 512 MB
  - sudo dphys-swapfile swapoff
  - sudo nano /etc/dphys-swapfile -> change CONF_SWAPSIZE=100 to CONF_SWAPSIZE=512
  - sudo dphys-swapfile setup
  - sudo dphys-swapfile swapon
- Reboot 
- Run WebApp installer
  - cd ~/RPi-Jukebox-RFID/src/webapp
  - ./run_rebuild.sh -u
- Run command for pyzmq from 
  - sudo pip uninstall pyzmq
  - sudo apt-get install libzmq3-dev
  - MQ_DRAFT_API=1 sudo pip3 install --no-binary=pyzmq pyzmq

Phoniebox is Running, WebApp is usable, Songs can be player and RC522 Rfid Card reader is also working

DivineDominion, 2023-11-09:
@AlvinSchiller Thank you! 

-----

With recent changes to the repo, this issue has been fixed. Thanks everyone for the help!

closed 2023-11-09 by DivineDominion