MiczFlor/RPi-Jukebox-RFID #2095: 🚀 | Update npm dependencies on future3

DivineDominion, 2023-11-08:
Checking out `future3/develop`, going through the setup, then after reboot rebuiding the webapp shows:

```
$ cd ~/RPi-Jukebox-RFID/src/webapp && ./run_rebuild.sh -u
Total phys memory: 921 MB
Free phys memory : 675 MB
Free swap memory : 299 MB
Free total memory: 975 MB
Setting Node Options:
NODE_OPTIONS=--max-old-space-size=512
npm WARN deprecated stable@0.1.8: Modern JS already guarantees Array#sort() is a stable sort, so this library is deprecated. See the compatibility table on MDN: https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Array/sort#browser_compatibility
npm WARN deprecated rollup-plugin-terser@7.0.2: This package has been deprecated and is no longer maintained. Please use @rollup/plugin-terser
npm WARN deprecated w3c-hr-time@1.0.2: Use your platform's native performance.now() and performance.timeOrigin.
npm WARN deprecated sourcemap-codec@1.4.8: Please use @jridgewell/sourcemap-codec instead
npm WARN deprecated querystring@0.2.0: The querystring API is considered Legacy. new code should use the URLSearchParams API instead.
npm WARN deprecated svgo@1.3.2: This SVGO version is no longer supported. Upgrade to v2.x.x.

added 1486 packages, and audited 1487 packages in 15m

251 packages are looking for funding
  run `npm fund` for details

13 vulnerabilities (6 moderate, 6 high, 1 critical)
```

I'll be tackling as many of these as I can. Tracking this in this issue for visibility.

`react-scripts build` also tells us about update to `caniuse-lite`, which aruably counts as a dependency, plus linter warnings, which don't :)

```
> webapp@0.1.0 build
> react-scripts build

Creating an optimized production build...
Browserslist: caniuse-lite is outdated. Please run:
  npx update-browserslist-db@latest
  Why you should do it regularly: https://github.com/browserslist/update-db#readme
Compiled with warnings.

[eslint]
src/components/Player/controls.js
  Line 55:6:  React Hook useEffect has missing dependencies: 'setState' and 'state'. Either include them or remove the dependency array  react-hooks/exhaustive-deps

src/components/Player/seekbar.js
  Line 51:6:  React Hook useEffect has missing dependencies: 'isSeeking' and 'updateTimeAndProgress'. Either include them or remove the dependency array  react-hooks/exhaustive-deps
```

pabera, 2023-11-08:
```
[eslint]
src/components/Player/controls.js
  Line 55:6:  React Hook useEffect has missing dependencies: 'setState' and 'state'. Either include them or remove the dependency array  react-hooks/exhaustive-deps

src/components/Player/seekbar.js
  Line 51:6:  React Hook useEffect has missing dependencies: 'isSeeking' and 'updateTimeAndProgress'. Either include them or remove the dependency array  react-hooks/exhaustive-deps
```

They existed from Day 1 🫣. Should be fixed but it's not business critical ;-)

s-martin, 2023-11-17:
This was fixed with #2103, right?

DivineDominion, 2023-11-17:
Correct, closing

closed 2023-11-17 by DivineDominion