MiczFlor/RPi-Jukebox-RFID #2104: 🐛 | Web audio controls dont work on install

tombruton87, 2023-11-12:
## Bug

I installed using the script on the install page using the legacy image in the PI image maker. Everything is working apart from audio volume controls in the web interface. This includes Up down and mute. I have not tested changing volume via GPIO 

### What happened

Volume buttons not working
I have followed https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#raspberry-pi-update-may-2020-broke-my-audio including steps my marcohorstman however I didnt need to make any changes as using interface 0 the built in sound

and all instances calling headphone

AlvinSchiller, 2023-11-12:
Hi, which raspberry pi and os version do you use?
Can you post the output of `amixer scontrols`?

tombruton87, 2023-11-12:
Rasberry Pi4 , version is bullseye

Simple mixer control 'Master',0
Simple mixer control 'Capture',0

AlvinSchiller, 2023-11-12:
> Simple mixer control 'Master',0

Ok then change 'headphone' to 'master', reboot and try again

tombruton87, 2023-11-12:
In 1 or all 3 places?

AlvinSchiller, 2023-11-12:
In all places

tombruton87, 2023-11-12:
changed it in all 3 places in the first post, I also get an error when using USB buttons (custom script)

expr: syntax error: unexpected argument '3'
/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh: line 412: [: -le: unary operator expected
OK MPD 0.22.4

AlvinSchiller, 2023-11-12:
Did you use lowercase 'master'? Please try 'Master' as this could make a difference.
Also what type of audio output do you use?

AlvinSchiller, 2023-11-12:
> expr: syntax error: unexpected argument '3'
> /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh: line 412: [: -le: unary operator expected
> OK MPD 0.22.4

Can you post the content of your "settings/global.conf" file? Looks like "AUDIOVOLMAXLIMIT" is not defined.

AlvinSchiller, 2023-11-12:
If setting to 'Master' alone does not help, you could try setting "mixer_type" in mpd.conf from "hardware" to "software" (https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2023)

tombruton87, 2023-11-12:
output of global config

AUDIOFOLDERSPATH="/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
PLAYLISTSFOLDERPATH="/home/pi/RPi-Jukebox-RFID/playlists"
SWIPEORPLACE="SWIPENOTPLACE"
SECONDSWIPE="RESTART"
SECONDSWIPEPAUSE="2"
SECONDSWIPEPAUSECONTROLS="ON"
AUDIOIFACENAME="Master"
AUDIOIFACEACTIVE="0"
VOLUMEMANAGER="mpd"
AUDIOVOLCHANGESTEP="3"
AUDIOVOLMAXLIMIT="90"
AUDIOVOLMINLIMIT="1"
AUDIOVOLSTARTUP="0"
AUDIOVOLBOOT="30"
VOLCHANGEIDLE="TRUE"
IDLETIMESHUTDOWN="0"
POWEROFFCMD="sudo poweroff"
SHOWCOVER="ON"
READWLANIPYN="OFF"
EDITION="classic"

AlvinSchiller, 2023-11-15:
So this can be closed?

closed 2023-12-04 by s-martin