MiczFlor/RPi-Jukebox-RFID #2108: ISSUE SUMMARY on future3 | PyZMQ | Websocket Problems | Install Script

s-raeuchle, 2023-11-13:
### Describe your problem 
If I start the web application, i get no data. Dev Console tells me websocket error, connection refused
"WebSocket connection to 'ws://192.168.178.39:5556/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED"

It seems the websocket connection using PyZMQ is not working (see error.log)

During installation there was an error related to that pip package. I installed it afterwards using pip install pyzmq without luck.
Looking in to the requirements.txt tells me I need a special compiled version...

Here my knowledge ends without further invesigation

#### What's your hardware set up? 
- RPi 4, 32 Bit Legacy lite Bullseye, fresh install on ssd over usb

#### If possible, try to attach logs from ... (paths from RPi)

[INSTALL.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/13341933/INSTALL.log)
[error.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/13341935/error.log)

AlvinSchiller, 2023-11-13:
Hi,

these problems should be fixed on the current `future3/develop` branch.
To install use this command 
`cd; GIT_USER='MiczFlor' GIT_BRANCH='future3/develop' bash <(wget -qO- https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/future3/develop/installation/install-jukebox.sh)`

pabera, 2023-11-14:
Also, we suggest to go with Dabian Bookworm now as well.

AlvinSchiller, 2023-11-15:
@s-raeuchle did it work with `future3/develop` ?

s-martin, 2023-11-16:
Possible duplicate to #2050

s-raeuchle, 2023-11-16:
Wow. didn't expect a fast response on that open source project. Thanks a lot.

I'll try and give some Feedback.

@pabera should I go for 32 Bit or 64 Bit Bookworm? 
I would like to use 64Bit because I need extra config 'arm_64bit=0' on /boot/config.txt to make it work.

s-martin, 2023-11-16:
I think only 32bit works at the moment.

See 
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2041
for problems with 64bit.

s-raeuchle, 2023-11-16:
> Hi,
> 
> these problems should be fixed on the current `future3/develop` branch. To install use this command `cd; GIT_USER='MiczFlor' GIT_BRANCH='future3/develop' bash <(wget -qO- https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/future3/develop/installation/install-jukebox.sh)`

OK on the dev of course there is no GUI and I need to build it somehow..badly your docs are down.
see: https://rpi-jukebox-rfid.readthedocs.io/en/latest/install.html

What do I do?

AlvinSchiller, 2023-11-16:
The Docs have been moved to the Repo. Currently available on future3/develop -> https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/future3/develop/documentation

To build the WebApp (and chose this Options during installation) you need to run this commands.
```
cd ~/RPi-Jukebox-RFID/src/webapp;
./run_rebuild.sh -u
```

s-raeuchle, 2023-11-16:
confirm. the bug is fixed on dev! I have no websocket problems at all now!

But I hat to install node & npm myself. The import script routine was not working properly!

this issue can be closed

closed 2023-11-16 by s-raeuchle

AlvinSchiller, 2023-11-16:
Great :) Hope the rest is working too.

> But I had to install node & npm myself. The import script routine was not working properly!

This sould be fixed with https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2111.

s-martin, 2023-11-16:
@s-raeuchle  Please reach us at https://matrix.to/#/#phoniebox_community:gitter.im

Future3 does not use mopidy anymore and we don't plan to do that, so Tidal integration could be more complicated. Also we don't plan to add new features to the old V2.x.

pabera, 2023-11-16:
But you can still join the chat ;-)