MiczFlor/RPi-Jukebox-RFID #2120: 🐛 🛡️ | Security vulnerability disclosure

kazet, 2023-11-21:
Hello,

CERT PL found a security vulnerability in this repository. How can we report this privately? We don't see any security policy describing how such vulnerabilities should be reported.

Groovylein, 2024-01-01:
Hi,
How did CERT PL got attention to this repository?

Do you have a recommendation or example how this can be handled or is handled in other repositories?

lukigruszka, 2024-01-26:
Hi,
CERT PL has performed a broad vulnerability scan of open source projects and that one was tested as well.

Different ways of handling such reports are possible, it depends on the level of confidentiality one wants to achieve. Solutions start from specifying a dedicated e-mail address and a corresponding PGP public key, through web forms with TLS encryption, ending with publicly discussing them as regular issues on GitHub

closed 2024-04-19 by s-martin