MiczFlor/RPi-Jukebox-RFID #2138: Cannot upload audio via web app on future3

Weischer, 2023-12-08:
Hi,

I have setup future3 on a raspberry pi 4b. Everything is working fine, but as I understood I should be able to upload files via the webapp? Currently I can only do so via Windows Explorer.
I am using Chrome. See the following screenshot. I am missing a + or anything to add files. Sorry, if this is not a bug and simply not supported.

[image]

pabera, 2023-12-08:
Uploading via the Webapp is currently NOT possible. Only through Samba (Windows Share).

s-martin, 2023-12-28:
Ideas: 
- https://uppy.io/
- https://github.com/pqina/react-filepond with https://pqina.nl/filepond/
- just a file drag n drop zone without an uploader: https://react-dropzone.js.org/

PNG-pyro, 2026-10-25:
I did this on my own box with copyparty: https://github.com/9001/copyparty 

It's python, so it might be a good fit with this project, and it's MIT licensed, so that's similar too. The interface is a bit quirky, but it's super fast and has all the features I might want.