MiczFlor/RPi-Jukebox-RFID #2183: 🐛 | Wifi Settings in Bookworm

AlvinSchiller, 2023-12-28:
## Bug

### What I did

Set a Wifi via the WebUi

### What happened

The new settings are not used

### I expected this to happen

The settings are used

### Further information that might help
With Bookworm the network settings have changed. Now `NetworkManager` is used.
This breaks settings of "old" wpa_supplicant.conf and dhcpcd.conf.

Static IP and Autohotspot are also affected.

## Software

### Base image and version
Bookworm lite

### Branch / Release
master 2.5.0

### Installscript
`scripts/installscripts/install-jukebox.sh`

## Hardware

### RaspberryPi version
3 A+

s-martin, 2023-12-28:
This does effect both V2 and V3, right?

AlvinSchiller, 2023-12-28:
yep, couldn't test on v3 right now, which features are effected.
But i assume static ip and autohotspot (already known).

s-martin, 2023-12-28:
Maybe useful references:
- https://forums.raspberrypi.com/viewtopic.php?t=357623
- https://networkmanager.dev/docs/api/latest/

`nmcli` may be the command to go.

AlvinSchiller, 2023-12-28:
Yep nmcli was working on manual test. Need to implement a switch for the features, depending on OS Version. 
Or maybe there is still comon ground?

AlvinSchiller, 2024-01-15:
### Temporary workaround
Before running the installation, the network config has to be changed via raspi-config, to use the "old" dhcpcd network settings.

> [!IMPORTANT]
> If the settings are changed, your network will reset and Wifi will not be configured, so you lose ssh access via wireless network.
> So make sure you perform the following steps in a local terminal with a connected monitor and keyboard.

Change network config
* run `sudo raspi-config`
* select `6 - Advanced Options`
* select `AA - Network Config`
* select `dhcpcd`

If you need Wifi, add the information now
* select `1 - System Options`
* select `1 - Wireless LAN`
* enter Wifi information

closed 2024-03-19 by AlvinSchiller