MiczFlor/RPi-Jukebox-RFID #2191: Websocket connections fail on firefox

powertomato, 2024-01-01:
### Describe your problem 
The UI only loads partially, dynamic values like volume, CPU temperature ect. display a permanent spinner-icon.
In the dev-console it is apparent that the connection to the websocket ws://192.168.1.222:5556 and *:5557, respectively (where the IP is the IP of the box)

Connection on same machine using chrome browser works.

The only difference I can spot is the header value `Sec-Websocket-Extensions` on the ws-request:

Firefox: `Sec-WebSocket-Extensions: permessage-deflate`
Chrome: `Sec-Websocket-Extensions: permessage-deflate; client_max_window_bits`

#### What's your hardware set up? 

Rock64 with DietPi

Log:
```
01.01.2024 20:51:01 -   41:daemon.py          - jb.daemon            - MainThread      - INFO     - Starting Jukebox Daemon (Version 3.5.0-alpha)
01.01.2024 20:51:01 -   44:daemon.py          - jb.daemon            - MainThread      - INFO     - Git state: c122387e [2024-01-01] Add NFCPy support  (HEAD -> feature/nfcpy_backend, private/feature/nfcpy_backend) [v3.5.0-alpha-7-gc122387e]
01.01.2024 20:51:01 -  317:cfghandler.py      - jb.cfghandler        - MainThread      - INFO     - (jukebox) Loading yaml file '/home/pi/RPi-Jukebox-RFID/shared/settings/jukebox.yaml'
01.01.2024 20:51:01 -  230:cfghandler.py      - jb.cfghandler        - MainThread      - DEBUG    - (jukebox) Replacing current config data
01.01.2024 20:51:01 -   53:daemon.py          - jb.daemon            - MainThread      - INFO     - Welcome to Jukebox
01.01.2024 20:51:01 -   54:daemon.py          - jb.daemon            - MainThread      - INFO     - Time of start: Mon Jan  1 20:51:01 2024
01.01.2024 20:51:01 -  549:plugs.py           - jb.plugin            - MainThread      - INFO     - Loading plugin 'components.publishing' as 'publishing'
01.01.2024 20:51:01 -  281:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Enlisting function 'components.publishing.republish' as 'publishing.republish' (<class 'function'>)
01.01.2024 20:51:01 -  574:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Package load initializer: calling publishing.initialize()
01.01.2024 20:51:01 -  229:server.py          - jb.pub.server        - MainThread      - DEBUG    - PublishServer initialized (Pyzmq version: 25.1.2; ZMQ version: 4.3.4; has draft API: False)
01.01.2024 20:51:01 -  549:plugs.py           - jb.plugin            - MainThread      - INFO     - Loading plugin 'components.volume' as 'volume'
01.01.2024 20:51:01 -  464:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.toggle_output
01.01.2024 20:51:01 -  464:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.get_outputs
01.01.2024 20:51:01 -  464:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.publish_volume
01.01.2024 20:51:01 -  464:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.publish_outputs
01.01.2024 20:51:01 -  464:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.set_volume
01.01.2024 20:51:01 -  464:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.get_volume
01.01.2024 20:51:01 -  464:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.change_volume
01.01.2024 20:51:01 -  464:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.get_mute
01.01.2024 20:51:01 -  464:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.mute
01.01.2024 20:51:01 -  464:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.set_output
01.01.2024 20:51:01 -  464:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.set_soft_max_volume
01.01.2024 20:51:01 -  464:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging PulseVolumeControl.get_soft_max_volume
01.01.2024 20:51:01 -  574:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Package load initializer: calling volume.initialize()
01.01.2024 20:51:01 -  249:__init__.py        - jb.pulse             - PulseMon        - INFO     - Start Pulse Monitor Thread
01.01.2024 20:51:01 -  264:server.py          - jb.pub.server        - PubServer       - INFO     - New subscription for topic b'batt_status'
01.01.2024 20:51:01 -  264:server.py          - jb.pub.server        - PubServer       - INFO     - New subscription for topic b'core.plugins.loaded'
01.01.2024 20:51:01 -  264:server.py          - jb.pub.server        - PubServer       - INFO     - New subscription for topic b'core.started_at'
01.01.2024 20:51:01 -  264:server.py          - jb.pub.server        - PubServer       - INFO     - New subscription for topic b'core.version'
01.01.2024 20:51:01 -  614:__init__.py        - jb.pulse             - MainThread      - INFO     - Ignoring secondary audio output configuration because it is missing or incomplete
01.01.2024 20:51:01 -  264:server.py          - jb.pub.server        - PubServer       - INFO     - New subscription for topic b'host.temperature.cpu'
01.01.2024 20:51:01 -  353:__init__.py        - jb.pulse             - MainThread      - DEBUG    - Configured audio sinks: [PaSink(alias='Built-in speakers', pulse_sink_name='alsa_output.platform-analog-sound.stereo-fallback', volume_limit=100)]
01.01.2024 20:51:01 -  264:server.py          - jb.pub.server        - PubServer       - INFO     - New subscription for topic b'host.timer.cputemp'
01.01.2024 20:51:01 -  549:plugs.py           - jb.plugin            - MainThread      - INFO     - Loading plugin 'components.jingle' as 'jingle'
01.01.2024 20:51:01 -  264:server.py          - jb.pub.server        - PubServer       - INFO     - New subscription for topic b'playerstatus'
01.01.2024 20:51:01 -  264:server.py          - jb.pub.server        - PubServer       - INFO     - New subscription for topic b'rfid.card_id'
01.01.2024 20:51:01 -  264:server.py          - jb.pub.server        - PubServer       - INFO     - New subscription for topic b'volume.level'
01.01.2024 20:51:01 -  281:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Enlisting function 'components.jingle.play' as 'jingle.play' (<class 'function'>)
01.01.2024 20:51:01 -  281:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Enlisting function 'components.jingle.play_startup' as 'jingle.play_startup' (<class 'function'>)
01.01.2024 20:51:01 -  281:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Enlisting function 'components.jingle.play_shutdown' as 'jingle.play_shutdown' (<class 'function'>)
01.01.2024 20:51:01 -  574:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Package load initializer: calling jingle.initialize()
01.01.2024 20:51:01 -  549:plugs.py           - jb.plugin            - MainThread      - INFO     - Loading plugin 'components.jingle.alsawave' as 'jingle.alsawave'
01.01.2024 20:51:02 -  464:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Tagging AlsaWave.play
01.01.2024 20:51:02 -  321:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Decorating class components.jingle.alsawave.AlsaWave for auto-assignment into package 'jingle.alsawave'
01.01.2024 20:51:02 -  574:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Package load initializer: calling jingle.alsawave.initialize()
01.01.2024 20:51:02 -  340:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Instantiating decorated class 'components.jingle.alsawave.AlsaWave' and register as 'jingle.alsawave.alsawave'
01.01.2024 20:51:02 -  293:plugs.py           - jb.plugin            - MainThread      - DEBUG    - Enlisting an instance of class 'components.jingle.alsawave.AlsaWave' as 'jingle.alsawave.alsawave'
01.01.2024 20:51:02 -   24:__init__.py        - jb.jingle            - MainThread      - DEBUG    - Register 'wav' in <class 'components.jingle.JingleFactory'>.
01.01.2024 20:51:02 -  549:plugs.py           - jb.plugin            - MainThread      - INFO     - Loading plugin 'components.jingle.jinglemp3' as 'jingle.jinglemp3'
[... cut ...]
01.01.2024 20:51:04 -  151:__init__.py        - nfc.clf              - read_00Thread   - INFO     - using ACS ACR122U PN532v1.6 at usb:001:002
01.01.2024 20:51:04 -  155:__init__.py        - jb.rfid(read_00)     - read_00Thread   - DEBUG    - card_removal_timer_thread.native_id = 281472726135264
01.01.2024 20:51:04 -   69:generic_nfcpy.py   - jb.rfid.new(read_00) - read_00Thread   - DEBUG    - Wait for card
01.01.2024 20:51:04 -  699:plugs.py           - jb.plugin.call       - StartJingle     - DEBUG    - Calling: volume.ctrl.set_volume(args=[40], kwargs=None)
```

s-martin, 2024-01-01:
@powertomato, which branch/version are you using?

@pabera, is this
```
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 807, in get
    raise NameError(f"Plugin object '{plugin}' not registered in package '{package}'")
NameError: Plugin object 'ctrl' not registered in package 'player'
```

fixed by #2179?

AlvinSchiller, 2024-01-01:
Firefox in genereal support should be fixed with the last release.
Though its not tested and officially supported for Rock64 or dietpi.

Did you update to 3.5.0-develop or made a new installation? Can you share your jukebox.yaml and installation log?

AlvinSchiller, 2024-01-01:
> ZMQ version: 4.3.4; has draft API: False

The correct version should be `4.3.5` with draft API. That's the reason Firefox is not working.
The installation log would be needed to investigate why this version is installed.

pabera, 2024-01-01:
> NameError: Plugin object 'ctrl' not registered in package 'player'

This shouldn't be a Firefox issue. It's usually a hint that the jukebox-daemon or the MPD is not running.

It's what your stack trace also reports.

```
01.01.2024 20:51:02 -  592:plugs.py           - jb.plugin            - MainThread      - ERROR    - Ignoring failed package load for 'playermpd'
01.01.2024 20:51:02 -  593:plugs.py           - jb.plugin            - MainThread      - ERROR    - Reason: TypeError: expected str, bytes or os.PathLike object, not NoneType
01.01.2024 20:51:02 -  594:plugs.py           - jb.plugin            - MainThread      - ERROR    - Detailed reason:
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 589, in load_all_named
    load(package, load_as, prefix)
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 579, in load
    raise e
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 576, in load
    func()
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/components/playermpd/__init__.py", line 689, in initialize
    player_ctrl = PlayerMPD()
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/components/playermpd/__init__.py", line 161, in __init__
    self.coverart_cache_manager = CoverartCacheManager(os.path.expanduser(coverart_cache_path))
  File "/usr/lib/python3.9/posixpath.py", line 231, in expanduser
    path = os.fspath(path)
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

The issue is related to the recent PR #2177. Maybe there is a bug which we have not captured yet.

Can you check if your `jukebox.yaml` has the following [2 lines added](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/develop/resources/default-settings/jukebox.default.yaml#L148-L149)?

powertomato, 2024-01-02:
Since I deleted the original installation log (sorry), I reran the setup-script. Here are the logs that resulted from that:

[INSTALL-1704151233.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/13806348/INSTALL-1704151233.log)
[app.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/13806350/app.log)
[jukebox.yaml.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/13806351/jukebox.yaml.txt)

All the Errors are now gone from the log. But the Firefox issue persists.

Note:
The setup is using my fork, which branched from commit 3a904b5f97d107806a8fe31dc21256577b3996bf (future3/develop)
The only changes are an addition of a card-reader backend (see [PR #2190](https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2190))

powertomato, 2024-01-02:
> > ZMQ version: 4.3.4; has draft API: False
> 
> The correct version should be `4.3.5` with draft API. That's the reason Firefox is not working. The installation log would be needed to investigate why this version is installed.

This was the issue. I removed pyzmq and rerun the setup, now the correct version was built and the webapp works in firefox too.

So the only problem I see is, the setup should check for the correct pyzmq version, not only if pyzmq is installed.

AlvinSchiller, 2024-01-02:
There are different problems here:
- Your pyzmq was installed with the wrong libzmq version (4.3.4) and without draft support. Thats why firefox is not working.
- Your playermpd couldn't be initialised `ERROR    - Ignoring failed package load for 'playermpd'`. As @pabera wrote in his [comment](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2191#issuecomment-1873505246) this is propably caused due to the missing config entry in your `jukebox.yaml`. 

**pyzmq**
- Its currently not supported to rerun the installation script, so this installation log is sadly not getting us closer to the initial problem (additionally the pyzmq installation was skipped as it was already installed). Can you recall if you chose the same options in the initial installation? e.g. webapp, node, ...?
- The logs write out you kernel architecture as "aarch64". This should abort the installation as it is currently not supported either. Did you manually changed the script after download on the device?

**playermpd**
- Did you copy this from an old installation or was your initial installation prior this commit (< 22.12.23) and you updated later? Otherwise i dont know why it should be missing. You fork has the commit included.

AlvinSchiller, 2024-01-02:
> All the Errors are now gone from the log. 

Thats strange, as the entry is still missing in your `jukebox.yaml`. So this should not behave different.

> This was the issue. I removed pyzmq and rerun the setup, now the correct version was built and the webapp works in firefox too.

Good to know :) Strange it was a wrong version in the first place. Do you maybe have a second sd card to test a fresh installation?

> So the only problem I see is, the setup should check for the correct pyzmq version, not only if pyzmq is installed.

Agree, we should evaluate if this can be easily tested. The problem is its not the pyzmq version itself, but the dependency to libzmq its build with.

s-martin, 2024-01-02:
@pabera @AlvinSchiller 
I was wondering, if we find a mechanism to add config entries with default values to jukebox.yml, if the entry is not there yet.

Especially the cover art cache isn't something a user would need to customize, but usually could go with a default entry.

In general this could help us to make the installation process more "updateable".
Right now it's "flash the card", but if something goes wrong many users will try to rerun the script anyway.

s-martin, 2024-01-02:
> Good to know :) Strange it was a wrong version in the first place. Do you maybe have a second sd card to test a fresh installation?
>
>
> Agree, we should evaluate if this can be easily tested. The [...]

Do we need to pin the versions of libzmq and pyzmq?
Could also help with #2167

AlvinSchiller, 2024-01-02:
>  I was wondering, if we find a mechanism to add config entries with default values to jukebox.yml, if the entry is not there yet.

I've done something similiar for the reader config, only there the check is for the file existing. Could also be possible for single entries? https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/3a904b5f97d107806a8fe31dc21256577b3996bf/src/jukebox/components/rfid/reader/__init__.py#L241

AlvinSchiller, 2024-01-02:
> Do we need to pin the versions of libzmq and pyzmq?

libzmq cant be pinned, as a prebuild bundle is used or it needs to be build locally. And i think thats also true for pyzmq as it needs to be build against the local libzmq to support draft API.

s-martin, 2024-01-02:
> I've done something similiar for the reader config, only there the check is for the file existing. Could also be possible for single entries? [...]

Yeah, something like that.

In the past I had a project, which had a mechanism for creating default values, if the parameters or values weren't in the actual file. Made it also easier to always create a config file matching the code instead of making sure the shipped file matches the code. But this was for ini files :-/

s-martin, 2024-03-03:
Closing now as completed.

Feel free to reopen, if the issue persists for you.

closed 2024-03-03 by s-martin