MiczFlor/RPi-Jukebox-RFID #2203: cover art causes corrupts player state on future3

AlvinSchiller, 2024-01-03:
### Describe your problem 

After copying some sound files into the shared folder (fresh installation) with embeded covers and access the library in the webui the covers are correctly shown, but no playback is possible.
Only after a restart of the service (or pi), the playback is working and cover art is correctly shown.

This can behavior can be recreated if the `cover-cache` folder is emptied.

#### What's your hardware set up? 
Pi 3 A+, Rasberry Pi OS Bookworm 32 bit lite, RC522 RFID Reader

#### If possible, try to attach logs from ... (paths from RPi)
```
04.01.2024 00:37:05 -  247:__init__.py        - jb.PlayerMPD         - mpd.timer_status - ERROR    - UnicodeDecodeError: 'utf-8' codec can't decode byte 0xac in position 1: invalid start byte
04.01.2024 00:38:06 -  526:__init__.py        - jb.PlayerMPD         - MainThread      - ERROR    - ConnectionError: Connection lost while reading binary data: expected 8192 bytes, got 180 at uri Electric Callboy - 2022 - Tekkno/Electric Callboy - Tekkno - 01 - Pump It.mp3
```

s-martin, 2024-01-04:
Cheap fix would be a refresh button to restart the service, but not very user friendly

pabera, 2024-01-04:
I am aware of it. It's a problem with MPD and occurs randomly. I have an idea, but it will take some time to be resolved.

s-martin, 2024-04-15:
#2325 fixed this on future3/develop.

Closing. Feel free to reopen, if your issue still persists.