MiczFlor/RPi-Jukebox-RFID #2209: KioskMode not working on armv6 on future3

AlvinSchiller, 2024-01-06:
### Describe your problem 

Since `Bullseye` the installation fails on choosing option "KioskMode" on an armv6 device cause `chromium-browser` has no installation candidate.

#### What's your hardware set up? 

Docker with armv6 qemu.
Raspberry Pi OS bullseye 32bit lite + bookworm 32bit lite

#### If possible, try to attach logs from ... (paths from RPi)

> Setup Kiosk Mode
>   Install Kiosk Mode dependencies
> E: Package 'chromium-browser' has no installation candidate
>   Configure Kiosk Mode

s-martin, 2024-01-06:
The package is called `chromium` vs `chromium-browser`?

AlvinSchiller, 2024-01-06:
`chromium-browser`
Don't know if it would work with just `chromium` as the code later calls also `chromium-browser`.

I get a Pi Zero 1 the next days, so i could then test this on the device.

s-martin, 2024-01-06:
Here I find only the package name `chromium` for all Debian versions and architectures https://packages.debian.org/search?suite=all&arch=any&searchon=names&keywords=Chromium

But this would mean it never worked... 🤔

s-martin, 2024-01-06:
According to git blame it was changed 3 years ago: https://github.com/MiczFlor/RPi-Jukebox-RFID/blame/03e64ff8680cd214f61fd56a2b6049c82b1b2d36/installation/routines/setup_kiosk_mode.sh#L18

AlvinSchiller, 2024-01-07:
`chromium` apperently has also not installation candidate for armv6 platform
```
E: Package 'chromium' has no installation candidate
```

s-martin, 2024-01-07:
Isn't `armv6` (and `armv7`) rather `armhf` in Debian nomenclature?

AlvinSchiller, 2024-01-07:
As far as i understood till `armv6` it's `armel` and from `armv7` it's `armhf`

s-martin, 2024-01-07:
Googling provided mixed results, which architecture name matches which release name. 😏🫣🤔

AlvinSchiller, 2024-01-07:
https://www.baeldung.com/linux/arm64-armel-armhf-overview
https://wiki.debian.org/ArmPorts
https://wiki.debian.org/RaspberryPi

s-martin, 2024-01-07:
I understand this means at least Raspian uses `armhf` for `armv6` architecture?

_There are or have-been at least two unofficial ports outside the Debian infrastructure.

Raspbian - Uses the hard-float version of the "new" ABI (EABI) like Debian armhf but targets v6 rather than v7. Currently tracks Debian jessie and stretch. Primarily but not exclusively targeted at the Raspberry Pi. GNU Triplet: arm-linux-gnueabihf_

AlvinSchiller, 2024-01-07:
Looks like. But that could mean it behaves different on a real pi.
I will test this as soon as I get the pi zero

pabera, 2024-01-07:
I just installed `chromium-browser` on my Pi 1 successfully. My question would be, does it really make sense to actually offer Kiosk Mode on `armv6` machines, cause it'll be pretty slow anyways. (just this installation took 15 mins)
It uses `armhf`.

I didn't connect a screen yet to actually test it, it's just a wild guess.

Thoughts?

s-martin, 2024-01-07:
Possible approach: detecting slow machines and then suggesting the user on slow machines to not install kiosk mode?

Minimum suggestion ARMv7?

AlvinSchiller, 2024-01-07:
> I just installed `chromium-browser` on my Pi 1 successfully. My question would be, does it really make sense to actually offer Kiosk Mode on `armv6` machines, cause it'll be pretty slow anyways. (the installation itself took 15 mins) It uses `armhf`.
> 
> I didn't connect a screen yet to actually test it, it's just a wild guess.
> 
> Thoughts?

Good to know its working and its "just" an emulation problem. 
Thought of this too. As of what I read it was discontinued for armv6 as it was nearly unusable slow.

I would also suggest either don't offer this feature for armv6 devices or give a warning that it can be slow, like other features/installations already do.

Maybe depends if it would still be usable ? @pabera ?

s-martin, 2024-01-17:
> > Possible approach: detecting slow machines and then suggesting the user on slow machines to not install kiosk mode?
> > Minimum suggestion ARMv7?
> 
> I think we should add this, but give the message that its not supportedf on this device model and can't be installed. I would not just leave it out, as users might ask, why this is not available.

yeah, let's add this, so users know that it will not work

AlvinSchiller, 2024-01-17:
Would it be desired to fix this with another browser?
Some suggest `dillo` as replacement, but i dont know if there are any tutorials.

pabera, 2024-01-17:
Let's not go this route now. We can create an issue to investigate dillo another time.

AlvinSchiller, 2024-01-18:
Kiosk mode install option deactivated with https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2217

closed 2024-01-18 by AlvinSchiller