MiczFlor/RPi-Jukebox-RFID #2210: MP3 files with "-" don't work

s-martin, 2024-01-07:
I'm using 2.5.0 - 4101984b. No Spotify is set up, don't use it.
Regarding the logs, thanks, that's helpful. Here's what I get 
```
Jan  6 20:08:37 jukepi mpd: player: played "NEEDTOBREATH/Favorites/04. Dreams.flac"
Jan  6 20:08:37 jukepi mpd: exception: No such song
Jan  6 20:08:37 jukepi lighttpd[93753]: error adding file:///home/pi/RPi-Jukebox-RFID/shared/audiofolders/Fun Music/Hello - Martin Solveig and Dragonette.mp3: No such song
```
The file is definitely there. When I use SMB to navigate the audio files, I can even play it on my computer. Are there file naming conventions that need to be followed?

Actually, that appears to be the issue. Renaming the file to get rid of the dash seemed to resolve it. Seems weird that a dash would throw this mess though...

_Originally posted by @AStoker in https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/2208#discussioncomment-8034254_

s-martin, 2024-01-07:
Need to check, why dashes in file name don't work, see also https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/2208#discussioncomment-8035377

AStoker, 2024-01-08:
Well, adding more details. After modifying an existing file and trying to play that, it works perfectly fine. So I decide to do a "stupid check" and change the original file that wasn't working (that suddenly did start working after removing the dash) back to the original name that had the dash in it to see if it wouldn't work. Lo and behold, it does work (no errors in the log about missing songs).
I'm trying to reproduce the error (go figure it's reproducible until I'm trying to reproduce it), and I'll see if I can't get consistent results. But while I'm doing this, is there any kind of file scanning process that might be the culprit? Perhaps some process is supposed to be scanning for changes to the filesystem and got stuck, so it never registered that that file existed? Spit-balling here.

s-martin, 2024-01-08:
You could try `mpc update`, which should update the audio database of the mpc after you added/updated audio files.

AStoker, 2024-01-11:
Alright, was able to reproduce it again (files had periods in it this time). After doing `mpc update`, the files played fine. So perhaps it is just revolving around the files not being recognized yet?

s-martin, 2024-01-11:
I would assume it’s because of missing update.

AStoker, 2024-01-12:
Is it not the latest release? Or is v3 ready to rock and I should upgrade to that?

s-martin, 2024-01-17:
> Is it not the latest release? Or is v3 ready to rock and I should upgrade to that?

If you have a working Phoniebox I would not upgrade to v3, because it’s a completely reinstall. Currently there’s no upgrade path, see https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/builders/update.md

If you build new Phoniebox, I would start with [version 3](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/future3/main/documentation) - as long as version 3 already has the features you need, see https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/developers/status.md